genodelabs/genode #1502: Dvorak keymap support in terminal

ehmry, 2015-05-08:
Some of us have forgotten how to use qwerty.

nfeske, 2015-06-02:
With the Turmvilla scenario of issue #1552, I face similar problems. As a vim user, I'd like to swap escape and capslock. However, instead of hard-coding different keyboard layouts in the terminal, I would like to move the keyboard layout configuration to a dedicated ROM module. A ROM module called "keyboard_layout" could contain an XML configuration with all the key mappings. The terminal would obtain this ROM module on startup. Furthermore, it could even register for ROM-module updates. This way, we could switch the keyboard layout at runtime. With this mechanism, the customization of keyboard layouts would be just a matter of adjusting the "keyboard_layout" ROM according to the user's preferences.

chelmuth, 2015-06-02:
This sounds pretty versatile but pragmatic. Could you imagine how this could be extended to keyboard-specific layouts, e.g., notebook-internal DE and external US? Or is this just a bad idea and should better switch layout explicitly when switching the keyboard?

nfeske, 2015-06-02:
I cannot see how we could implicitly address this problem by the keyboard-layout mechanism. But the keyboard-specific remapping of physical keys could be achieved by a lower-level component similar to our input_merger. Ideally, sometime later, my might come up with something similar to the VFS, but for input streams where we could define a chain of filters that calibrate, merge, filter, and remap input events.

To further clarify the difference between the terminal's keyboard layout mechanism and the input-session interface: In contrast to low-level stream of input events (press, release of physical switches), the keyboard layout has the functionality of mapping a key + modifier state to a character. So it provides a level of abstraction, namely the abstraction of a "character".

ehmry, 2015-06-02:
I would prefer doing this with a layout ROM rather than each config. :+1:

nfeske, 2015-06-02:
@ehmry I'm afraid I caused a bit of confusion by mentioning the VFS. I actually meant to use a separate component if there is a need to remap keys on a physical level (like when connecting different keyboards to the same machine). Sorry for mentioning the VFS.

ehmry, 2015-10-06:
Perhaps input history could also be handled in an input component. This way something like a shell command history can be used in applications like cli_monitor that lack persistent storage. The input component would enter and exit a recall mode when a key combo was entered. Naturally there would have to be some sort of session policy to prevent history leaking where it shouldn't.

nfeske, 2015-10-07:
That is a very interesting idea. It would even allow different applications to share the same history (e.g., browsers of different security domains where the history of one domain would become visible to the browser of the other domain but not vice versa). Food for thought. :-)

cnuke, 2015-10-07:
That is indeed an interesting idea — keep in mind though that merely wrapping the input session may not suffice in all cases, i.e., a shell history should at best only contain valid commands. That would require some kind of feedback mechanism that is used by the client processing the actual input.

nfeske, 2015-10-07:
Indeed, the input session may be not the right interface. The problem has some similarities with the clipboard topic that I have been working on recently. It could be a point of integration of multiple components that complies a configured information-flow policy. But in contrast to the clipboard where I was able to use ROM and Report sessions, those interface do not look like a good match for a history, which could become very large and is frequently updated. So maybe introducing a dedicated session interface would make sense here.

We may also consider addressing not just a command history but the entire class of history-like problems including the browser history or recently used email-addresses. If we tagged history entries with type information (URL, bash command, CLI monitor command, email address, the translation by a dictionary...), one history could actually contain various pieces of data whereby each application could grep for types in the history that are meaningful in its context. For example, it might be pretty cool if an URL that was recently visited in the web browser would automatically pop up in the history of a textual Genode shell as a command for opening the URL.

Jookia, 2015-10-28:
This might not be too relevant, but perhaps it'd be useful to introduce an abstraction of a Unicode stream versus keypresses, this could help in adding IMEs for things like CJK.

ehmry, 2015-10-28:
That couldn't work any worse than SCIM/Fcitx :-P

ehmry, 2016-03-17:
As was brought up on the secure-desktops mailing list, keystroke patterns can reveal the identity of the user, so we have yet another reason for an abstraction between input and a character stream.

http://arstechnica.com/security/2015/07/how-the-way-you-type-can-shatter-anonymity-even-on-tor/

ehmry, 2016-06-13:
I was thinking about this over the weekend, and after looking at the Input session I had an idea. If keycodes are expressed as **int** on input events and there are less than 2^16 keycodes, than a UTF-16 codepoint could be stored in the high keycode bits. This would send both the raw keycodes to SDL applications and expose a character stream to components like terminals.

I think that to map keys to unicode at each input source is more intuitive than to use the same map for keys at every input source at each application. Applications could still remap keys if they wished using the raw codes.

I have very simple mapping working as a static library linked to ``ps2_drv``, but this isn't good enough because real mapping needs a state machine to handle modifier keys. That makes it more complicated, but it would give users the freedom to implement multi-layer and corded keymaps with a relatively simple interface.

This is not a priority for me, but something that has been in the back of my mind for a while.

ehmry, 2017-01-23:
I'm closing this because ``input_remap`` works well enough for now. I am able to use it to create the Workman layout without modifying any code.

closed 2017-01-23 by ehmry