genodelabs/genode #1527: SD-card driver for i.MX6

nfeske, 2015-05-18:
In March, @srinivasprv posted the following SD-card driver for the i.MX6 platform to our mailing list:

https://github.com/srinivasprv/genode/commit/58d436c1419f918b9d8ceb6b71094b71a87405da

Back then, we deferred the inclusion of the driver until Genode officially supports i.MX6. Today, this is the case. The driver looks very good to me. But we will probably need to adapt it to the recent changes of Genode's device-driver infrastructure (e.g., the changed IRQ session interface) @srinivasprv would you like to rebase your driver onto the current version of Genode? On another note, I would very much appreciate you using an email address in your commit message that clarifies the origin of the work (from IIT Madras). Right now, the commit message specifies your gmail address.

nfeske, 2015-05-29:
The version https://github.com/iitmadras/genode/commit/a5cd4c7f8e0cd5cf224aa760bb1383459e3b25d6 seems more current. The commit has still a gmail address and no full name though.

srinivasprv, 2015-05-31:
I'm sorry.
I have a poor internet connection and could not respond sooner.

Unfortunately my project with IIT Madras was complete.
I'll  coordinate with our team at IIT Madras, rebase our work and create a proper commit.

nfeske, 2015-07-22:
Are there any news on this issue? @srinivasprv can you recommend me how to proceed?

srinivasprv, 2015-07-23:
I have modified the code but could not test it due to unavailability of i.MX6 board.
The student, at IIT Madras, who was helping me, is new to this area and of little help.  
Anyway I am going to visit IIT Madras tomorrow. May be I'll be able to test it and update in two days.

nfeske, 2015-07-24:
Thank you for your efforts, which are very much appreciated.

nfeske, 2017-02-28:
Fixed in Genode 17.02

closed 2017-02-28 by nfeske