genodelabs/genode #1528: hw_x86_64: Support for dynamic IRQ mode setting

Kensan, 2015-05-18:
Currently the IRQ trigger mode and polarity on hw_x86_64 is hardcoded. Remove this limitation by adding support for dynamic IRQ mode handling similar to how this is treated on base-foc.

Kensan, 2015-05-26:
I rebased the change on the latest staging branch (394e8f8...) and verified that the **run/libc_ffat** script works on real hardware.

chelmuth, 2015-05-27:
@m-stein do you confirm the patch?

m-stein, 2015-05-27:
@Kensan, @chelmuth 
I enjoyed reading this new and essential feature.
I've added a fix-up with some minor modifications.
Do you agree?

Kensan, 2015-05-27:
@m-stein
I am fine with the changes but would like to point out that I followed the function naming of the base-foc implementation, see https://github.com/genodelabs/genode/blob/master/repos/base-foc/src/core/include/platform.h#L134. Maybe that implementation could be renamed accordingly for consistency's sake?

chelmuth, 2015-05-27:
@m-stein I agree with most of the fixup changes beside renaming `setup_irq_mode` as the function is more than just a plain _setter_.

m-stein, 2015-05-27:
@Kensan @chelmuth 
I removed the renaming from the fix-up.

commit 3fd03657 (2015-05-27): hw_x86_64: Support for dynamic IRQ mode setting

Add a Platform::setup_irq_mode function which enables the IRQ session to
update the trigger mode and polarity of the associated IRQ according to
the session parameters. On ARM this function is a nop.

This change enables the x86_64 platform to support devices which use
arbitrary trigger modes and polarity settings, e.g. AHCI on QEMU and
real hardware.

Fixes #1528.
commit 965d85d5 (2015-05-28): hw_x86_64: Support for dynamic IRQ mode setting

Add a Platform::setup_irq_mode function which enables the IRQ session to
update the trigger mode and polarity of the associated IRQ according to
the session parameters. On ARM this function is a nop.

This change enables the x86_64 platform to support devices which use
arbitrary trigger modes and polarity settings, e.g. AHCI on QEMU and
real hardware.

Fixes #1528.
closed 2015-05-28 by chelmuth