genodelabs/genode #1537: hw: destroy active scheduling context

m-stein, 2015-05-21:
Because of helping, it is possible that a core thread that wants to destroy another thread at the kernel is using the scheduling context of the thread that shall be destroyed at this point in time. When building without ``GENODE_RELEASE`` defined, this always triggers an assertion in the kernel. When building with ``GENODE_RELEASE`` defined, this might silently lead to kernel-memory corruption.

m-stein, 2015-05-21:
I started to enhance the scheduler so that it is aware of destroying the active context but the solution tends to become more complex than I thought. Thus, I've created a simple patch that, at least, prevents silent memory corruption due to this bug for now.

skalk, 2015-06-05:
Already brought up by cproc in issue #1495.

m-stein, 2015-06-08:
Thanks for the hint to the test case.

skalk, 2017-03-08:
@chelmuth @nfeske: please consider commit c8634e1 for staging
@m-stein: I've also tested the cpu_scheduler.run, it still worked

commit 81ae098d (2017-03-08): hw: destroy active scheduling context (fix #1537)

* In addition fixes routes of the cpu_scheduler.run test
closed 2017-03-15 by chelmuth