genodelabs/genode #1538: log_file server

ehmry, 2015-05-23:
A server to direct log messages to a native FS session. Log files are creating with directories and names from session labels. Errors from writing to file are not reported as short writes to log.

ehmry, 2015-05-23:
There are now also copies of utility functions from fs_rom in os/include/file_system/util.h

nfeske, 2015-05-29:
I like the new implementation. But let us consider two points:

* It would be nice to have a separate commit for moving the utility functions from fs_rom to file_system/util.h.
* In my opinion, the new service should replace the existing fs_log service. Otherwise the situation and the naming of the components becomes quite confusing. How about removing the original fs_log service from the libports repository? @astarasikov: Would you be ok with replacing your original implementation with the new one? Since the new server does not depend on the libc, it should reside in the os repository but be named fs_log.

ehmry, 2015-05-29:
I trimmed the utilites off fs_rom and moved them into a second commit. I also added some equity operators to node handles.
Fs_rom is removed and replaced with the new version.

nfeske, 2015-05-30:
Thanks a lot for following my suggestions. One last little thing: Would you mind to add a README file to the component so that users can quickly determine its purpose and the information of how to use it?

ehmry, 2015-05-30:
Yes, I've pushed one since I've opened the issue https://github.com/ehmry/genode/blob/c90f3225f87a5e05913ebcdd638f16349c5aad2f/repos/os/src/server/fs_log/README

ehmry, 2015-05-30:
I changed the noux_bash scenario to use fs_log to make logs available from noux, its on the head of my fs_log-native branch.

nfeske, 2015-05-30:
@ehmry Mea culpa! Sorry for having missed it. :-) That's actually perfect. Also, I like your idea to showcase the server in the noux_bash scenario.

I will still wait one day or two for a comment by @astarasikov before merging your patches, ok?

ehmry, 2015-05-30:
Sounds good

astarasikov, 2015-05-30:
I'm fine with whatever brings Genode development forward :)

ehmry, 2015-05-31:
I just realized that message buffering at the stream is bounded by the size of packets queue, not the bulk buffer, so no real message buffering.

chelmuth, 2015-09-16:
@ehmry I merged f046e2b13e352c6e98f8f4b1d7090d748a3e85e1 and 3679fe3 into staging. If you want to amend something, feel free to add patches.

ehmry, 2015-09-16:
I have a patch for the version I'm currently using, but it requires my ascii_to bool patch to handle the policy 091828a17b04b501a1e88b4024aa35bf96f8bd56

ehmry, 2015-09-29:
I say drop the passthru patch, but https://github.com/ehmry/genode/commit/e87cefa81bc499b44379de1d64eb88564a12a5d2 is good, it makes log merging what it should be.

chelmuth, 2015-09-29:
@ehmry please check staging, which now contains 2b8ccad, cbd3498, and 2b21cef (as fixup for cbd3498).

ehmry, 2015-09-29:
@chelmuth, sorry, but you'll want https://github.com/ehmry/genode/commit/f6fb4f6612a66f5c4d6079ea4a8e52fb780e845e

I used throw(); when I should throw; :\

chelmuth, 2015-09-29:
So shall it be...

closed 2015-09-30 by chelmuth

ehmry, 2015-10-05:
@chelmuth I have a new patch to fix logs names '.log' and to re-truncate files when they are reopened. https://github.com/ehmry/genode/commit/07a569d3211ad67c871c690ddd769509ddea1009

chelmuth, 2015-10-05:
@ehmry I'd prefer if you would split up the two aspects of the commit into separate commits. This makes it easier to review your contribution and to spot specific changes based on the commit message.

ehmry, 2015-10-05:
@chelmuth No problem

commit 2b8ccad4 (2015-09-16): fs_log: new log file server using native FS sessions

Fixes #1538
commit ce1e6c16 (2015-09-30): fs_log: new log file server using native FS sessions

Fixes #1538