genodelabs/genode #1542: x86: consolidate pci and acpi driver into single platform driver

alex-ab, 2015-05-26:


nfeske, 2015-05-26:
Are you considering our discussion about decoupling the ACPI functionality from the PCI using the report-rom mechanism?

alex-ab, 2015-05-26:
Yes, I remember the discussion ;-)

alex-ab, 2015-05-26:
After some direction discussion with @nfeske and @chelmuth it turned out that the both drivers should stay as separate components. So, I will just update the way the acpi_drv communicate the necessary information to the pci_drv (via report_rom) and finally rename the pci_drv to platform_drv.

chelmuth, 2015-05-27:
This issue is closely related to #1359.

alex-ab, 2015-05-27:
Commit 7a8f9ca4189ddd2eb62910db7a422293e0341e23 use a report_rom to transfer the information from the acpi_drv to the pci_drv. The acpi_drv now don't provide any service anymore, just the rom. The pci_drv must now be configured explicitly via the config node, whether it is running with acpi support or not. I adjusted for now 3 run scripts to showcase the changes. Please drop your opinions, before I start renaming all pci_* into platform_*

alex-ab, 2015-05-27:
dd78f527692a022ccfa3e024f6dbd980ef4af1ec showcase the renaming changes (usb_hid, input and lwip run as use cases)

nfeske, 2015-05-28:
That was quick! :-)

I like the patch. There are a few minor suggestions though:
* How do you think about moving the platform_drv part of the run scripts to a separate include file, let's say base/run/platform_drv.inc. This file could extend the ```build_components```, ```config```, and ```boot_modules``` variables while taking the platform differences into account. This would avoid duplicating the config snippet in all run scripts. In principle, this could be done as a separate commit preceding your actual change. Your "convert to platform_drv" patch would then merely touch platform_drv.inc.
* The report_rom should be named acpi_report_rom to avoid a naming conflict when another instance of report_rom is used.

alex-ab, 2015-05-28:
> How do you think about moving the platform_drv part of the run scripts to a separate include file, let's say base/run/platform_drv.inc. This file could extend the build_components, config, and [...]

In principal yes, actually you referring to something intended by #765. The only missing thing is, that for each platform driver you actually need a run script specific policy. Just assume a variable, e.g. '$platform_drv_policy' which must be available if used together with the platform_drv.inc ?

alex-ab, 2015-05-28:
Ok, playing with the way #765 suggests. So I can still specify the policy per run script as argument to the config_platform_drv procedure.

alex-ab, 2015-05-28:
What do you think about 8c744bed728e4ad2c22a5aee34f124ed5e34462a ? I adjusted input.run and lwip.run to show how one could still apply a platform_drv specific policy and if wanted (input.run) a routing node specific to the use case.

chelmuth, 2015-05-28:
Looks quite good except the mandatory `join`. Could we use `append build_components [build_platform_drv]`?

nfeske, 2015-05-28:
This looks already very nice!

I wonder, what is the use case for customizing the session routing for the platform driver? Doesn't it always talk to exclusively to core? If yes, we could drop the second argument of config_platform_drv in my opinion.

Another question, would it be sensible to have some kind of default policy that covers the platform-driver clients that we have in our scenarios, e.g., ps2_drv, usb_drv, fb_drv, pit_drv? Most scenarios would not need all of them but having them in the policy would not hurt either, or would it? This way, most of our run scripts would not need to specify any arguments to the config_platform_drv function.

alex-ab, 2015-05-28:
> Looks quite good except the mandatory join. Could we use append build_components [build_platform_drv]?  

I also don't like the join, but did not found a better solution so far. Your suggestions leads to:

```
building targets:  core init drivers/input drivers/timer test/input
drivers/acpi drivers/pci
spawn make core init drivers/input drivers/timer test/input
invalid command name "drivers/acpi"
    while executing
"drivers/acpi drivers/pci"
    ("eval" body line 2)
    invoked from within

```

alex-ab, 2015-05-28:
Next iteration b9af9b980f7175fd697fbf17250f5bc705207161. Contains updates to all autopilot run scripts as demo.run

nfeske, 2015-05-29:
I propose to simplify the mechanism by introducing the convention that there exist the three global variables ```boot_modules```, ```build_components```, and ```config``` - as is the case in most non-trivial run scripts anyway. Once we have this convention fixed, we can have functions that modify those variables directly. For example, instead of calling ```lappend build_components [build_platform_drv]``` (followed by the non-intuitive ```join```), we would call a function ```append_platform_drv_build_components```. The name says what the function does:

```
proc append_platform_drv_build_components {} {
  global build_components
  lappend ...
  ...
}
```

Granted, this programming style is quite different from our usual best practices with C++ where we generally try to avoid side effects. But since we use Tcl as some kind of a domain-specific language, we may just choose the way that simplifies the run scripts as much as possible.

As a further technique that can be leveraged in Tcl: It is possible to overwrite functions. For example, we could have a function ```platform_drv_policy```, which returns the default policy. In the run script, this function can be implemented as well - somewhere after we included the platform_drv.inc file. This way, the original function gets overwritten by the new one. By representing different aspects as functions (where we would normally pass argument values), we automatically have hooks for customizing them. Another example would be ```acpi_drv_priority```. The benefit is that we don't need to spoil run scripts with arguments when we don't have anything to customize. Since the functions have an unambiguous name, any customization as done by the run script becomes almost self-explanatory (in contrast to argument values passed to functions). For example:

```
source ${genode_dir}/repos/base/run/platform_drv.inc
...
# override defaults of platform_drv.inc
proc acpi_drv_priority {} { return -1 }
proc platform_drv_policy {} {
  return { <policy label="nic_drv" ... > }
}
...
set config ....
...
append_platform_drv_config
...
```

alex-ab, 2015-06-01:
Ok. Reworked the commit again 6b9e03123b9ba742dbb8afe97293782239061121 for autopilot run scripts and demo.run. Please re-check.

nfeske, 2015-06-01:
The new version looks superb (at least to my eyes)!

alex-ab, 2015-06-01:
Then it would be good to add to staging in the next days. If all runs fine, I would actually re-apply the changes of this issue regarding renaming pci driver into platform driver.

alex-ab, 2015-06-04:
Please review dcec21a20aee82f8c2569ec35b3fb0798e28b8a9 and if ok, please add to staging. If it does not break any tests, I would finally rework the renaming patch (pci_* to platform_*) and adjust all drivers accordingly.

nfeske, 2015-06-04:
The patch looks very good. Let's go for it!

alex-ab, 2015-06-08:
Commit 8ad28b4c4869937485271d607e9004738d5bb4ff renames finally all pci_* stuff into platform_*

alex-ab, 2015-06-08:
85ec460e46c4d3537c7eb4f5284093721265effe contains adjustments of pci_drv -> platform_drv for all run scripts

chelmuth, 2015-06-09:
@alex-ab could you please rebase alex-ab/issue_1542 on current staging?

alex-ab, 2015-06-09:
@chelmuth: done

chelmuth, 2015-06-17:
I updated staging (including fixup squashing) just now.

commit 0ec8e43e (2015-06-09): pci: convert to platform_drv

Fixes #1542
commit 0b51aaf6 (2015-06-10): pci: convert to platform_drv

Fixes #1542
commit 8e8f66af (2015-06-10): pci: convert to platform_drv

Fixes #1542
commit 49691079 (2015-06-11): fixup "pci: convert to platform_drv"

Fixes #1542
closed 2015-06-22 by alex-ab