genodelabs/genode #1543: hw & arm_v7: detect faults in core and kernel

m-stein, 2015-05-26:
By now, only faults in core threads are detected and whether the detection succeeds depends on pretty much that isn't needed in this case because it is done pretty late in the corresponding kernel pass. I think it should be detected as early as possible and make use of as few core data as possible to be more reliable. In case, the developer may still add temporary functionality on his own risk.

m-stein, 2015-05-27:
My branch 1543_hw_arm_detect_faults_in_core_and_kernel is based on staging and fixes the issue.

@skalk What do you think about it?

m-stein, 2015-05-29:
@skalk 
I've updated the branch: There was a problem with stack overflows in ARMv6. After solving it was little additional effort for a fully generic exception entry. So, I added that too.

skalk, 2015-06-05:
I very much appreciate the second commit about unifying the assembler pathes for ARMv7 and ARMv6.
To be honest, I do not like the solution to detect kernel page faults. This pure debug feature seems to me far to invasive (touching 138 lines, that are mostly sensitive assembler instructions, and moreover adding 100 lines). In the past when I assumed a kernel fault, I've added a 3 liner for the non-SMP case that worked in 99% to get the offending instruction pointer of the kernel, and worked 100% to get the fault address. Admittedly, my interims solution was nothing that can be used as a generic one, nevertheless I have a bad feeling in putting so much code for a small debug helper into the kernel.
I do not get the point that we have to detect a kernel entry in the kernel as soon as possible. To me it is normally enough to have the fault address and offending instruction pointer. As long as we can detect that the kernel lock-holder is trying to aquire the lock twice (which should be less invasive), we can retrieve the ip saved during the kernel entry, and read the fault address from either IFAR of DFAR.

m-stein, 2015-06-08:
The invasive nature of the commit doesn't arise from fault detection only. Most of the modifications are related to generalization and clean-up. Merely 4 lines were added to the "normal" entry path for realizing the detection. The rest of the detection code can't trigger issues as long as no core fault is in play. I don't think that keeping the crash landing in case of a core fault as short and clear as possible isn't worth the 4 additional lines. In a late approach we would unnecessarily switch the MMU context, save the kernel context to some memory it doesn't belong to and, on core faults, touch several kernel objects (scheduler, threads, drivers, etc.). A long way to crash again and get stuck without providing any useful information to the developer.

m-stein, 2015-06-08:
I have not updated the commits yet as there is a problem with detection on SMP and I'm at other stuff right now.

chelmuth, 2015-06-10:
Reading your discussion I came to the conclusion the issue is not fixed yet. Please correct me if I'm wrong.

m-stein, 2015-06-10:
@chelmuth You're right. Thanks for the correction.

chelmuth, 2018-01-03:
Is this issue still not fixed?

skalk, 2018-01-08:
It is partly fixed. Page-faults in core are detected, but kernel page-faults still lead to a dead-lock (lock taken twice). I feel no urgent pressure to solve this issue. From my point of view it can be closed.

closed 2018-01-09 by skalk