genodelabs/genode #1544: Planning of Hack'n'Hike 2016

m-stein, 2015-05-26:
Like in 2015, we're planning to celebrate a Hack'n'Hike gathering again in May 2016. If you're interested in Genode and the community behind it, you're welcome to join us. The idea behind the Hack'n'Hike is to meet in a cozy accommodation in the country side, experience activities in nature during the day and encourage programming projects at the evening/night.

This issue shall serve as platform for the collaborative planning of the Hack'n'Hike 2016 and as information source for all participants. For private requests regarding the event, please send an email to hacknhike@genode-labs.com. If you're planning to join us, please let us know as soon as possible to simplify the organization.

chelmuth, 2015-05-27:
Maybe we'll have a chance to book http://www.gutshaus-heeselicht.de/ this time?

m-stein, 2015-05-29:
I've contacted them yesterday.

m-stein, 2015-06-01:
Gutshaus Heeselicht has answered and they say that in May 2016, they have room for us expect at the weekends of Ascension Day and Pentecost. I liked the Monday-to-Friday constellation this year. So, I would propose 20th-23rd or 27th-30th of May for the next Hack'n'Hike. Any opinions on that?

The Gutshaus has room for 14 persons at a max. It provides a kitchen and dining room for self-catering only. I would make use of this offer and assign myself to the organization of breakfast and dinner if someone (or several people) would like to join me. Volunteers?

chelmuth, 2015-06-02:
:+1: from me for both dates. I'll gladly join you preparing meals - maybe @skalk could help us prepare the delicious pasta with egg plants we had in Saupsdorf 2012...

nfeske, 2015-06-02:
It would be interesting to know if there is the opportunity to camp, in the case the 14 beds are not enough.

m-stein, 2015-06-02:
@chelmuth 
Cool that you want to join in catering :-). @ssumpf had some cooking ideas too. A [bakery](https://www.google.de/maps/place/Frank+Tschulik+B%C3%A4ckerei/@51.0065178,14.0990097,15z/data=!4m2!3m1!1s0x4709b00fb94a3f61:0xfd5f689cafa09b29) is near the Gutshaus. That should simplify breakfast preparation.

@nfeske 
I had misunderstood something. They have room for up to 23 persons.

m-stein, 2015-06-08:
If there is no objection against it until Wednesday, I would start reservation of the accommodation for 27th-30th of May.

m-stein, 2015-06-12:
<b>Accommodation</b>

We recently have finished reservation of the next Hack'n'Hike accommodation.

<i>When</i>
From the 27th till the 30th of May

<i>Where</i>
Gutshaus Heeselicht ([website](http://www.gutshaus-heeselicht.de))
Germany, 01833 Stolpen, district Heeselicht,  Hofestr. 1a ([maps](https://www.google.de/maps/place/Gutshaus+Heeselicht/@51.006601,14.099013,17z/data=!3m1!4b1!4m2!3m1!1s0x4709b0047bc32663:0xe599b8be1a2da696?hl=en))

<i>What</i>
* Six rooms with a double bed each and one room with a double and a single bed
* If required, there is further capacity through the means of camping in the garden of the Gutshaus and some bunk beds
* Four baths and a separate lavatory
* A 60 m² sized seminar room, a cozy lounge and the Gutshaus park
* A self-catering kitchen and a dining area
* All exclusively used by us during the Hack'n'Hike

<i>Catering</i>
As already mentioned,

chelmuth, 2015-06-14:
@m-stein I'm a bit confused as I was notified of a comment by you (https://github.com/genodelabs/genode/issues/1544#issuecomment-111603204) Did you remove it later?

m-stein, 2015-06-15:
@chelmuth 
Yes. I was hitting the return key mistakingly. Sorry for the confusion :)

m-stein, 2015-06-15:
### Accommodation

We recently have finished reservation of the next Hack'n'Hike accommodation. If you like to join us, please **let us know till January 2016** to simplify organization. Following, some general information on it:

#### Time
* Check-in as of 3 pm on May 27th
* Check-out till 12 am on May 30th

#### Location
* Gutshaus Heeselicht ([website](http://www.gutshaus-heeselicht.de))
Germany, 01833 Stolpen, district Heeselicht,  Hofestr. 1a ([maps](https://www.google.de/maps/place/Gutshaus+Heeselicht/@51.006601,14.099013,17z/data=!3m1!4b1!4m2!3m1!1s0x4709b0047bc32663:0xe599b8be1a2da696?hl=en))

#### Room
* Six rooms with a double bed each and one room with a double and a single bed
* If required, there is further capacity through means of camping in the Gutshaus park and bunk beds
* Four baths and a separate lavatory
* A 60 m² sized lockable seminar room, a cozy lounge and the Gutshaus park
* A self-catering kitchen and a dining area
* A fireplace and a grill area in the Gutshaus park
* All exclusively used by us during the Hack'n'Hike
* The Gutshaus does **not** provide an internet connection

#### Catering
* Christian and I will do the catering for breakfast and dinner in the Gutshaus
* Normally, we stop by a restaurant or tavern for lunch during the day trips
* If you have special requirements regarding the food (vegetarian, vegan, gluten-free, lactose-free, ...), **please let us know as soon as possible**

m-stein, 2016-01-11:
The Gutshaus Heeselicht now needs a first estimation of the amount of participants. If you like to join the Hack'n'Hike 2016, this would be a good time to let us know. However, if you don't know by now whether you can participate, you may join us later as well.

nfeske, 2016-01-11:
I'm in! :-)

ehmry, 2016-01-11:
I can make it.

chelmuth, 2016-01-11:
Count on me.

vladimir-lu, 2016-01-11:
I can see why you picked the location, looks very nice!

I can make it also.

parfito, 2016-01-18:
Hello, I'm new here, but I want to join too.

ponast, 2016-01-19:
For estimation purposes: count on 50% of me. Will know for sure at FOSDEM.

trimpim, 2016-01-20:
I will also come.

m-stein, 2016-01-20:
@ponast If this is possible for you, it would be good to have your decision till January 26th, i.e. before the FOSDEM, because this simplifies organization.

m-stein, 2016-01-22:
I attached a list of the fastest ways to get from Dresden main station to Heeselicht at May 27th using public transport. At least one of the Genode Labs crew will travel using one of the attached plans, so others can join him. If you like to do so, please let us know in advance and tell us when and where you will arrive in Dresden. If you have further questions regarding this topic, don't hesitate to post it here or send it to hacknhike@genode-labs.com .

```
Plan 1
------
From Dresden central station Platform 18 13:59 train "S1" direction "Bad Schandau"
To   Pirna   central station Platform 3  14:20
From Pirna   central bus station         14:25 bus 237 direction "Sebnitz Busbahnhof"
To   Heeselicht Markt (Stolpen)          14:57

Plan 2
------
From Dresden central station Platform 18 14:29 train "S1" direction "Schöna"
To   Pirna   central station Platform 3  14:50
From Pirna   central bus station         14:55 bus 236 direction "Sebnitz Busbahnhof"
To   Heeselicht Markt (Stolpen)          15:35

Plan 1 + 1h (From Dresden central station at 14:59)
Plan 2 + 1h (From Dresden central station at 15:29)

Plan 3
------
From Dresden central station Platform 18 16:11 train "S2" direction "Pirna"
To   Pirna   central station Platform 2  16:33
From Pirna   central bus station         16:40 bus 236 direction "Sebnitz Busbahnhof"
To   Heeselicht Markt (Stolpen)          17:18

Plan 4
------
From Dresden central station Platform 18 18:11 train "S2" direction "Pirna"
To   Pirna   central station Platform 2  18:33
From Pirna   central bus station         18:40 bus 237 direction "Sebnitz Busbahnhof"
To   Heeselicht Markt (Stolpen)          19:09
```

m-stein, 2016-01-25:
A short reminder: Tomorrow I'll have to report the number of participants to the Gutshaus Heeselicht. So, if you like to join us, post a short hint in here or send it to hacknhike@genode-labs.com. If you, however, are not able to determine till tomorrow whether you can make it, there are also ways to join us later.

ponast, 2016-01-25:
Hello Martin,

I have all the permits in order now! I'll sign up!

Pontus

m-stein, 2016-01-25:
Thank you Pontus :)

parfito, 2016-01-26:
Hello Martin, Sorry for my late reply, the Plan 1 is good for me. I recall that we are two (me and my collegue Emery).

ehmry, 2016-01-26:
A second Emery?!

parfito, 2016-01-27:
Yeah, an other Emery. So that makes 2.

m-stein, 2016-01-31:
I'm happy to announce that we have successfully booked the Gutshaus Heeselicht for 27 May, 3pm till 30 May, 12am. In addition to the [information that you can find further up](https://github.com/genodelabs/genode/issues/1544#issuecomment-111988235) in the discussion, please consider also the following points if you'll stay with us in the Gutshaus:

* Please **bring your own towel** (and Hitchhiker’s Guide to the Galaxy)
* If you like, you can also bring a tent and camp in the park of the Gutshaus but please let us know in advance if you're planning to do so
* Unfortunately, pets are not allowed in the Gutshaus, so please no dogs, cats or naked mole-rats
* The reservation includes the bedding, so you don't have to bring your own
* If you're planning to join the outdoor activities I recommend you to bring some clothes that can get dirty, hardwearing shoes, and, just in case, rainwear

gwitmond, 2016-02-03:
If there is still a place available, I'd like to join too.

m-stein, 2016-02-04:
@gwitmond Yes there is. Nice to have you in again!

m-stein, 2016-02-10:
I have a suggestion for an outdoor activity:

If we drive to Wehlen (20 minutes) we can do the [circular walk "Höllengrund"](http://www.wandern-saechsische-schweiz.de/Rathen_Hoellengrund.asp). It's 11,5 kilometers long and should take us about 3 hours. It goes through the Uttewalder and Wehlener coulee to the Felsentor (a rock gate), through the coulees Kohlgrund and Reingrund to the [Bastei](https://en.wikipedia.org/wiki/Bastei), then to the inn "Steinerner Tisch", and through the coulees Höllengrund and Zscherregrund back to the castle of Wehlen.

Any opinions on that? Is there someone who thinks this walk is too hard? Is there someone who doesn't like the POIs of the walk? Please let me know ;-)

chelmuth, 2016-02-22:
:+1: for the hiking tour!

trimpim, 2016-02-23:
:+1: for the "Höllengrund" tour

I'll bring my RPI with me, so I can work on the I2C and SPI drivers.

If there would be other people interested in porting Genode to the RPI2, I could bring this with me too.

ehmry, 2016-02-23:
Hiking is good.

ponast, 2016-04-14:
Limo service is provided for free from Berlin Tegel for anybody arriving there on May 27th at 16:00. Return trip back to Tegel circa 09:00 Monday.

vladimir-lu, 2016-05-11:
Unfortunately I won't be able to make this :disappointed:

m-stein, 2016-05-11:
@vladimir-lu Pity! Hopefully for enjoyable reasons.

m-stein, 2016-05-18:
I've just received a note from the Gutshaus Heeselicht that everyone should bring his/her __own slippers__! As mentioned earlier, please also bring your __own towel__ and the other __stuff listed [here](https://github.com/genodelabs/genode/issues/1544#issuecomment-177642147)__. Happy packing ;)

m-stein, 2016-05-18:
I've summarized the plan so far for the Hack'n'Hike. Any questions or suggestions? (Please view it on github to get a pretty table)

 | Fr | Sa | Su | Mo
---|---|---|---|---
Breakfast |                  | Genode Labs Catering in our [accomodation](http://www.gutshaus-heeselicht.de/)         | Genode Labs Catering in our [accomodation](http://www.gutshaus-heeselicht.de/) | Genode Labs Catering in our [accomodation](http://www.gutshaus-heeselicht.de/)
Activity 1   | | Hiking from Heeselicht to Stolpen | [Hiking from Wehlen to the Bastei](https://github.com/genodelabs/genode/issues/1544#issuecomment-182541847) | Check-out till 12 am  
Lunch     |              | [Inn "Goldner Löwe" in Stolpen](http://www.loewe-stolpen.de/) | [Inn at the Bastei](http://www.bastei-berghotel.de/gastronomie/restaurants/panoramarestaurant.html)    |
Activity 2 | Check-in as of 3 pm  | Visit Stolpen [Castle](http://www.burg-stolpen.org/en/homepage/) | [Hiking from the Bastei back to Wehlen](https://github.com/genodelabs/genode/issues/1544#issuecomment-182541847) |
Dinner    | [Inn "Erbgericht" in Heeselicht](http://www.erbgericht.de/) | Genode Labs Barbecue in our [accomodation](http://www.gutshaus-heeselicht.de/)         | [Inn "Erbgericht" in Heeselicht](http://www.erbgericht.de/)     |
Evening + Night | Hacking | Hacking | Hacking |

m-stein, 2016-05-18:
@trimpim Could you please bring your RPI2? I would be interested in porting.

ehmry, 2016-05-18:
@ponast Is that a ride from Berlin all the way to Saxony?

ponast, 2016-05-18:
@emery The ride is from Berlin Tegel  to wherever you want to go. Most
likely Inn Erbgericht in Heeslicht. Want to join?

ehmry, 2016-05-18:
@ponast Yes please, I'll email you next week for the details.

trimpim, 2016-05-19:
Yes, I will bring it, together with all documentation I'll be able to collect about it.

m-stein, 2016-05-19:
@trimpim Very cool :)

gwitmond, 2016-05-19:
Hi Pontus,

Does the limo have a third sofa? I'd like to hitch a ride too.

Cheers, Guido.

gwitmond, 2016-05-20:
I've cleaned up the house and found some old flight tickets in the attic. I got myself a limo that fits the hand luggage so feel free to hop on at Berlin Schönefeld around 9:00 at 27th.

ponast, 2016-05-22:
@gwitmond: Sorry for the late response, on vacation over the weekend on the
countryside.

Pontus

ponast, 2016-05-24:
@ehmry: I have actually lost your email from today regarding the meeting
coordinates so I want you to resend it.

trimpim, 2016-05-25:
@chelmuth my flight will arrive at Dresden Airport around 09:00. Is it okay if I come to the Genode office?

chelmuth, 2016-05-25:
@trimpim That's perfectly okay, but be aware that we'll be busy packing all the needed stuff and appreciate any helping hand ;-) There's no fixed time yet but I expect we'll not leave before 13:00.

alex-ab, 2016-05-26:
Currently it looks like that nobody of the Genode will take the public transport options as advertised by @m-stein. @parfito and and Emery: if you want to join us and get a ride by car please come to the Genode office (till approx. 13:30).

ehmry, 2016-05-26:
I'm getting a ride from @ponast, so no problem.

parfito, 2016-05-26:
hello, I want to apologize on behalf of my colleague Emery Assogba who is unable to come at the last minute. So, unfortunately I'll come alone.
Really sorry!

m-stein, 2016-05-31:
Hey guys, all got home in one piece?
I'd like to thank you for coming and hope you've enjoyed this years Hack'n'Hike as much as me?
I just started planning for the Hack'n'Hike 2017 (#1985), so have a look ;-)

parfito, 2016-05-31:
I am back in Brussels safely...
It was very nice, eagerly waiting for the next.

chelmuth, 2016-06-16:
So it seems I have the pleasure to close this issue. I had a great time in Heeselicht and am looking forward to #1985.

closed 2016-06-16 by chelmuth