genodelabs/genode #1547: AHCI fails on (some) real hardware

NobodyIII, 2015-05-29:
This issue seems to be related to IRQs.

chelmuth, 2015-05-30:
[Hardware specs for the machine](https://github.com/NobodyIII/genode/blob/master/lshw.log)

@NobodyIII please add a clean and complete log file too. The issue description does not tell anything about your issue currently.

NobodyIII, 2015-05-31:
What do you mean by "clean"? And by complete, do you mean that you want me
to set verbose=true on ahci and/or acpi, or do you need more than that?
Also, I've been wading through the ahci, and pci code to try to find a good
place to hardcode IRQ 19, but I haven't found one yet. Given a few hours, I
could probably find a place to hardcode IRQ 19, but I'm betting you know
the code well enough that you could help me find one much faster without
you spending much time wading through code. Could you please point me in
the right direction?

chelmuth, 2015-06-01:
The [log](https://gist.github.com/chelmuth/598916fcb30a4c131062#file-nobody-unclean-log) you sent to on the list is not clean in the way that several lines of log are overwritten by later output. I'd like to read the complete log output where each input line also takes one output line.

ssumpf, 2015-06-01:
@NobodyIII: Could you send us a dump of the ACPI DSDT table of the machine? If you don't want to see the dump online a private email does suffice.

alex-ab, 2015-06-01:
> Also, I've been wading through the ahci, and pci code to try to find a good place to hardcode IRQ 19, but I haven't found one yet. Given a few hours, I could probably find a place to hardcode IRQ [...]

@NobodyIII:  try  commit adf784fa07f3af65821255c3f49f72d1e26e7217.

NobodyIII, 2015-06-01:
I used minicom. Should I use something else?

NobodyIII, 2015-06-01:
The commit didn't work. It did give the debug output about overwriting IRQ
11 with 19 though.

chelmuth, 2015-06-01:
Whatever you like as long as the output stays intact. AFAIK minicom uses Ctrl-l to define a log file to write to.

NobodyIII, 2015-06-01:
Thanks. I just copied the terminal output.

ssumpf, 2015-06-04:
@NobodyIII: Can you please try out the following branch (https://github.com/ssumpf/genode/tree/issue_1547_dsdt) and send the log output. Thx.

NobodyIII, 2015-06-04:
Here's the output:

Bender: Hello World.
Need 07068000 bytes to relocate modules.
Relocating to 78f98000:
Copying 6053840 bytes...
Copying 10485760 bytes...
Copying 7559296 bytes...
Copying 14463248 bytes...
Copying 7114256 bytes...
Copying 8398360 bytes...
Copying 6123064 bytes...
Copying 6853608 bytes...
Copying 1298720 bytes...
Copying 6721728 bytes...
Copying 6645048 bytes...
Copying 6341296 bytes...
Copying 6395352 bytes...
Copying 6024024 bytes...
Copying 6637760 bytes...
Copying 1870 bytes...
Copying 8878520 bytes...
Copying 404264 bytes...
Copying 695472 bytes...
Copying 728940 bytes...

L4 Bootstrapper
  Build: #1 Wed May 27 02:30:51 UTC 2015, x86-64, 4.9.2
  RAM: 0000000000000000 - 000000000009e7ff: 634kB
  RAM: 0000000000100000 - 00000000be10ffff: 3113024kB
  RAM: 00000000bee5d000 - 00000000bee5dfff: 4kB
  RAM: 00000000bf064000 - 00000000bf462fff: 4092kB
  RAM: 00000000bf7f3000 - 00000000bf7fffff: 52kB
  RAM: 0000000100001000 - 000000043effffff: 13615100kB
  Total RAM: 16340MB
  Moving up to 19 modules behind 1100000
  moving module 00 { 7904a000-790f3caf } -> { 80000000-800a9caf } [695472]
  moving module 01 { 790f4000-79156b27 } -> { 800aa000-8010cb27 } [404264]
  moving module 02 { 79157000-799ce9b7 } -> { 8010d000-809849b7 } [8878520]
  moving module 03 { 799cf000-799cf74d } -> { 1100000-110074d } [1870]
  moving module 04 { 799d0000-7a0248bf } -> { 1101000-17558bf } [6637760]
  moving module 05 { 7a025000-7a5e3b57 } -> { 1756000-1d14b57 } [6024024]
  moving module 06 { 7a5e4000-7abfd5d7 } -> { 1d15000-232e5d7 } [6395352]
  moving module 07 { 7abfe000-7b20a2af } -> { 232f000-293b2af } [6341296]
  moving module 08 { 7b20b000-7b861537 } -> { 293c000-2f92537 } [6645048]
  moving module 09 { 7b862000-7becb0bf } -> { 2f93000-35fc0bf } [6721728]
  moving module 10 { 7becc000-7c00911f } -> { 35fd000-373a11f } [1298720]
  moving module 11 { 7c00a000-7c6933e7 } -> { 373b000-3dc43e7 } [6853608]
  moving module 12 { 7c694000-7cc6ae37 } -> { 3dc5000-439be37 } [6123064]
  moving module 13 { 7cc6b000-7d46d617 } -> { 439c000-4b9e617 } [8398360]
  moving module 14 { 7d46e000-7db36e0f } -> { 4b9f000-5267e0f } [7114256]
  moving module 15 { 7db37000-7e90210f } -> { 5268000-603310f } [14463248]
  moving module 16 { 7e903000-7f03887f } -> { 6034000-676987f } [7559296]
  moving module 17 { 7f039000-7fa38fff } -> { 676a000-7169fff } [10485760]
  moving module 18 { 7fa3a000-7fffffcf } -> { 716a000-772ffcf } [6053840]
  Scanning /fiasco/fiasco -serial_esc
  Scanning /fiasco/sigma0
  Scanning /stage1/core
  Bootloader MMAP:
    [        0,     9e800) RAM (1)
    [    9e800,     a0000) reserved (2)
    [    e0000,    100000) reserved (2)
    [   100000,  be110000) RAM (1)
    [ be110000,  be4e0000) reserved (2)
    [ be4e0000,  be8cf000) ACPI NVS (4)
    [ be8cf000,  bee5d000) reserved (2)
    [ bee5d000,  bee5e000) RAM (1)
    [ bee5e000,  bf064000) ACPI NVS (4)
    [ bf064000,  bf463000) RAM (1)
    [ bf463000,  bf7f3000) reserved (2)
[... 562 lines cut ...]
[init -> acpi] Rewriting 01:00.1 IRQ: 11 -> GSI: 25
[init -> acpi] void dump_bdf(Genode::uint32_t, Genode::uint32_t,
Genode::uint32_t): Device bdf 02:00.0 (200) bridge 00:09.0 (48) Pin: 0
[init -> acpi] Rewriting 02:00.0 IRQ: 10 -> GSI: 48
[init -> acpi] void dump_bdf(Genode::uint32_t, Genode::uint32_t,
Genode::uint32_t): Device bdf 03:05.0 (328) bridge 00:14.4 (a4) Pin: 0
[init -> acpi] Rewriting 03:05.0 IRQ: 10 -> GSI: 20
[init -> acpi] void dump_bdf(Genode::uint32_t, Genode::uint32_t,
Genode::uint32_t): Device bdf 05:00.0 (500) bridge 00:15.3 (ab) Pin: 0
[init -> acpi] Rewriting 05:00.0 IRQ: 11 -> GSI: 19
[init] child "acpi" announces service "PCI"
[init -> ahci_drv] AHCI found (vendor: 1002 device: 4391 class: 00010601)
[init] child "acpi" announces service "IRQ"
[init -> ahci_drv] flag 1
[init -> acpi -> pci_drv] 0:11.0 uses IRQ, vector 0xb
[init -> ahci_drv] flag 2
[init -> ahci_drv]     version: 1.0200
[init -> ahci_drv]     command slots: 32
[init -> ahci_drv]     native command queuing: yes
[init -> ahci_drv]     64 bit support: yes
[init -> ahci_drv]     number of ports: 4 pi: f
[init -> ahci_drv]         #0: ATA
[init -> ahci_drv]         #1: ATA
[init -> ahci_drv]         #2: ATAPI
[init -> ahci_drv]         #3: off

ssumpf, 2015-06-05:
Update: The ACPI driver does not resolve any host-bridge GSIs. This is caused by a strange _ADR entry in SB0.PCI0. The entry shows 0x00180000 for the _ADR which our driver resolves to 00:c0.0. The real BDF (00:00.0) can be found in the first 'Device' entry of SB0.PCI0 that describes the bridge. Since on all previous platforms SB0.PCI0._ADR has been 0x0 and because there is nothing much to be found on the super-information highway (www), I've reached out to vmmon.

ssumpf, 2015-06-09:
According to vmmon we have to read the bus number from the 00:18.0 PCI config space, this will tell us which bus the interrupt routings are meant for. Fix is in the works.
@NobodyIII: Right now we observe some issues with interrupts on Fiasco.OC in general, which means they do not work on some platforms. Since @alex-ab told me that the NOVA kernel does not come up on your box, we will have to fix the Fiasco thing additionally.

alex-ab, 2015-06-09:
@NobodyIII :  please try the following branch "https://github.com/alex-ab/genode/tree/issue_1547" and send the log output.

NobodyIII, 2015-06-09:
Actually, the NOVA kernel does work—it just looked like it didn't since it
didn't output much to the screen. Now that I've got a working serial
connection, I can tell that it actually works. However, I get an
error/warning from device_pd. I mentioned it earlier on the non-github
thread related to the AHCI issue.

alex-ab, 2015-06-09:
Cool, please send nevertheless the serial log from the nova run. Based on the output I will rework the patch and make it ready for staging.

NobodyIII, 2015-06-09:
Here's the log output:

NOVA Microhypervisor v7-9916e62 (x86_64): May 27 2015 01:04:57 [gcc 4.9.2]

[ 0] CORE:0:0:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 1] CORE:0:1:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 4] CORE:0:4:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 5] CORE:0:5:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 3] CORE:0:3:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 6] CORE:0:6:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 2] CORE:0:2:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 7] CORE:0:7:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
Hypervisor does not feature VMX
Hypervisor features SVM
Hypervisor reports 8x1 CPUs - boot CPU is 0
Hypervisor info page contains 42 memory descriptors:
detected physical memory: 0x0000000000000000 - size: 0x9e800
use      physical memory: 0x0000000000000000 - size: 0x9e000
detected physical memory: 0x0000000000100000 - size: 0xbe010000
use      physical memory: 0x0000000000100000 - size: 0xbe010000
detected physical memory: 0x00000000bee5d000 - size: 0x1000
use      physical memory: 0x00000000bee5d000 - size: 0x1000
detected physical memory: 0x00000000bf064000 - size: 0x3ff000
use      physical memory: 0x00000000bf064000 - size: 0x3ff000
detected physical memory: 0x00000000bf7f3000 - size: 0xd000
use      physical memory: 0x00000000bf7f3000 - size: 0xd000
detected physical memory: 0x0000000100001000 - size: 0x33efff000
use      physical memory: 0x0000000100001000 - size: 0x33efff000
map multi-boot module: physical 0x76a68000 -> [0x001c2000-0x00b06340) - core
map multi-boot module: physical 0x773ad000 -> [0x00002000-0x000027cc) -
config
map multi-boot module: physical 0x773ae000 -> [0xb0000000-0xb072ef90) - init
map multi-boot module: physical 0x77add000 -> [0x00b0a000-0x011a2500) -
timer
map multi-boot module: physical 0x78176000 -> [0x011a4000-0x01819a38) -
log_report
map multi-boot module: physical 0x787ec000 -> [0x0181b000-0x01f25248) -
rom_blk
map multi-boot module: physical 0x78ef7000 -> [0x01f27000-0x025fede0) -
part_blk
map multi-boot module: physical 0x795cf000 -> [0x02600000-0x02c97250) -
rump.lib.so
map multi-boot module: physical 0x79c67000 -> [0x02c99000-0x033020d0) -
rump_fs.lib.so
map multi-boot module: physical 0x7a2d1000 -> [0x03304000-0x03486af0) -
rump_fs
map multi-boot module: physical 0x7a454000 -> [0x03488000-0x03c32248) -
ld.lib.so
map multi-boot module: physical 0x7abff000 -> [0x03c34000-0x042db628) -
fs_rom
map multi-boot module: physical 0x7b2a7000 -> [0x042dd000-0x04bf6618) -
pci_drv
map multi-boot module: physical 0x7bbc1000 -> [0x04bf8000-0x05235778) -
pci_device_pd
map multi-boot module: physical 0x7c1ff000 -> [0x05237000-0x05a76808) -
fb_drv
map multi-boot module: physical 0x7ca3f000 -> [0x05a78000-0x061619b0) -
ps2_drv
map multi-boot module: physical 0x7d129000 -> [0x06163000-0x067bda00) -
acpi_drv
map multi-boot module: physical 0x7d784000 -> [0x067bf000-0x077741f8) -
usb_drv
map multi-boot module: physical 0x7e73a000 -> [0x07776000-0x07fbd0b8) -
ahci_drv
map multi-boot module: physical 0x7ef82000 -> [0x07fbf000-0x089bf000) -
ata.raw
map multi-boot module: physical 0x7f983000 -> [0x089bf000-0x0903b048) -
test-blk-cli-nowrite
:virt_alloc: Allocator 17ce90 dump:
 Block: [00002000,00004000) size=00002000 avail=00000000 max_avail=00000000
[... 392 lines cut ...]
[init -> log_report]     <routing gsi="0x34" bridge_bdf="0xc0" device="0x5"
device_pin="0x0"/>
[init -> log_report]     <routing gsi="0x34" bridge_bdf="0xc0" device="0x4"
device_pin="0x3"/>
[init -> log_report]     <routing gsi="0x34" bridge_bdf="0xc0" device="0x4"
device_pin="0x2"/>
[init -> log_report]     <routing gsi="0x34" bridge_bdf="0xc0" device="0x4"
device_pin="0x1"/>
[init -> log_report]     <routing gsi="0x34" bridge_bdf="0xc0" device="0x4"
device_pin="0x0"/>
[init -> log_report]     <routing gsi="0x34" bridge_bdf="0xc0" device="0x2"
device_pin="0x3"/>
[init -> log_report]     <routing gsi="0x34" bridge_bdf="0xc0" device="0x2"
device_pin="0x2"/>
[init -> log_report]     <routing gsi="0x34" bridge_bdf="0xc0" device="0x2"
device_pin="0x1"/>
[init -> log_report]     <routing gsi="0x34" bridge_bdf="0xc0" device="0x2"
device_pin="0x0"/>
[init -> log_report]     <routing gsi="0x37" bridge_bdf="0xc0" device="0x0"
device_pin="0x0"/>
[init -> log_report]     <routing gsi="0x13" bridge_bdf="0xc0"
device="0x11" device_pin="0x0"/>
[init -> log_report] </acpi>
[init -> log_report] end of report
int main(): --- init created, waiting for exit condition ---

alex-ab, 2015-06-09:
The relevant information of the pci_drv is unfortunately missing. Sorry, I forgot that you have to setup a separate report_rom server now. Please have a look for an example in repos/libports/run/libc_ffat.run and adjust your run script. The generated config of libc_ffat.run looks like this for base-nova:

```
<config verbose="yes">
	<parent-provides>
		<service name="ROM"/>
		<service name="RAM"/>
		<service name="IRQ"/>
		<service name="IO_MEM"/>
		<service name="IO_PORT"/>
		<service name="CAP"/>
		<service name="PD"/>
		<service name="RM"/>
		<service name="CPU"/>
		<service name="LOG"/>
		<service name="SIGNAL"/>
	</parent-provides>
	<default-route>
		<any-service> <parent/> <any-child/> </any-service>
	</default-route>
	<start name="timer">
		<resource name="RAM" quantum="1M"/>
		<provides> <service name="Timer"/> </provides>
	</start>
	<start name="test-libc_ffat">
		<resource name="RAM" quantum="8M"/>
		<config>
			<iterations value="1"/>
			<libc stdout="/dev/log" stderr="/dev/log">
				<vfs>
					<dir name="dev">
						<log/>
					</dir>
				</vfs>
			</libc>
		</config>
	</start>
	<start name="acpi_drv" >
		<resource name="RAM" quantum="2M"/>
		<route>
			<service name="LOG"> <parent/> </service>
			<service name="IO_MEM"> <parent/> </service>
			<service name="RM"> <parent/> </service>
			<service name="Report"> <child name="acpi_report_rom"/> </service>
		</route>
	</start>
	<start name="acpi_report_rom" >
		<binary name="report_rom"/>
		<resource name="RAM" quantum="2M"/>
		<provides>
			<service name="ROM" />
			<service name="Report" />
		</provides>
		<config>
			<rom> <policy label="pci_drv -> acpi" report="acpi_drv -> acpi"/> </rom>
		</config>
		<route>
			<service name="LOG"> <parent/> </service>
			<service name="SIGNAL"> <parent/> </service>
			<service name="CAP"> <parent/> </service>
			<service name="RM"> <parent/> </service>
		</route>
	</start>
	<start name="pci_drv" >
		<resource name="RAM" quantum="3M" constrain_phys="yes"/>
		<provides> <service name="PCI"/> </provides>
		<route>
			<service name="ROM">
				<if-arg key="label" value="acpi"/> <child name="acpi_report_rom"/>
			</service>
			<any-service> <parent/> </any-service>
		</route>
		<config>
			<policy label="ps2_drv">  <device name="PS2"/> </policy>
			<policy label="nic_drv">  <pci class="ETHERNET"/> </policy>
			<policy label="fb_drv">   <pci class="VGA"/> </policy>
			<policy label="wifi_drv"> <pci class="WIFI"/> </policy>
			<policy label="usb_drv">  <pci class="USB"/> </policy>
			<policy label="ahci_drv"> <pci class="AHCI"/> </policy>
		</config>
	</start>
	<start name="ahci_drv">
		<resource name="RAM" quantum="10M" />
		<provides><service name="Block" /></provides>
		<route>
			<any-service> <parent /> <any-child /></any-service>
		</route>
		<config>
			<policy label="test-libc_ffat" device="0" />
		</config>
	</start>
</config>

```

NobodyIII, 2015-06-09:
I'm a bit confused as to what's going on. Why are both acpi_drv and pci_drv
started separately? I though acpi_drv was supposed to start pci_drv. Please
explain well enough what's going on so I can adapt the config to my needs.

ssumpf, 2015-06-09:
@NobodyIII: Because your box seems to trigger several bugs or missing features. One can be found in our ACPI-DSDT parsing heuristics  00:11.0 ->  GSI 19 is not found. The other thing is that @alex-ab rewrote most of the pci_drv which will be called "platform driver" in the future. One of the changes is that one has to route PCI devices explicitly (see config above). By now the PCI driver handles all interrupt related things on x86. Hence the mix up. So please try out the config above and thanks for testing!

nfeske, 2015-06-09:
For reference, the current changes regarding the roles of pci_drv, acpi_drv, and platform_drv are discussed at issue #1542.

alex-ab, 2015-06-09:
> I'm a bit confused as to what's going on. Why are both acpi_drv and pci_drv started separately? I though acpi_drv was supposed to start pci_drv. Please explain well enough what's going on so I can adapt the config to my needs.

We are de-coupling the apci and pci driver currently. The acpi driver does not start anymore the pci driver. The acpi parser just parses the tables and generates a report. The report service stores this information. The pci driver asks for this report, and uses this information to rewrite IRQs and other stuff. So, you will need now ever 3 components for foc and nova, acpi_drv, pci_drv and a report_rom service.

ssumpf, 2015-06-09:
@alex-ab  and @nfeske:  Thanks for your help!

NobodyIII, 2015-06-09:
Here's the new log:

Bender: Hello World.
Need 09d9a000 bytes to relocate modules.
Relocating to 76266000:
Copying 6799432 bytes...
Copying 10485760 bytes...
Copying 8679608 bytes...
Copying 16470520 bytes...
Copying 6662656 bytes...
Copying 7248304 bytes...
Copying 8648712 bytes...
Copying 6543224 bytes...
Copying 9541144 bytes...
Copying 6977064 bytes...
Copying 8036936 bytes...
Copying 1583856 bytes...
Copying 6721744 bytes...
Copying 6910544 bytes...
Copying 7175648 bytes...
Copying 7381576 bytes...
Copying 7258376 bytes...
Copying 6773304 bytes...
Copying 6915328 bytes...
Copying 7532432 bytes...
Copying 2709 bytes...
Copying 9716544 bytes...
Copying 1132544 bytes...

NOVA Microhypervisor v7-9916e62 (x86_64): May 27 2015 01:04:57 [gcc 4.9.2]

[ 0] CORE:0:0:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 4] CORE:0:4:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 5] CORE:0:5:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 7] CORE:0:7:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 6] CORE:0:6:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 2] CORE:0:2:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 1] CORE:0:1:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 3] CORE:0:3:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
Hypervisor does not feature VMX
Hypervisor features SVM
Hypervisor reports 8x1 CPUs - boot CPU is 0
Hypervisor info page contains 43 memory descriptors:
detected physical memory: 0x0000000000000000 - size: 0x9e800
use      physical memory: 0x0000000000000000 - size: 0x9e000
detected physical memory: 0x0000000000100000 - size: 0xbe010000
use      physical memory: 0x0000000000100000 - size: 0xbe010000
detected physical memory: 0x00000000bee5d000 - size: 0x1000
use      physical memory: 0x00000000bee5d000 - size: 0x1000
detected physical memory: 0x00000000bf064000 - size: 0x3ff000
use      physical memory: 0x00000000bf064000 - size: 0x3ff000
detected physical memory: 0x00000000bf7f3000 - size: 0xd000
use      physical memory: 0x00000000bf7f3000 - size: 0xd000
detected physical memory: 0x0000000100001000 - size: 0x33efff000
use      physical memory: 0x0000000100001000 - size: 0x33efff000
map multi-boot module: physical 0x7637b000 -> [0x001c2000-0x00b06340) - core
map multi-boot module: physical 0x76cc0000 -> [0x00002000-0x00002a95) -
config
map multi-boot module: physical 0x76cc1000 -> [0xb0000000-0xb072ef90) - init
map multi-boot module: physical 0x773f0000 -> [0x00b0a000-0x011a2500) -
timer
map multi-boot module: physical 0x77a89000 -> [0x011a4000-0x01819a38) -
log_report
map multi-boot module: physical 0x780ff000 -> [0x0181b000-0x01f07108) -
report_rom
map multi-boot module: physical 0x787ec000 -> [0x01f09000-0x02613248) -
rom_blk
map multi-boot module: physical 0x78ef7000 -> [0x02615000-0x02cecde0) -
part_blk
map multi-boot module: physical 0x795cf000 -> [0x02cee000-0x03385250) -
[... 490 lines cut ...]
bus number 0 - is pci bridge 0 - bridge bdf 0 classcode=60000
[init -> pci_drv] device 0 device_pin=0 gsi 55 bridge bdf 0:18.0 (c0) ->
bus number 0 - is pci bridge 0 - bridge bdf 0 classcode=60000
[init -> pci_drv] device 11 device_pin=0 gsi 19 bridge bdf 0:18.0 (c0) ->
bus number 0 - is pci bridge 0 - bridge bdf 0 classcode=60000
[init] child "pci_drv" announces service "PCI"
[init -> pci_drv] static short unsigned int
Pci::Irq_routing::rewrite(unsigned char, unsigned char, unsigned char,
unsigned char): rewrite 0:11.0 bridge bdf 0
[init -> pci_drv] 0:11.0 rewriting IRQ: 11 -> 19
[init -> ahci_drv] AHCI found (vendor: 1002 device: 4391 class: 00010601)
[init -> pci_drv -> pci_device_pd] assignment of PCI device failed
[init -> ahci_drv] flag 1
[init -> pci_drv] 0:11.0 uses IRQ, vector 0x13
[init -> ahci_drv] flag 2
[init -> ahci_drv]     version: 1.0200
[init -> ahci_drv]     command slots: 32
[init -> ahci_drv]     native command queuing: yes
[init -> ahci_drv]     64 bit support: yes
[init -> ahci_drv]     number of ports: 4 pi: f
[init -> ahci_drv]         #0: ATA
int main(): --- init created, waiting for exit condition ---
[init -> ahci_drv]         #1: ATA
[init -> ahci_drv]         #2: ATAPI
[init -> ahci_drv]         #3: off

NobodyIII, 2015-06-09:
And by the way, I have an Asrock 970 Extreme 2.0 motherboard if that help
at all.

NobodyIII, 2015-06-09:
edit: helps, not help.

alex-ab, 2015-06-17:
@NobodyIII: Does it mean that the ahci driver on nova works with the irq re-routing patch finally ?

NobodyIII, 2015-06-17:
No. My test still isn't starting, indicating that it cannot get a block
connection.

alex-ab, 2015-06-17:
What is with the other drivers on your machine, nic_drv (lwip.run script), usb_drv (usb_hid.run), ps2_drv (input.run) (if you have ps2 ...) ? Do they work?

NobodyIII, 2015-06-17:
I haven't tested nic_drv, but usb_drv doesn't work with either my mouse or
keyboard. I removed my wireless adapter to see whether that would help, but
it still didn't show any signs of recieving input when I moved the mouse,
clicked, and typed. Here's the output:

NOVA Microhypervisor v7-9916e62 (x86_64): May 27 2015 01:04:57 [gcc 4.9.2]

[ 0] CORE:0:0:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 6] CORE:0:6:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 7] CORE:0:7:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 3] CORE:0:3:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 1] CORE:0:1:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 2] CORE:0:2:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 5] CORE:0:5:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
[ 4] CORE:0:4:0 15:2:0:0 [0] AMD FX(tm)-8320 Eight-Core Processor
Hypervisor does not feature VMX
Hypervisor features SVM
Hypervisor reports 8x1 CPUs - boot CPU is 0
Hypervisor info page contains 44 memory descriptors:
detected physical memory: 0x0000000000000000 - size: 0x9e800
use      physical memory: 0x0000000000000000 - size: 0x9e000
detected physical memory: 0x0000000000100000 - size: 0xbe010000
use      physical memory: 0x0000000000100000 - size: 0xbe010000
detected physical memory: 0x00000000bee5d000 - size: 0x1000
use      physical memory: 0x00000000bee5d000 - size: 0x1000
detected physical memory: 0x00000000bf064000 - size: 0x3ff000
use      physical memory: 0x00000000bf064000 - size: 0x3ff000
detected physical memory: 0x00000000bf7f3000 - size: 0xd000
use      physical memory: 0x00000000bf7f3000 - size: 0xd000
detected physical memory: 0x0000000100001000 - size: 0x33efff000
use      physical memory: 0x0000000100001000 - size: 0x33efff000
map multi-boot module: physical 0x76329000 -> [0x001c2000-0x00b06340) - core
map multi-boot module: physical 0x76c6e000 -> [0x00002000-0x000028f0) -
config
map multi-boot module: physical 0x76c6f000 -> [0xb0000000-0xb072ef90) - init
map multi-boot module: physical 0x7739e000 -> [0x00b0a000-0x011a2500) -
timer
map multi-boot module: physical 0x77a37000 -> [0x011a4000-0x01819a38) -
log_report
map multi-boot module: physical 0x780ad000 -> [0x0181b000-0x01f07108) -
report_rom
map multi-boot module: physical 0x7879a000 -> [0x01f09000-0x02613248) -
rom_blk
map multi-boot module: physical 0x78ea5000 -> [0x02615000-0x02cecde0) -
part_blk
map multi-boot module: physical 0x7957d000 -> [0x02cee000-0x03385250) -
rump.lib.so
map multi-boot module: physical 0x79c15000 -> [0x03387000-0x039f00d0) -
rump_fs.lib.so
map multi-boot module: physical 0x7a27f000 -> [0x039f2000-0x03b74af0) -
rump_fs
map multi-boot module: physical 0x7a402000 -> [0x03b76000-0x04320248) -
ld.lib.so
map multi-boot module: physical 0x7abad000 -> [0x04322000-0x049c9628) -
fs_rom
map multi-boot module: physical 0x7b255000 -> [0x049cb000-0x052e4618) -
pci_drv
map multi-boot module: physical 0x7bb6f000 -> [0x052e6000-0x05923778) -
pci_device_pd
map multi-boot module: physical 0x7c1ad000 -> [0x05925000-0x06164808) -
fb_drv
map multi-boot module: physical 0x7c9ed000 -> [0x06166000-0x0684f9b0) -
ps2_drv
map multi-boot module: physical 0x7d0d7000 -> [0x06851000-0x06eaba00) -
acpi_drv
map multi-boot module: physical 0x7d732000 -> [0x06ead000-0x07e621f8) -
usb_drv
map multi-boot module: physical 0x7e6e8000 -> [0x07e64000-0x086ab0b8) -
ahci_drv
map multi-boot module: physical 0x7ef30000 -> [0x086ad000-0x090ad000) -
[... 468 lines cut ...]
[init -> pci_drv] 0:13.2 rewriting IRQ: 11 -> 17
[init -> pci_drv -> pci_device_pd] assignment of PCI device failed
[init -> usb_drv] dev_info: EHCI Host Controller
[init -> usb_drv] dev_info: new USB bus registered, assigned bus number 4
[init -> usb_drv] dev_info: applying AMD SB700/SB800/Hudson-2/3 EHCI dummy
qh workaround
[init -> usb_drv] dev_info: irq 17, io mem 0xfeb07000
[init -> usb_drv] dev_info: USB 2.0 started, EHCI 1.00
[init -> usb_drv] dev_info: USB hub found
[init -> usb_drv] dev_info: 5 ports detected
[init -> pci_drv] static short unsigned int
Pci::Irq_routing::rewrite(unsigned char, unsigned char, unsigned char,
unsigned char): rewrite 0:16.2 bridge bdf 0
[init -> pci_drv] 0:16.2 rewriting IRQ: 11 -> 17
[init -> pci_drv -> pci_device_pd] assignment of PCI device failed
[init -> usb_drv] dev_info: EHCI Host Controller
[init -> usb_drv] dev_info: new USB bus registered, assigned bus number 5
[init -> usb_drv] dev_info: applying AMD SB700/SB800/Hudson-2/3 EHCI dummy
qh workaround
[init -> usb_drv] dev_info: irq 17, io mem 0xfeb04000
[init -> usb_drv] dev_info: USB 2.0 started, EHCI 1.00
[init -> usb_drv] dev_info: USB hub found
[init -> usb_drv] dev_info: 4 ports detected
[init -> test-input] --- Input test is up ---
[init -> test-input] input buffer at 4000

alex-ab, 2015-06-19:
Ok, lets make sure that we get actually any irq. Please run the timer.run script of this updated branch https://github.com/alex-ab/genode/tree/issue_1547. It uses as user-mode timer the old PIT and enables on the hypervisor the so called "spinner", which shows you whether and which irqs fires. Please post the output of this run here, and please look at the monitor of your machine where you should see output as shown in the the nova documentation, V appendix, C console https://github.com/udosteinberg/NOVA/raw/master/doc/specification.pdf. There should by a yellow number at the 2. position flickering (means the pit irq fires and the number indicates the rate).
If this test works, please use the vary same branch and try run lwip.run and usb_hid.run and look for the irqs firing. Please describe what you see at your monitor, especially which position/which irq triggers.

NobodyIII, 2015-06-19:
Okay. Thanks! I'll try that. I still don't know what to do exactly for lwip
given that the run script is designed to be run in a VM. I should be able
to figure it out, but if I do something wrong it might just be my error
rather than an IRQ problem. Also, should I try my AHCI config with spinner?

alex-ab, 2015-06-19:
Yes, you may also run your ahci setup, just make sure _not_ to use the framebuffer driver. The lwip run script has nothing to do with a VM. It is useful to test the network driver with the lwIP TCP/IP stack.

NobodyIII, 2015-06-20:
Okay, thanks. As for lwip, what I was referring to was combining the client
and server, one in the VM and one on the host. From what you've said, it
sounds like it's pretty simple to do with two separate computers. I just
didn't want to get my configuration wrong.

NobodyIII, 2015-06-22:
The spinner output for timer looks good. I'll try the other scenarios
tomorrow. By the way, when I switch branches, I have to re-download ports
and rebuild things. Is there a good solution to this? I don't want to cause
unnecessary wear to my SSD.

NobodyIII, 2015-06-23:
I just tested usb_hid. No interrupts are triggered by mouse movements,
clicks, key presses, or removing USB devices when using a USB 2.0 port. The
only interrupts triggered for USB 2.0 are at the 2nd 1 position when I
insert USB devices. Interrupts are however triggered (and test-input gives
output) when using any of my USB 3.0 ports (both front and back). Note that
uhci, ehci, and xhci are all set to yes.
As for ahci, I get 2 interrupts in the 2nd 3 position.

NobodyIII, 2015-06-23:
Note that the interrupts for USB 3.0 are all in the last column and are
purple. As for lwip, it works, and its interrupts are also purple and in
the last column. Also, my power button creates interrupts in the first 9
position. Genode doesn't respond to it, but I don't have any reason to
believe it's supposed to yet.

alex-ab, 2015-06-26:
@NobodyIII: Thank you for testing. Please use the updated branch 
https://github.com/alex-ab/genode/tree/issue_1547 and run timer.run. Please post the lines of the kernel debug output - should look like this:
```
[ 0] GSI 2 ---------------------
[ 0] APIC reg_base=0xffffffffbfdfe000 ID:0x2 VER:0x20 PRQ:0 GSI_BASE:0 IRT_MAX=39
[init -> client] --- timer test ---
[ 0] GSI 2 ---------------------
[ 0] APIC reg_base=0xffffffffbfdfe000 ID:0x2 VER:0x20 PRQ:0 GSI_BASE:0 IRT_MAX=39
```

NobodyIII, 2015-06-27:
Well, here's the output:
[ 0] GSI 2 ---------------------
[ 0] APIC reg_base=0xffffffffbfdfe000 ID:0x9 VER:0x21 PRQ:1 GSI_BASE:0
IRT_MAX=23
[ 0] APIC reg_base=0xffffffffbfdfd000 ID:0xa VER:0x21 PRQ:1 GSI_BASE:24
IRT_MAX=31
[init -> client] --- timer test ---
[ 0] GSI 2 ---------------------
[ 0] APIC reg_base=0xffffffffbfdfe000 ID:0x9 VER:0x21 PRQ:1 GSI_BASE:0
IRT_MAX=23
[ 0] APIC reg_base=0xffffffffbfdfd000 ID:0xa VER:0x21 PRQ:1 GSI_BASE:24
IRT_MAX=31

chelmuth, 2015-06-29:
Hm, your board has two IO-APICs with the second (0xffffffffbfdfd000) using a GSI vector base of 24. I'm anxious to hear what @alex-ab thinks about this tomorrow.

NobodyIII, 2015-06-29:
Sorry, but what exactly does that mean? How common is it to have 2
IO-APICs? And what does the base vector of 24 mean?

chelmuth, 2015-06-29:
From our (limited) experience this is uncommon with standard client PC hardware. Let's wait until @alex-ab sheds some light on this.

alex-ab, 2015-06-30:
From my perspective, this looks all ok. 2 IO-APICs are not uncommon, according to the www. After re-reading the messages regrading the irqs for ahci and usb (yellow spinners) they match actually. So for ahci you get already 2 irqs and for usb_2.0 1 irq. Maybe we should add some debugging to the ahci driver, already asked @ssumpf to support us here.

chelmuth, 2015-06-30:
@alex-ab http://www/ ?

alex-ab, 2015-06-30:
Sorry, I just meant the worldwideweb. But more specifically the Intel MultiProcessor Specification in Appendix D mentions this setup. Addjusted the link of the orginal message.

ssumpf, 2015-07-01:
@NobodyIII: Can you try out the commit above send any
'irq: x state: x' or 'loop irq: x state: x' outputs?
Thx

NobodyIII, 2015-07-26:
I tried it. The AHCI driver started fine (it detected the controller), but it didn't give any of those messages. I double-checked that I had updated ata_driver.h as well as the ahci_drv binary.

alex-ab, 2015-08-05:
@NobodyIII: please try commits 9198bc2450deee1aa45a5466b998d7b0e1a3b051 and 0b997c9b15cb359f2600327871ce9f15f1613ce8 and send output done by the ahci driver and the platform driver

NobodyIII, 2015-08-08:
[init -> ahci_drv] --- Starting AHCI driver -> done right .-) --
[init -> test-blk-cli-nowrite] starting
[init -> test-blk-cli-nowrite] alloc done
[init] child "timer" announces service "Timer"
[init] child "rom_blk" announces service "Block"
[init -> platform_drv] platform driver started
[init -> platform_drv] Could not open ROM session for module "device_pd"
[init -> platform_drv] PCI device protection domain for IOMMU support is
not available
[init -> acpi_drv] void Acpi_table::_parse_tables(T*, Genode::uint32_t)
[with T = long long unsigned int; Genode::uint32_t = unsigned int]: Found
MADT
[init -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0
[init -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: f
[init -> acpi_drv] void Acpi_table::_parse_tables(T*, Genode::uint32_t)
[with T = long long unsigned int; Genode::uint32_t = unsigned int]: Found
MCFG
[init -> acpi_drv] MCFG BASE 0xe0000000 seg 00 bus 00-ff
[init] child "platform_drv" announces service "Platform"
[init -> platform_drv] 0:11.0 rewriting IRQ: 11 -> 19
[ 0] GSI 19 ---------------------
[ 0] APIC reg_base=0xffffffffbfdfe000 ID:0x9 VER:0x21 PRQ:1 GSI_BASE:0
IRT_MAX=23
[ 0] APIC reg_base=0xffffffffbfdfd000 ID:0xa VER:0x21 PRQ:1 GSI_BASE:24
IRT_MAX=31
[init -> ahci_drv] AHCI found (vendor: 1002 device: 4391 class: 00010601)
[init -> platform_drv] 0:11.0 uses IRQ, vector 0x13
[init -> ahci_drv]     version: 1.0200
[init -> ahci_drv]     command slots: 32
[init -> ahci_drv]     native command queuing: yes
[init -> ahci_drv]     64 bit support: yes
[init -> platform_drv] bool Genode::Irq_proxy<THREAD>::ack_irq() [with
THREAD = NoThread]: irq=19 1/1 woken_up=0
[ 0] GSI 19 ---------------------
[ 0] APIC reg_base=0xffffffffbfdfe000 ID:0x9 VER:0x21 PRQ:1 GSI_BASE:0
IRT_MAX=23
[ 0] APIC reg_base=0xffffffffbfdfd000 ID:0xa VER:0x21 PRQ:1 GSI_BASE:24
IRT_MAX=31
[init -> platform_drv] virtual void Platform::Irq_component::_ack_irq():
irq.cc ack irq=19
[init -> ahci_drv]     number of ports: 4 pi: f
[init -> ahci_drv]         #0: ATA
int main(): --- init created, waiting for exit condition ---
[init -> ahci_drv]         #1: ATA
[init -> ahci_drv]         #2: ATAPI
[init -> ahci_drv]         #3: off
[init -> platform_drv] virtual void Platform::Irq_thread::entry(): got
signal for a irq
[init -> platform_drv] void
Genode::Irq_proxy<THREAD>::notify_about_irq(unsigned int) [with THREAD =
NoThread]: irq=19 count=1
[init -> platform_drv] void
Genode::Irq_proxy<THREAD>::notify_about_irq(unsigned int) [with THREAD =
NoThread]: irq=19 - submit 1. signal
[init -> ahci_drv] void Ahci::handle_irq(unsigned int): got irq -
port_list=4
[init -> platform_drv] bool Genode::Irq_proxy<THREAD>::ack_irq() [with
THREAD = NoThread]: irq=19 1/1 woken_up=0
[init -> platform_drv] virtual void Platform::Irq_component::_ack_irq():
irq.cc ack irq=19
[init -> platform_drv] virtual void Platform::Irq_thread::entry(): got
signal for a irq
[init -> platform_drv] void
Genode::Irq_proxy<THREAD>::notify_about_irq(unsigned int) [with THREAD =
NoThread]: irq=19 count=1
[init -> platform_drv] void
Genode::Irq_proxy<THREAD>::notify_about_irq(unsigned int) [with THREAD =
NoThread]: irq=19 - submit 1. signal
[init -> ahci_drv] void Ahci::handle_irq(unsigned int): got irq -
port_list=4
[init -> platform_drv] bool Genode::Irq_proxy<THREAD>::ack_irq() [with
THREAD = NoThread]: irq=19 1/1 woken_up=0
[init -> platform_drv] virtual void Platform::Irq_component::_ack_irq():
irq.cc ack irq=19

NobodyIII, 2015-09-03:
Any thoughts about where to go from here? (see my previous comment)

ssumpf, 2015-09-04:
@NobodyIII: Our current suspicion is that the interrupts are delivered to the ATAPI device (#2), since they do not end up in the ATA-driver. In the next step we would have to verify this and, if that is the case, the only thing that we can do from here is to dump the port registers of the ATA devices. If Alex or myself find some spare time, we will submit a debugging patch to this issue.

NobodyIII, 2015-09-05:
I unplugged all but my first SATA drive, but it still didn't work. Here's
the output:

[init] child "acpi_report_rom" announces service "Report"
[init] child "acpi_report_rom" announces service "ROM"
[init -> ahci_drv] --- Starting AHCI driver -> done right .-) --
[init -> test-blk-cli-nowrite] starting
[init -> test-blk-cli-nowrite] alloc done
[init] child "timer" announces service "Timer"
[init] child "rom_blk" announces service "Block"
[init -> platform_drv] platform driver started
[init -> platform_drv] Could not open ROM session for module "device_pd"
[init -> platform_drv] PCI device protection domain for IOMMU support is
not available
[init -> acpi_drv] void Acpi_table::_parse_tables(T*, Genode::uint32_t)
[with T = long long unsigned int; Genode::uint32_t = unsigned int]: Found
MADT
[init -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0
[init -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: f
[init -> acpi_drv] void Acpi_table::_parse_tables(T*, Genode::uint32_t)
[with T = long long unsigned int; Genode::uint32_t = unsigned int]: Found
MCFG
[init -> acpi_drv] MCFG BASE 0xe0000000 seg 00 bus 00-ff
[init] child "platform_drv" announces service "Platform"
[init -> platform_drv] 0:11.0 rewriting IRQ: 11 -> 19
[ 0] GSI 19 ---------------------
[ 0] APIC reg_base=0xffffffffbfdfe000 ID:0x9 VER:0x21 PRQ:1 GSI_BASE:0
IRT_MAX=23
[ 0] APIC reg_base=0xffffffffbfdfd000 ID:0xa VER:0x21 PRQ:1 GSI_BASE:24
IRT_MAX=31
[init -> ahci_drv] AHCI found (vendor: 1002 device: 4391 class: 00010601)
[init -> platform_drv] 0:11.0 uses IRQ, vector 0x13
[init -> ahci_drv]     version: 1.0200
[init -> ahci_drv]     command slots: 32
[init -> ahci_drv]     native command queuing: yes
[init -> ahci_drv]     64 bit support: yes
[init -> platform_drv] bool Genode::Irq_proxy<THREAD>::ack_irq() [with
THREAD = NoThread]: irq=19 1/1 woken_up=0
[ 0] GSI 19 ---------------------
[ 0] APIC reg_base=0xffffffffbfdfe000 ID:0x9 VER:0x21 PRQ:1 GSI_BASE:0
IRT_MAX=23
[ 0] APIC reg_base=0xffffffffbfdfd000 ID:0xa VER:0x21 PRQ:1 GSI_BASE:24
IRT_MAX=31
[init -> platform_drv] virtual void Platform::Irq_component::_ack_irq():
irq.cc ack irq=19
[init -> ahci_drv]     number of ports: 4 pi: f
[init -> ahci_drv]         #0: ATA
int main(): --- init created, waiting for exit condition ---
[init -> ahci_drv]         #1: off
[init -> ahci_drv]         #2: off
[init -> ahci_drv]         #3: off

alex-ab, 2015-09-07:
@NobodyIII: may you please apply following patch to the kernel and rebuild it and rerun the scenario, does it change respectively do you see the kernel message of the patch ?

```

diff --git a/include/sm.hpp b/include/sm.hpp
index c14b072..9bf3dc5 100644
--- a/include/sm.hpp
+++ b/include/sm.hpp
@@ -28,6 +28,7 @@
 class Sm : public Kobject, public Refcount, public Queue<Ec>, public Queue<Si>, public Si
 {
     private:
+public:
         mword counter;
         bool  rcu;
 
diff --git a/src/si.cpp b/src/si.cpp
index 760f2ed..6ebcb0f 100644
--- a/src/si.cpp
+++ b/src/si.cpp
@@ -64,6 +64,12 @@ void Si::chain(Sm *si)
 
     if (sm)
         sm->add_ref();
+
+    if (kern_sm && kern_sm->counter) {
+      trace(0, " we have already irq pending %lu", kern_sm->counter);
+      kern_sm->submit();
+    }
+
 }
 
 void Si::submit()

```

NobodyIII, 2015-09-08:
I applied the patch. Here's the output:

[init] child "acpi_report_rom" announces service "Report"
[init] child "acpi_report_rom" announces service "ROM"
[init -> ahci_drv] --- Starting AHCI driver -> done right .-) --
[init -> test-blk-cli-nowrite] starting
[init -> test-blk-cli-nowrite] alloc done
[init] child "timer" announces service "Timer"
[init] child "rom_blk" announces service "Block"
[init -> platform_drv] platform driver started
[init -> platform_drv] Could not open ROM session for module "device_pd"
[init -> platform_drv] PCI device protection domain for IOMMU support is
not available
[init -> acpi_drv] void Acpi_table::_parse_tables(T*, Genode::uint32_t)
[with T = long long unsigned int; Genode::uint32_t = unsigned int]: Found
MADT
[init -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0
[init -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: f
[init -> acpi_drv] void Acpi_table::_parse_tables(T*, Genode::uint32_t)
[with T = long long unsigned int; Genode::uint32_t = unsigned int]: Found
MCFG
[init -> acpi_drv] MCFG BASE 0xe0000000 seg 00 bus 00-ff
[init] child "platform_drv" announces service "Platform"
[init -> platform_drv] 0:11.0 rewriting IRQ: 11 -> 19
[ 0] GSI 19 ---------------------
[ 0] APIC reg_base=0xffffffffbfdfe000 ID:0x9 VER:0x21 PRQ:1 GSI_BASE:0
IRT_MAX=23
[ 0] APIC reg_base=0xffffffffbfdfd000 ID:0xa VER:0x21 PRQ:1 GSI_BASE:24
IRT_MAX=31
[init -> ahci_drv] AHCI found (vendor: 1002 device: 4391 class: 00010601)
[init -> platform_drv] 0:11.0 uses IRQ, vector 0x13
[init -> ahci_drv]     version: 1.0200
[init -> ahci_drv]     command slots: 32
[init -> ahci_drv]     native command queuing: yes
[init -> ahci_drv]     64 bit support: yes
[init -> platform_drv] bool Genode::Irq_proxy<THREAD>::ack_irq() [with
THREAD = NoThread]: irq=19 1/1 woken_up=0
[ 0] GSI 19 ---------------------
[ 0] APIC reg_base=0xffffffffbfdfe000 ID:0x9 VER:0x21 PRQ:1 GSI_BASE:0
IRT_MAX=23
[ 0] APIC reg_base=0xffffffffbfdfd000 ID:0xa VER:0x21 PRQ:1 GSI_BASE:24
IRT_MAX=31
[init -> platform_drv] virtual void Platform::Irq_component::_ack_irq():
irq.cc ack irq=19
[init -> ahci_drv]     number of ports: 4 pi: f
[init -> ahci_drv]         #0: ATA
int main(): --- init created, waiting for exit condition ---
[init -> ahci_drv]         #1: ATA
[init -> ahci_drv]         #2: ATAPI
[init -> ahci_drv]         #3: off
[init -> platform_drv] virtual void Platform::Irq_thread::entry(): got
signal for a irq
[init -> platform_drv] void
Genode::Irq_proxy<THREAD>::notify_about_irq(unsigned int) [with THREAD =
NoThread]: irq=19 count=1
[init -> platform_drv] void
Genode::Irq_proxy<THREAD>::notify_about_irq(unsigned int) [with THREAD =
NoThread]: irq=19 - submit 1. signal
[init -> ahci_drv] void Ahci::handle_irq(unsigned int): got irq -
port_list=4
[init -> platform_drv] bool Genode::Irq_proxy<THREAD>::ack_irq() [with
THREAD = NoThread]: irq=19 1/1 woken_up=0
[init -> platform_drv] virtual void Platform::Irq_component::_ack_irq():
irq.cc ack irq=19
[init -> platform_drv] virtual void Platform::Irq_thread::entry(): got
signal for a irq
[init -> platform_drv] void
Genode::Irq_proxy<THREAD>::notify_about_irq(unsigned int) [with THREAD =
NoThread]: irq=19 count=1
[init -> platform_drv] void
Genode::Irq_proxy<THREAD>::notify_about_irq(unsigned int) [with THREAD =
NoThread]: irq=19 - submit 1. signal
[init -> ahci_drv] void Ahci::handle_irq(unsigned int): got irq -
port_list=4
[init -> platform_drv] bool Genode::Irq_proxy<THREAD>::ack_irq() [with
THREAD = NoThread]: irq=19 1/1 woken_up=0
[init -> platform_drv] virtual void Platform::Irq_component::_ack_irq():
irq.cc ack irq=19

alex-ab, 2015-09-10:
NobodyIII: may I ask how you configured your BIOS regarding your disks -
ftp://europe.asrock.com/manual/970%20Extreme3.pdf - there seem to be several options how to do it (sorry if you already answered this question in some of the older mails)

alex-ab, 2015-09-10:
@NobodyIII: please re-run with commit f7b077423e59c9df3c60c03b7b1f13384c6b7a19 and provide the output of the added lines - thanks

chelmuth, 2015-09-10:
The 970 has an IOMMU. May the IRQ remapping hit us in some way? @NobodyIII Could you disable the IOMMU in the BIOS (just to be sure)?

NobodyIII, 2015-09-11:
Current configuration:
SATA Mode: AHCI mode
AMD AHCI BIOS ROM: disabled
SATA IDE Combined Mode: enabled
Aggressive Link Power Management: enabled
Hard Disk S.M.A.R.T: disabled

NobodyIII, 2015-09-11:
I disabled the IOMMU, both with and without my other SATA drives, and I saw
no difference in the output.

NobodyIII, 2015-09-11:
Here's are the new results:

[init] child "acpi_report_rom" announces service "Report"
[init] child "acpi_report_rom" announces service "ROM"
[init -> ahci_drv] --- Starting AHCI driver -> done right .-) --
[init -> test-blk-cli-nowrite] starting
[init -> test-blk-cli-nowrite] alloc done
[init] child "timer" announces service "Timer"
[init] child "rom_blk" announces service "Block"
[init -> platform_drv] platform driver started
[init -> platform_drv] Could not open ROM session for module "device_pd"
[init -> platform_drv] PCI device protection domain for IOMMU support is
not available
[init -> acpi_drv] void Acpi_table::_parse_tables(T*, Genode::uint32_t)
[with T = long long unsigned int; Genode::uint32_t = unsigned int]: Found
MADT
[init -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0
[init -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: f
[init -> acpi_drv] void Acpi_table::_parse_tables(T*, Genode::uint32_t)
[with T = long long unsigned int; Genode::uint32_t = unsigned int]: Found
MCFG
[init -> acpi_drv] MCFG BASE 0xe0000000 seg 00 bus 00-ff
[init] child "platform_drv" announces service "Platform"
[init -> platform_drv] 0:11.0 rewriting IRQ: 11 -> 19
[ 0] GSI 19 ---------------------
[ 0] APIC reg_base=0xffffffffbfdfe000 ID:0x9 VER:0x21 PRQ:1 GSI_BASE:0
IRT_MAX=23
[ 0] APIC reg_base=0xffffffffbfdfd000 ID:0xa VER:0x21 PRQ:1 GSI_BASE:24
IRT_MAX=31
[init -> ahci_drv] AHCI found (vendor: 1002 device: 4391 class: 00010601)
[init -> platform_drv] 0:11.0 uses IRQ, vector 0x13
[init -> ahci_drv]     version: 1.0200
[init -> ahci_drv]     command slots: 32
[init -> ahci_drv]     native command queuing: yes
[init -> ahci_drv]     64 bit support: yes
[init -> ahci_drv]     extended CAP2: 0
[init -> platform_drv] bool Genode::Irq_proxy<THREAD>::ack_irq() [with
THREAD = NoThread]: irq=19 1/1 woken_up=0
[ 0] GSI 19 ---------------------
[ 0] APIC reg_base=0xffffffffbfdfe000 ID:0x9 VER:0x21 PRQ:1 GSI_BASE:0
IRT_MAX=23
[ 0] APIC reg_base=0xffffffffbfdfd000 ID:0xa VER:0x21 PRQ:1 GSI_BASE:24
IRT_MAX=31
[init -> platform_drv] virtual void Platform::Irq_component::_ack_irq():
irq.cc ack irq=19
[init -> ahci_drv]     number of ports: 4 pi: f
[init -> ahci_drv]         #0: ATA
[init -> ahci_drv]         #0: sleep state 0
int main(): --- init created, waiting for exit condition ---
[init -> ahci_drv]         #1: ATA
[init -> ahci_drv]         #1: sleep state 0
[init -> ahci_drv]         #2: ATAPI
[init -> ahci_drv]         #2: sleep state 0
[init -> ahci_drv]         #3: off
[init -> platform_drv] virtual void Platform::Irq_thread::entry(): got
signal for a irq
[init -> platform_drv] void
Genode::Irq_proxy<THREAD>::notify_about_irq(unsigned int) [with THREAD =
NoThread]: irq=19 count=1
[init -> platform_drv] void
Genode::Irq_proxy<THREAD>::notify_about_irq(unsigned int) [with THREAD =
NoThread]: irq=19 - submit 1. signal
[init -> platform_drv] bool Genode::Irq_proxy<THREAD>::ack_irq() [with
THREAD = NoThread]: irq=19 1/1 woken_up=0
[init -> platform_drv] virtual void Platform::Irq_component::_ack_irq():
irq.cc ack irq=19
[init -> platform_drv] virtual void Platform::Irq_thread::entry(): got
signal for a irq
[init -> platform_drv] void
Genode::Irq_proxy<THREAD>::notify_about_irq(unsigned int) [with THREAD =
NoThread]: irq=19 count=1
[init -> platform_drv] void
Genode::Irq_proxy<THREAD>::notify_about_irq(unsigned int) [with THREAD =
NoThread]: irq=19 - submit 1. signal
[init -> platform_drv] bool Genode::Irq_proxy<THREAD>::ack_irq() [with
THREAD = NoThread]: irq=19 1/1 woken_up=0
[init -> platform_drv] virtual void Platform::Irq_component::_ack_irq():
irq.cc ack irq=19

[8 more replies]

closed 2019-03-14 by alex-ab