genodelabs/genode #1549: Add IoMux to raspberry Pi platform driver

trimpim, 2015-05-30:
Add a RPC interface to configure the IoMux to the platform driver of the Raspberry Pi.

nfeske, 2015-06-04:
I wonder, is there a need to change the IoMux configuration dynamically at runtime? If the IoMux configuration is just done a single time at the start of the system, it might be sufficient to perform this configuration by the platform driver and not offer any session interface at all. @trimpim Could you shed some light on the need for the dynamic adjustment of the IoMux settings? Which kind of clients do you have in mind?

trimpim, 2015-06-06:
After thinking about it, there is no need for an RPC interface, a static configuration is sufficient for my current plans. My current project uses SPI/I2C co display information on an E-Paper display and some GPIO to interact with the system.

trimpim, 2016-04-02:
At last I have found some time to continue the implementation of the IoMux.
As the configuration for the platform_drv is automatically provided by the append_platform_drv_config function I implemented the IoMux as a separate program (iomux_drv).
When I searched in the repository on how to use the classes to access IO-ports I found out that the platform driver for the iMx53 platform provides an RPC interface to enable i2c, buttons and PWM.

@nfeske how should I continue now? I see three possibilities:
 1) platform_drv of RPI also provides an RPC interface for setting the ports (the same as iMx53)
 2) continue with the separate program iomux_drv
 3) combine both whereas iomux_drv reads the configuration and uses the RPC to set the ports in platform_drv (I think this is a bit overkill)

My current progress can be found at:
https://github.com/trimpim/genode/tree/issue1549_new

nfeske, 2016-04-04:
As far as I can see, there is no overlap or inter-dependency between the platform driver and your iomux_drv. So I see no compelling reason to merge both into one program and would rather keep them separate. Do you have the need to access the IOMUX hardware via an RPC interface? I would try to avoid that if possible.

For the dynamic access to GPIO pins, it might make sense to let the iomux_drv provide the existing "Gpio" session interface (https://github.com/genodelabs/genode/tree/master/repos/os/include/gpio_session).

trimpim, 2016-04-04:
My thoughts were more in the direction of consistency between the different hardware platforms than the need for a IOMUX RPC interface.
My plan was, that the IOMUX reads the configuration from its parent, sets the IOMUX accordingly and exits.
But It could be easily expanded to provide the "Gpio" session interface.
An other option could be, that it uses the gpio_drv to set the pins if they are configured as GPIO, but from an user perspective dosn't make much sense IMHO as the user needs an Gpio session later on anyway to read and write the GPIO.
My preferred solution would be to extract the code to set up the IOMUX from the gpio_drv, integrate it into iomux_drv and remove the "setup" functions from the "Gpio" session interface. The problem I see in this, is that it breaks all existing code of other users (not fun for all other users).

skalk, 2016-04-05:
I agree to trimpim that having the iomux driver within the platform driver is good for consistency reasons. In general, the iomux controller settings are fully static and fixed for ready-to-use boards. On the other hand those settings are relevant form more than one device, so it should not be accessed individually by the drivers. Therefore, they can be part of the platform initialization, and normally does not need to be configured at all, but can be hard-coded.

@trimpim: To me it is not quite clear to this point in which manner the iomux settings you need for the e-paper display differ from the current ones. Do you need some additions so that the I2C controller works nicely, or does it really breaks current device usage of the Rasperry PI? If not, I would vote for simply setting what is need regarding the iomux controller during initialization of the platform driver.

trimpim, 2016-04-05:
The e-paper needs nearly all of the 40 pins. On the 26 pin RPI some functions (e.g. Buttons aren't) available.
The break for other users is, that the pins 6 and 8 (default UART0) are used for the Display power control. For me the fix will be to use the pins 32 and 33, that aren't used by the board and can be configured as UART0 on an alternate function.
The schematics are at: https://github.com/PiSupply/PaPiRus/blob/master/hardware/Old%20Version%20-%20v1.8/2014-035-01-Pi_ePaper-circuit_v1_8.pdf

In general there are many Extension Boards for the RPI, and IMHO it would be nice, if there is a easy way to configure the IOMUX depending on the module that is used.

skalk, 2016-04-12:
@trimpim sorry for my delay. Thanks for the explanation! Now I better understand your undertaking. Given the circumstances, I would vote to do the initialization of the IOMUX controller within the initialization of the platform driver, and that using the settings found in the platform driver's configuration. Thereby, no additional RPC interface is needed, and the IOMUX controller that is critical for potentially more than one device is not dependent on different device drivers. On the other hand no hotplugging of different peripherals at the pins is supported without restarting the platform driver, but I assume this is not necessarily needed right now. Do you agree?

trimpim, 2016-04-12:
@skalk yes, not having hot plug functionality is not an issue for me, so a static IOMUX configuration is also preferred by me.
As for how to bring the IOMUX configuration in to the platform_driver, I had a talk with @alex-ab at the coffee table. Maybe it could be done as in the x86 platform for the ACPI configuration. Where the platform driver waits for an initial acpi report before configuring the hardware.
Doing so allows for easy changes in the configuration for the IOMUX without modifying the file ${genode_dir}/repos/base/run/platform_drv.inc where the configuration for the platform driver normally comes from.
If this is okay, I would implement it like this.
 - Implement the iomux_drv (or just iomux as an app?). This just reads the configuration from the Parent and writes it in to the iomux report_rom. If no configuration is present, it creates a default configuration and sends this. after sending the configuration it exits.
 - Change platform_drv to wait for the configuration report ROM and interpret it.
 - Remove the setup functions from the GPIO session interface.
 - ensure the documentation is up to date and accurate.

chelmuth, 2016-04-13:
This sounds like a well-conceived plan. Regarding your first bullet point, I'd like to ask if it could be possible to drop the *iomux app* completely in the case of static - integration time - configuration. This could be done with a overloaded hook function in *platform_drv.inc* like `platform_drv_policy` for the x86 policy configuration.

skalk, 2016-04-13:
That sounds good, having a default iomux setting when no report is available, and otherwise to follow the protocol developed by trimpim.

chelmuth, 2016-04-13:
@skalk that's not what I wanted to express. I was thinking in the line of: The platform_drv requests the *iomux configuration* ROM, which can be provided by a boot module (i.e., ROM) in the case of static configuration. If the requirement for a dynamic (or boot-time generated) configuration arises, then the described iomux app plus report_rom can be implemented. For now, it sounds like creating the *iomux configuration* on integration time is sufficient. This way, we only need to enhance the platform_drv at the moment.

skalk, 2019-03-22:
@trimpim There is not much progress in this issue. I would like to close it for now, until there is a unified concept for a platform_drv for ARM available. It seems to me that hard-coding is an option everybody can live with for the time-being?

trimpim, 2019-03-25:
As I don't have time to work on it currently and wasn't even aware of the issue, I'll close it.

closed 2019-03-25 by trimpim