genodelabs/genode #1554: VirtualBox: dynamic adaptation to screen-size changes

nfeske, 2015-06-02:
Thanks to the VirtualBox guest additions, VirtualBox is principally able to dynamically respond to screen-size changes. E.g., when changing the size of the window that contains the guest OS, the guest OS adapts its desktop environment to the new window size. This feature is desired in scenarios like issue #1552. However, our version of VirtualBox does not support it yet.

nfeske, 2015-06-16:
Fantastic! It works perfectly. Thanks @cproc :-)

chelmuth, 2015-06-16:
Most interestingly the fix does not change anything when running Windows 8.1. Don't ask me why...

chelmuth, 2015-06-17:
Please ignore my last comment as it applies to #1589 - I apologize for any confusion caused.

commit 6110a67e (2015-06-16): vbox: dynamic adaptation to screen-size changes

Fixes #1554
closed 2015-06-22 by cproc