genodelabs/genode #1557: VBox pointer: policy selection via nitpicker session labels

nfeske, 2015-06-02:
The current version of the vbox_pointer assumes that VirtualBox has a dedicated nitpicker domain "vbox". The code has an annotation (XXX) that it should read this policy from the configuration. Moreover, it would be nice to have a way to express a policy do a specific nitpicker session label (alternatively to a nitpicker domain). For example, in the Turmvilla scenario of issue #1552, VirtualBox runs in the same domain as all regular GUI clients and has the label "wm -> cli_monitor -> linux -> nit_fb". It would be ideal to be able to define a policy that matches all views that belong to "wm -> cli_monitor -> linux".

chelmuth, 2015-06-02:
Exactly what I stumbled over today: I have more than one instance of VirtualBox running, each of them providing a shape report. In this scenario vbox_pointer has to display the shape associated to the currently hovered session. One special domain seems to be just a stop gap solution I have to generalize now.

chelmuth, 2015-06-09:
@nfeske please have a close look at cbef9d2b9da87b5210f9733711e9aef95d67254f and 120cfb66fc8616c0f37667a161f2097ea43088c3.

nfeske, 2015-06-09:
Delicious! :-)

Just a small question crossed my mind: Why are you naming the children and labels all uppercase? This is quite a derivation from from our usual practice.

chelmuth, 2015-06-09:
You're right, this looks confusing and is just a remnant from my early testing phase. I will polish tomorrow ;-)

commit fec76af2 (2015-06-10): vbox_pointer: policy-based shape selection

Fixes #1557
commit 4e8d5952 (2015-06-17): vbox_pointer: policy-based shape selection

Fixes #1557
commit 7897e522 (2015-06-22): vbox_pointer: policy-based shape selection

Fixes #1557
closed 2015-06-22 by chelmuth