genodelabs/genode #1559: pci: regression in policy and interrupt handling

chelmuth, 2015-06-03:
Commit 1592e78 of issue #1541 affects the behavior of pci_drv in a strange way. Currently, I do not suspect the commit to be buggy, instead it seems to uncover another shortcoming. The following diff reflects the changes to `run/usb_hid` if the commit is removed from master. Please note that the IRQ rerouting seems broken.
```diff
--- /tmp/master.log	2015-06-03 12:28:33.528074001 +0200
+++ /tmp/master-without.log	2015-06-03 12:28:33.528074001 +0200
@@ -2,12 +2,12 @@
 
 [ 0] CORE:0:0:0 6:3c:3:1 [12] Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz
 [ 3] CORE:0:3:0 6:3c:3:1 [12] Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz
-[ 4] CORE:0:0:1 6:3c:3:1 [12] Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz
-[ 6] CORE:0:2:1 6:3c:3:1 [12] Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz
 [ 1] CORE:0:1:0 6:3c:3:1 [12] Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz
 [ 5] CORE:0:1:1 6:3c:3:1 [12] Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz
 [ 2] CORE:0:2:0 6:3c:3:1 [12] Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz
 [ 7] CORE:0:3:1 6:3c:3:1 [12] Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz
+[ 6] CORE:0:2:1 6:3c:3:1 [12] Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz
+[ 4] CORE:0:0:1 6:3c:3:1 [12] Intel(R) Core(TM) i7-4770 CPU @ 3.40GHz
 [ 0] DMAR:0xffffffff81023078 FRR:0 FR:0x6 BDF:0:1d:0 FI:0xd869f000
 Hypervisor features VMX
 Hypervisor does not feature SVM
@@ -106,7 +106,7 @@
  Block: [da000000,100000000) size=26000000 avail=26000000 max_avail=fffffffbe01ff000
  Block: [41fe00000,fffffffffffff000) size=fffffffbe01ff000 avail=fffffffbe01ff000 max_avail=fffffffbe01ff000
  => mem_size=18446744056664612864 (17592186028160 MB) / mem_avail=18446744056664612864 (17592186028160 MB)
-Genode 15.05-9-g2c09011
+Genode 15.05-9-g2c09011 <local changes>
 int main(): --- create local services ---
 int main(): --- start init ---
 int main(): transferred 16214 MB to init
@@ -128,6 +128,17 @@
 [init -> acpi -> pci_drv -> pci_device_pd] PCI device pd started
 [init -> acpi -> pci_drv] 00 BDF start 0, functions: 0x4000, physical base 0xf8000000
 [init -> acpi] ACPI table format is supported - rewrite GSIs
+[init -> acpi] Rewriting 00:02.0 IRQ: 11 -> GSI: 16
+[init -> acpi] Rewriting 00:03.0 IRQ: 11 -> GSI: 16
+[init -> acpi] Rewriting 00:14.0 IRQ: 10 -> GSI: 21
+[init -> acpi] Rewriting 00:16.0 IRQ: 11 -> GSI: 16
+[init -> acpi] Rewriting 00:19.0 IRQ: 07 -> GSI: 20
+[init -> acpi] Rewriting 00:1b.0 IRQ: 14 -> GSI: 22
+[init -> acpi] Rewriting 00:1d.0 IRQ: 11 -> GSI: 23
+[init -> acpi] Rewriting 00:1f.2 IRQ: 15 -> GSI: 19
+[init -> acpi] Rewriting 00:1f.3 IRQ: 06 -> GSI: 18
+[init -> acpi] Rewriting 02:00.0 IRQ: 11 -> GSI: 16
+[init -> acpi] Rewriting 03:00.0 IRQ: 06 -> GSI: 18
 [init -> usb_drv] Services::Services(): Could not read screen resolution in config node
 [init -> usb_drv] Services::Services(): No <storage> config node found - not starting the USB Storage (Block) service
 [init -> usb_drv] Services::Services(): No <nic> config node found - not starting the USB Nic (Network) service
@@ -137,7 +148,7 @@
 [init -> usb_drv] Enabled XHCI (USB 3.0) support
 [init -> usb_drv] dev_info: xHCI Host Controller
 [init -> usb_drv] dev_info: new USB bus registered, assigned bus number 1
-[init -> acpi -> pci_drv] 0:14.0 uses MSI 64bit, vector 0x7f, address 0xfee00018
+[init -> acpi -> pci_drv] 0:14.0 uses MSI 64bit, vector 0x7d, address 0xfee00018
 [init -> usb_drv] dev_info: USB hub found
 [init -> usb_drv] dev_info: 15 ports detected
 [init -> usb_drv] dev_info: xHCI Host Controller
@@ -147,8 +158,8 @@
 [init -> usb_drv] dev_info: EHCI Host Controller
 [init -> usb_drv] dev_info: new USB bus registered, assigned bus number 3
 [init -> usb_drv] dev_warn: Enabling legacy PCI PM
-[init -> acpi -> pci_drv] 0:1d.0 uses IRQ, vector 0xb
-[init -> usb_drv] dev_info: irq 11, io mem 0xf7e3b000
+[init -> acpi -> pci_drv] 0:1d.0 uses IRQ, vector 0x17
+[init -> usb_drv] dev_info: irq 23, io mem 0xf7e3b000
 int main(): --- init created, waiting for exit condition ---
 [init -> usb_drv] dev_info: USB 2.0 started, EHCI 1.00
 [init -> usb_drv] dev_info: USB hub found
@@ -162,5 +173,19 @@
 [init -> usb_drv] upgrading quota donation for SIGNAL session (8192 bytes)
 [init -> usb_drv] dev_info: new SuperSpeed USB device number 2 using xhci_hcd
 [init -> usb_drv] dev_info: new high-speed USB device number 2 using ehci-pci
-[init -> usb_drv] __wait_completion_timeout(): Timeout
+[init -> usb_drv] dev_info: USB hub found
+[init -> usb_drv] dev_info: 8 ports detected
+[init -> usb_drv] dev_info: new full-speed USB device number 3 using xhci_hcd
+[init -> usb_drv] dev_warn: ep <warning: unsupported format string argument>x - rounding interval to 129 microframes, ep desc says 1024 microframes
+[init -> usb_drv] dev_info: USB hub found
+[init -> usb_drv] dev_info: 4 ports detected
+Quota exceeded! amount=20480, size=8192, consumed=16384
+[init -> usb_drv] upgrading quota donation for SIGNAL session (8192 bytes)
+[init -> usb_drv] dev_info: new full-speed USB device number 4 using xhci_hcd
+[init -> usb_drv] dev_warn: ep <warning: unsupported format string argument>x - rounding interval to 129 microframes, ep desc says 64 microframes
+[init -> usb_drv] dev_warn: ep <warning: unsupported format string argument>x - rounding interval to 131 microframes, ep desc says 32 microframes
+[init -> usb_drv] dev_info: A C
+[init -> usb_drv] dev_info: input: USB HID v1.00 Keyboard [A C] on usb-dummy-14.4.1/input0
+[init -> usb_drv] dev_info: A C
+[init -> usb_drv] dev_info: input: USB HID v1.00 Mouse [A C] on usb-dummy-14.4.1/input1
 Expect: 'interact' received 'strg+c' and was cancelled
```

alex-ab, 2015-06-03:
I think the ALL rule will take somehow the "deny - return" of the commit, which it shouldn't - just a guess.

alex-ab, 2015-06-03:
This would explain that the acpi_drv can't rewrite the IRQs since it does not get access to the devices of the pci driver.

alex-ab, 2015-06-04:
Commit e2745aa19168db4ae85d1f6ff92b48ac4440f30a fixes the issue on my testmachine.

commit af29888a (2015-06-04): pci: fix 'ALL' policy rule handling

Related to #1541.

Fixes #1559
commit 8ccac9a3 (2015-06-09): pci: fix 'ALL' policy rule handling

Related to #1541.

Fixes #1559
closed 2015-06-09 by alex-ab