genodelabs/genode #1562: wifi_drv: remove need for writable file system

nfeske, 2015-06-03:
The current version of the wifi_drv generates a wpa_supplicant.conf file that is then processed by the WPA supplicant, which is part of the same program. This mechanism needs a writable file system, which is not intuitive as both the creator and the consumer of the wpa_supplicant.conf file are the same program. So the data could (and, in my opinion, should) be kept internal to the wifi_drv.

chelmuth, 2016-10-20:
In 7bdcb867802e29505c288a6b62b0a538b100fa1f I replaced the need for the ram_fs component by a VFS `<ram/>` node, which keeps the file local to the wifi_drv component. Does this solve the issue or is is meant as a hook for the renovation of the used wpa_supplicant interface (away from the conf file)?

cnuke, 2018-09-03:
Part of the update in 18.08.

closed 2018-09-03 by cnuke