genodelabs/genode #1564: Move file system helpers from fs_rom to os/include/file_system

ehmry, 2015-06-04:
I'm opening this as a seperate issue because I have a few branches based on splitting these utilities.

nfeske, 2015-06-09:
Thank you. I merged the patch to staging now. I hope you are fine with the abbreviated commit title. I shortened it to be compliant to git's 50-character convention.

ehmry, 2015-06-10:
@chelmuth I've fixed the 'string_contains' namespacing issue in https://github.com/ehmry/genode/commit/5d56eb83ca467e5b4001b4b47d3feecd607aa6d6.

chelmuth, 2015-06-10:
@ehmry thanks for dealing with the compilation issue that quickly. Unfortunately, I'd prefer to restrict the availability of `string_contains()` to the `File_system` namespace. The reason is that it's not needed by any `base` code and, therefore, does not belong into `base`. Would you mind to fix your fixup accordingly?

ehmry, 2015-06-10:
Ok, string_contains in now in the File_system namespace, and I sqashed that back to the fs_rom commit.

chelmuth, 2015-06-10:
Okay, I removed the old commit from staging, added 895ce50 and _force_ pushed. Please remember that it's easier for our work flow if you provide _fixup_ commits to your work-in-progress. Then we are able to gently push, which results in fast forwards to others merging their work into staging.

ehmry, 2015-06-10:
Sorry about that, I should have payed better attention to the issue history.

commit d08b9588 (2015-06-09): move fs utils from fs_rom to file_system/util.h

Fixes #1564
commit 3371d270 (2015-06-10): move fs utils from fs_rom to file_system/util.h

Fixes #1564
commit 895ce504 (2015-06-10): unify file system utilities to os/include/file_system

Fixes #1564
commit b3142e0b (2015-06-17): unify file system utilities to os/include/file_system

Fixes #1564
closed 2015-06-22 by chelmuth