genodelabs/genode #1566: nova: propagate write-combining policy to page attributes

nfeske, 2015-06-05:
The current version of the NOVA kernel provides no interface to tailor the page-attribute bits of memory mappings according to a user-level policy. This lacking feature severely impacts the interactive performance because the framebuffer is accessed as regular I/O memory resource via unbuffered bus transactions instead of write-combined burst transactions. To solve this issue, we need two things:

* NOVA must define a page attribute table (by writing to the MSR 0x277) with an entry that selects write combining.
* NOVA has to provide an interface to allow the user land to pass the write-combining policy to the kernel. For example, via a corresponding bit of a map item, similar to the DMA bit.

The second part raises the question of how to maintain the consistence of the caching policy across all mappings of a physical page. One possible solution would be to allow the specification of the write-combining policy only for mappings from the physical address space. This way, only core had the ability to define the bit. All subsequent mappings would inherit the bit.

nfeske, 2015-06-05:
Commit a9193d50b0ec9e8f2888817d5531172eca12c3df illustrates the issue. The fb_bench program measures the throughput when writing to the frame buffer. On NOVA, we get about 50 MiB/second on the Lenovo x201 (as used for Issue #1552). On Fiasco.OC, where the write-combining policy is propagated to the kernel, we get about 2 GiB/second.

nfeske, 2015-06-05:
@chelmuth came up with the following proof-of-concept patch for NOVA that hard-wires the page attributes for all mappings of the Lenovo x201 framebuffer.

```
diff --git a/include/hpt.hpp b/include/hpt.hpp
index 786438f..853807f 100644
--- a/include/hpt.hpp
+++ b/include/hpt.hpp
@@ -47,11 +47,14 @@ class Hpt : public Pte<Hpt, mword, PTE_LEV, PTE_BPL, false>
             HPT_P   = 1UL << 0,
             HPT_W   = 1UL << 1,
             HPT_U   = 1UL << 2,
+            HPT_PWT = 1UL << 3,
+            HPT_PCD = 1UL << 4,
             HPT_UC  = 1UL << 4,
             HPT_A   = 1UL << 5,
             HPT_D   = 1UL << 6,
             HPT_S   = 1UL << 7,
             HPT_G   = 1UL << 8,
+            HPT_PAT = 1UL << 12,
             HPT_NX  = 0,
 
             PTE_P   = HPT_P,
diff --git a/src/cpu.cpp b/src/cpu.cpp
index 40dd1e9..d405b5f 100644
--- a/src/cpu.cpp
+++ b/src/cpu.cpp
@@ -210,6 +210,13 @@ void Cpu::init()
     if (EXPECT_TRUE (feature (FEAT_SMEP)))
         set_cr4 (get_cr4() | Cpu::CR4_SMEP);
 
+    uint32 cr_pat = Msr::read<uint32>(Msr::IA32_CR_PAT) & 0xffff00ff;
+
+    cr_pat |= 1 << 8;
+    Msr::write<uint32>(Msr::IA32_CR_PAT, cr_pat);
+    cr_pat = Msr::read<uint32>(Msr::IA32_CR_PAT);
+    trace(0, "PAT: %08x", cr_pat);
+
     Vmcs::init();
     Vmcb::init();
 
diff --git a/src/space_mem.cpp b/src/space_mem.cpp
index 95f585f..d9d2f31 100644
--- a/src/space_mem.cpp
+++ b/src/space_mem.cpp
@@ -74,6 +74,8 @@ void Space_mem::update (Mdb *mdb, mword r)
         return;
 
     mword ord = min (o, Hpt::ord);
+    if (p >= 0xd0000000 && p < 0xe0000000)
+    	a |= Hpt::HPT_PWT;
     for (unsigned long i = 0; i < 1UL << (o - ord); i++)
         hpt.update (b + i * (1UL << (ord + PAGE_BITS)), ord, p + i * (1UL << (ord + PAGE_BITS)), Hpt::hw_attr (a), r ? Hpt::TYPE_DN : Hpt::TYPE_UP);
 
```

alex-ab, 2015-08-26:
Based on @chelmuth work (thx!) I extended the nova kernel accordingly.

commit d420ef60 (2015-08-27): nova: support write combining

Fixes #1566
commit 0f2c2a67 (2015-08-31): nova: support write combining

Fixes #1566
closed 2015-08-31 by alex-ab