genodelabs/genode #1568: hw: capability leak during thread construction/destruction

skalk, 2015-06-08:
When testing the bomb scenario, I experienced the leak of exactly one capability every time one bomb child was constructed and destructed again. As it turned out the leak has to do with thread construction. Any thread gets its UTCB dataspace capability as well as its parent capability transfered within its UTCB when getting started, although both capabilities are only used by the main thread.

When a capability hits the PD that is currently not part of it, a corresponding 'Object_identity_reference' is allocated for the PD within the kernel. The kernel waits for a hint by the PD to delete it, or otherwise keeps it until the PD gets destroyed. Normally when a capability enters the PD some thread received it via IPC in its UTCB. When the thread takes the capability out of the UTCB its lifetime gets recorded. This was not done for both mentioned capabilities (UTCB dataspace and parent capability). The parent capability was within the PD anyway, but the UTCB dataspace capability got already destroyed after successful attachment, therefore it entered the PD again and is constructed again. Due to the fact that the thread ignores that capability within the UTCB its lifetime gets not recorded and no hint is given to the kernel that this capability is not used anymore. Therefore this capability gets leaked, which is a problem for components with longer lifetime that repeatedly construct and destruct threads.

skalk, 2015-06-10:
@m-stein can you review above commit please?

m-stein, 2015-06-10:
Sorry for the late reaction. I'm fine with the commit.

commit ea599346 (2015-06-10): hw: seal capability leak (fix #1568)

Do not initially transfer UTCB dataspace and parent capability to a thread
that is no main thread.
commit 4858a424 (2015-06-10): hw: seal capability leak (fix #1568)

Do not initially transfer UTCB dataspace and parent capability to a thread
that is no main thread.
commit d2039028 (2015-06-10): hw: seal capability leak (fix #1568)

Do not initially transfer UTCB dataspace and parent capability to a thread
that is no main thread.
commit 51829a52 (2015-06-10): hw: seal capability leak (fix #1568)

Do not initially transfer UTCB dataspace and parent capability to a thread
that is no main thread.
closed 2015-06-22 by skalk