genodelabs/genode #1570: init: children with excessive priority levels fail to start

ehmry, 2015-06-08:
Setting a child priority level below the number of priority levels causes a thread spawn failure in the kernel, atleast on NOVA.

nfeske, 2015-06-22:
When looking at the patch, I wonder if there is an off-by-one bug. If the ```prio_levels``` value is 4, the valid range of ```priority``` values if 0...3. Your code clamps the values to be lower than 4. But it should actually clamp the values to be lower than 3.

nfeske, 2015-06-22:
Given your issue description, I was a bit worried about the kernel or core getting into trouble caused by the out-of-range values. This is fortunately not the case (I have just experimented a bit). But the symptom of failure is hard to track down to the misconfiguration of the priority value. I think that your patch, with the off-by-one problem addressed, would be a good improvement.

ehmry, 2015-06-25:
@nfeske You were right that a child prority level needs to be clamped below the number of levels, but you didn't catch that I made the more serious error of clamping to the log base two of priority levels, rather than the value written on the config node.

ehmry, 2015-06-25:
When I went back I also noticed that the test at https://github.com/ehmry/genode/blob/363fd6065d6e1ce8ed104ca68ce5d323d7def0a1/repos/os/src/init/main.cc#L38 is effectively (prio_levels && prio_levels) != (1 << log2(... not prio_levels && (prio_levels != (1 << log2(..., which you can test by setting an invalid error and looking for the error message.

It would be nice if invalid priorities could be caught at thread creation but I don't have AMT so I like init to handle these things gracefully when I boot :)

commit fe7ce075 (2015-07-07): init: child priorities shall not excede the lowest priority level

Fixes #1570
commit c72c9924 (2015-07-21): init: prios shall not exceed the lowest prio level

Fixes #1570
closed 2015-07-21 by nfeske