genodelabs/genode #1575: VirtualBox: allocate guest memory in chunks

nfeske, 2015-06-11:
The current version of VirtualBox allocates the entire guest memory as a single dataspace from core's RAM service. If equipping the VM of the Turmvilla scenario if issue #1552 with 2GiB, core happens to not have a contiguous dataspace of this size available, due to memory fragmentation. Even though VirtualBox has enough quota, the RAM allocation fails.

For VirtualBox, there is actually no requirement to back the guest memory with physically contiguous RAM. To alleviate the RAM fragmentation issue, it should weaken the allocation constraint by performing multiple allocations using, let's say, 16 MiB chunks. VirtualBox can map those chunks virtually contiguous using a managed dataspace. We already use the same technique with L4Linux.

alex-ab, 2015-06-11:
This assumes that we never use vbox with directly assigned host devices (a) doing DMA on systems without iommu  support (b). Currently we don't support (a) but if we would do - it depends on (b) whether this will still work.

nfeske, 2015-06-11:
@alex-ab Thanks for chiming in! Do you think it would be better to allocate the guest memory from the PCI driver then? The PCI driver could actually allocate chunks and map them sequentially within the device PD. Another (NOVA-specific) solution would be to associate the PCI devices directly to the VM (not the device PD).

chelmuth, 2015-06-11:
I doubt VirtualBox would be able to support (a) without (b).

nfeske, 2015-06-11:
@alex-ab I just noticed that I missed your remark about (b). I doubt that we will ever need to directly assign devices to VMs on non-IOMMU systems.

alex-ab, 2015-06-11:
Just wanted to point out, that if (a) gets a must used feature, the underlying kernel and hardware _must_ support a iommu with this change (up to now valid for one kernel running on Intel CPUs). Don't get me wrong, I support this change request! The standard for other potential base platforms using this vbox port becomes just a bit higher - in principal - sel4, hw-x86 or foc are candidates - in principal ;-)

nfeske, 2015-08-28:
@alex-ab Thank you very much for taking on this issue! Looking at your patch, I'm wondering, doesn't this approach cause a very high number of RM connections to be opened?

alex-ab, 2015-08-31:
9 rm_connections for virtualbox.run or vbox_auto_win8.run

nfeske, 2015-08-31:
Considering that a VM is quite large anyway (in terms of memory consumption), it does not seem worth to optimize those 9 RM session away. Should the need to conserve those session quotas arise, we can still optimize.

commit b6c44085 (2015-08-31): vbox: allocate guest memory in chunks

Fixes #1575
closed 2015-09-09 by alex-ab