genodelabs/genode #1576: os/part_blk.run test broken

ValiValpas, 2015-06-12:
...because sfdisk fails to set up a partition table with the provided input. This is due to the fact that the second partition (start 4096, size 16386) overlaps with the first partition (start 2048, size 4096). 

I'd propose to change it as follows:
```diff
--- a/repos/os/run/part_blk.run
+++ b/repos/os/run/part_blk.run
@@ -27,7 +27,7 @@ if { [file exists bin/ata.raw] == 0 } then {
 
        # create two primary partitions (one is extented) and two logical paritions
        puts "using sfdisk to partition disk image, requires root privileges"
-       catch { exec echo "2048,4096,c\n4096,16386,5\n0,0\n0,0\n6144,4096,c\n12288,8192,c\n" | $sfdisk -uS -f bin/ata.raw }
+       catch { exec echo "2048,2048,c\n4096,16386,5\n0,0\n0,0\n6144,4096,c\n12288,8192,c\n" | $sfdisk -uS -f bin/ata.raw }
 }
 
 create_boot_directory
```

chelmuth, 2015-06-12:
Thanks for reporting the issue but doesn't the table also exceeds the files size?

ValiValpas, 2015-06-12:
Haven't checked that actually, but you're right. The test was working though.

chelmuth, 2015-06-12:
IMO sfdisk is like reading cipher text. We clean this up and use fallocate and parted as in [`tool/create_grub2`](https://github.com/genodelabs/genode/blob/15.05/tool/create_grub2).

commit ff5c7a4d (2017-05-13): Use parted in part_blk test

Fixes #1576
closed 2017-05-31 by chelmuth