genodelabs/genode #1584: x86: add framebuffer resolutions

NobodyIII, 2015-06-15:
The vesa_fb driver lacks support for non-BIOS display resolutions. My monitor is 1920x1080, but none of the 3 computers I have used with it have support for this resolution in the BIOS. On XP (years ago), I had to use DTDCalculator to hack the settings to add a custom display resolution. Should we follow this route? Or should we make/port graphics drivers? In case it helps, I have an AMD Radeon HD 5450.

nfeske, 2015-06-16:
I have the same problem with Intel graphics on my Lenovo x201. The Linux KMS driver is able to set up the 1920x1200 mode of my monitor but VESA merely offers 1600x1200. I think there is no quick hack to overcome this problem. We ultimately need real graphics drivers on Genode instead of relying on VESA. I plan to explore the port of Intel's Linux KMS driver but I won't look into any other vendor.

NobodyIII, 2015-06-16:
Should we use the kernel drivers or the Xorg drivers? The Xorg drivers
might be easier to port, and possibly more complete as well.

nfeske, 2015-06-16:
My plan is to look into KMS as this is the code that is most commonly used and best maintained. As another benefit, when enabling KMS, we have a path to also port the in-kernel GPU drivers in the future. I developed a prototype of the latter some years ago: http://genode.org/documentation/release-notes/10.08#Gallium3D_and_Intel%27s_Graphics_Execution_Manager So it is definitely feasible to go this route.

NobodyIII, 2015-06-16:
What's my best bet on getting an AMD Radeon driver? I don't have experience
with porting or coding drivers. About the most I've really done in that
area is some AHCI debugging and driving an RGB LED with a Raspberry Pi.

nfeske, 2015-06-16:
I am afraid that in the near future, chances are very slim. Somebody would need to invest (significant) time to build up knowledge and experience with Radeon graphics hardware. I do not see a way around it.

For Intel graphics, the situation is actually the same. But we are willing to take the necessary steps because Intel graphics is the most widely used option and it is quite well documented. Furthermore, we don't need to start from scratch but can build upon our existing experience.

NobodyIII, 2015-06-16:
Is a DDE not feasible in this case?

nfeske, 2015-06-17:
Sure. Our Linux KMS driver will reside in the dde_linux repository and follow the DDE approach. But please consider that DDE does not save you from learning about how the hardware operates. Even though you won't need to understand all the tiny details and the meaning of each single register, a thorough understanding of the architecture and the operations is needed. Once you start debugging, you have to dig deep into the details anyway.

NobodyIII, 2015-06-17:
Okay. Tell me when the Intel KMS driver works, and at that point I might be
able to port the Radeon driver. I'm likely to need some help with this, but
I might be able to do it, and if I succeed, I'll probably learn a lot.

chelmuth, 2015-06-22:
Can we close this issue as N/A?

NobodyIII, 2015-06-22:
Sure.

closed 2015-06-22 by chelmuth