genodelabs/genode #1585: Synergy input server

ehmry, 2015-06-16:
http://synergy-project.org/

On a whim I thought I'd try and get a Synergy client working for Genode, and it turns out they have a small library for embedded clients that was trivial to hook into an input server. I find performance to be transparent both running natively and with QEMU.

The issue to be solved before it is done is getting the dimensions of the screen to report to the synergy server. The dimensions are required to connect to the synergy serevr, but the dimensions do not need to be known before the component can serve an input session. For now the resolution is set in the config, but it should be dynamic. I can open a session to nitpicker or framebuffer but I don't think I can try one and fall back to the other without extra threads, so I think the best thing to do for now is to specify in config which session to open.

Any suggestions on getting dimensions? Would there be any benefit to restricting the area that the client can move a pointer to?

chelmuth, 2015-06-22:
I have the feeling this feature could be better preserved in a feature repo of yours like @nfeske's [genode-world repo](https://github.com/nfeske/genode-world). What do you think?

ehmry, 2015-06-22:
I'm all for keeping things for clean, moved to feature repo
https://github.com/ehmry/genode-emery

closed 2015-06-22 by ehmry