genodelabs/genode #1588: hw: deadlock in core-paging

skalk, 2015-06-16:
Currently, a deadlock occurs when core's heap substitue, namely Core_mem_allocator, tries to establish a new mapping and thereby triggers that core's page allocator targets the Core_mem_allocator to gain more backing store.

m-stein, 2015-06-16:
Its cool to see the reduction in code complexity that this modification causes. I especially like that the early translations table stuff isn't anymore. Besides some minor comments I've added to the commit, I have no objections.

skalk, 2015-06-17:
@m-stein I've re-written the commit including using more consistent names instead of Page_allocator.

m-stein, 2015-06-17:
Besides the one comment I totally like the new version, especially of the allocator header. Much clearer. Thanks for taking the time to re-write it!

closed 2015-06-22 by skalk

skalk, 2015-06-26:
The solution provided in commit cc58b11 leads to a very large .bss segment for core, because the maximum page-table set for core is held within it. This leads to an unnecessary large binary size, which is unwanted on especially tiny embedded platforms. Therefore, a follow-up work is wanted that takes the memory for the page-tables of core from anonymous memory instead of the .bss segment.

skalk, 2015-06-26:
@nfeske, @chelmuth please have a look at the upper 4 commits, they solve the memory problems you have criticed and also solves current issues with Rasperry Pi detected by the nightly buildbot tests.

m-stein, 2015-06-27:
Despite the minor issues I've commented in the commits, this looks cool to me.

skalk, 2015-06-29:
@nfeske, @chelmuth: commits 86843c3df9fbd334386a63d6b541cc460c141bc5 cb7e5e007a7680241173975403eabc0261735dd7 f0d095c2a01596a6e7170dc5a0c94952149c9eb7 and 5a26ac16c4f7044a6377922f5c385c1d3582adf9 are the overworked variants to solve the binary bloat and current issues with Raspberry Pi after review by m-stein

m-stein, 2015-06-29:
@skalk Thanks for taking the time to apply my suggestions :)

commit 0d0c9d2f (2015-06-18): hw: replace page table allocator with static one

Instead of organizing page tables within slab blocks and allocating such
blocks dynamically on demand, replace the page table allocator with a
simple, static alternative. The new page table allocator is dimensioned
at compile-time. When a PD runs out of page-tables, we simply flush its
current mappings, and re-use the freed tables. The only exception is
core/kernel that should not produce any page faults. Thereby it has to
be ensured that core has enough page tables to populate it's virtual
memory.

A positive side-effect of this static approac [... cut]