genodelabs/genode #1594: usb_drv: Does not always come up

ssumpf, 2015-06-18:
When there is a lot of load or the 'platform_drv' has a lowr priority, we examine timeouts on device enumeration.

alex-ab, 2015-06-18:
Letting run the platform_driver at a lower priority band than a driver is IMO a bug by the scenario integrator in principal ...

chelmuth, 2015-06-18:
My scenario put platform_drv *and* usb_drv at -1 while other workload used -2 and core stayed at 0.

alex-ab, 2015-06-18:
This does not help if the driver does some busy stuff (which it did before the commit) and the platform driver has to wait until it gets scheduled to deliver the signal for the irq (which would stop the driver doing busy stuff). With the commit we decrease the probability (which is good), but it may happen again. What we actually want to, is that if Genode signals because of irqs are sent immediately. From my perspective the safest way is to set the platform driver priority wise higher than a device driver.

chelmuth, 2015-06-18:
To sum it up: platform_drv should use (at least) prio+1 from any device driver? So, the quick fix is also part of the solution.

nfeske, 2015-06-18:
Please let us distinguish two situations:

1. When we are running our regular tests, we want to immediately capture problems with spinning device drivers, which is an unexpected behavior that must be fixed. For this reason, I would deliberately run the platform driver at a *lower* priority than the drivers because this will make the driver bugs immediately visible to us developers.

2. When deploying a Genode system, no driver must be able to starve the platform driver (and thereby all other drivers). So the platform driver should be prioritized higher or equal than the drivers.

commit 1888efad (2015-06-18): usb_drv: Remove signaling in wait_event_timeout

With the server framework this becomes unnecessary. Also when the 'platform_drv'
has a lower priority, signaling will cause a constant load that starves the
'platform_drv'.

Fixes #1594
closed 2015-06-22 by ssumpf