genodelabs/genode #1596: foc: RaspberryPI support

rmillo, 2015-06-21:
RaspberryPi platform support on Genode using Fiasco.OC as microkernel.

chelmuth, 2015-06-30:
@rmillo I also tried to rebase your work on staging (as with #1597) and ended up with
```
    LINK     core
ipc_pager.o: In function `Genode::Ipc_pager::_parse_msg_type()':
/plain/krishna/src/genode/genode_staging.git/repos/base-foc/src/core/ipc_pager.cc:53: undefined reference to `Genode::Ipc_pager::_parse_exception()'
pager.o: In function `Genode::Pager_activation_base::entry()':
/plain/krishna/src/genode/genode_staging.git/repos/base-foc/src/core/pager.cc:115: undefined reference to `Genode::Ipc_pager::set_regs(Genode::Thread_state)'
/plain/krishna/src/genode/genode_staging.git/repos/base-foc/src/core/pager.cc:131: undefined reference to `Genode::Ipc_pager::get_regs(Genode::Thread_state*)'
/plain/krishna/src/genode/genode_staging.git/repos/base-foc/src/core/pager.cc:76: undefined reference to `Genode::Ipc_pager::get_regs(Genode::Thread_state*)'
collect2: error: ld returned 1 exit status
```
Chances are high that this is an artifact induced by our current developments. Could you please have a look?

chelmuth, 2015-06-30:
I forgot: The tree is https://github.com/chelmuth/genode/tree/issue1596

rmillo, 2015-07-01:
@chelmuth, I have reviewed and tested the examples and works fine. I have noted that i have created the branch from master branch, but have  tried to merge with the stagging branch, this is my error. It should be merged with the master branch like #1597.

chelmuth, 2015-07-01:
@rmillo In addition to my explanation in #1608 I'd like to refer you to [Genode Foundations](http://genode.org/documentation/genode-foundations-15-05.pdf) Section 5.5 Git flow, which describes our development work flow in much more detail.

chelmuth, 2015-07-01:
https://github.com/chelmuth/genode/tree/issue1596 is now rebased on genodelabs/master.

rmillo, 2015-07-01:
@chelmuth I have updated from master branch and fixed the build configuration (84f4cbe492fb013e0d818b0e16755f723ddb63ac)

chelmuth, 2015-07-01:
I gave your merge a spin: It build flawlessly and I put the commits in our staging branch. Unfortunately, I cannot add a builder for foc_rpi to our nightly build remotely, so we'll have to wait until tomorrow night if the buildbot is happy.

commit b0baece4 (2015-07-01): base-foc: support for Raspberry Pi

Fixes #1596
commit c60d6515 (2015-07-06): base-foc: support for Raspberry Pi

Fixes #1596
closed 2015-07-07 by chelmuth