genodelabs/genode #1597: foc: Odroid-X2 support

rmillo, 2015-06-21:
Odroid-X2 (Exynos4412) platform support on Genode using Fiasco.OC as microkernel

rmillo, 2015-06-26:
Based on Arndale support on Genode and Exynos4412 datasheet, we have supported the Odroid-X2 on base-foc (5bd7464622102e79da13050839773a6c61ea84f0).
We have tested succesfully some basic GenodeOS examples:

* affinity
* affinity_subspace
* blk
* blk_cache
* cap_integrity
* config_args
* ds_ownership
* dynamic_config_loader
* dynamic_config_slave
* failsafe
* fpu
* kdb_uart_drv
* mp_server
* new_delete
* part_blk
* printf
* ram_fs_chunk
* resource_request
* resource_yield
* rm_fault
* rom_blk
* signal
* sub_rm
* synced_interface
* tar_rom
* thread
* thread_join
* timed_semaphore
* timer
* trace
* util_mmio
* volatile_object
* weak_ptr
* xml_generator

Now we are working on some device drivers on the Odroid-X2.

chelmuth, 2015-06-30:
@rmillo I tried my luck to rebase your work on top of our current staging branch (see https://github.com/chelmuth/genode/tree/issue1597). Unfortunately, I get interactively prompted in the Fiasco.OC configuration step after `./tool/create_builddir foc_odroid_x2; make -C build/foc_odroid_x2 run/timer` like follows
```
Platform Selection
  1. Beagleboard (PLATFORM_TYPE_beagleboard)
> 2. Samsung Exynos4 (PLATFORM_TYPE_exynos4)
  3. Samsung Exynos5 (PLATFORM_TYPE_exynos5)
  4. Freescale i.MX21 (PLATFORM_TYPE_imx21)
  5. Freescale i.MX35 (PLATFORM_TYPE_imx35)
  6. Freescale i.MX51 (PLATFORM_TYPE_imx51)
  7. Freescale i.MX53 (PLATFORM_TYPE_imx53) (NEW)
  8. Freescale i.MX6 (PLATFORM_TYPE_imx6)
  9. ARM Integrator (PLATFORM_TYPE_integrator)
  10. Marvell Kirkwood (PLATFORM_TYPE_kirkwood)
  11. TI OMAP AM33xx (PLATFORM_TYPE_omap3_am33xx)
  12. TI OMAP3EVM (PLATFORM_TYPE_omap3evm)
  13. Pandaboard (PLATFORM_TYPE_pandaboard)
  14. Raspberry Pi, Model A (PLATFORM_TYPE_rpi_a)
  15. Raspberry Pi, Model B (PLATFORM_TYPE_rpi_b)
  16. ARM Realview (Generic) (PLATFORM_TYPE_rv)
  17. ARM Realview PBX (PLATFORM_TYPE_rv_pbx)
  18. ARM Versatile Express (PLATFORM_TYPE_rv_vexpress)
  19. ARM Versatile Express (PLATFORM_TYPE_rv_vexpress_a15)
  20. Nvidia Tegra 2 (PLATFORM_TYPE_tegra2)
  21. Nvidia Tegra 3 (PLATFORM_TYPE_tegra3)
  22. Custom platform (PLATFORM_TYPE_custom)
choice[1-22]: 
```
Could you help me to prevent this as our nightly (unattended) build would choke on it?

chelmuth, 2015-07-01:
https://github.com/chelmuth/genode/tree/issue1597 is now rebased on genodelabs/master.

rmillo, 2015-07-07:
@chelmuth I have updated the Odroid-X2 build configuration for Fiasco.OC

chelmuth, 2015-07-07:
@rmillo I had a look at your sources and like to merge them to staging. Would you mind if I squash the individual commits into one _foc: Odroid-X2 basic support_ commit on top of staging?

rmillo, 2015-07-07:
@chelmuth This sounds good.
Now we are working on USB device driver support on Odroid-X2. Should we use the same issue and branch or a new one?

chelmuth, 2015-07-07:
The USB driver work sounds exciting, I'm looking forward it. Regarding the development work flow, I suggest you use a new branch and submit your result in a new issue. This one will be close (automatically) when the Odroid-x2 commit enters genodelabs/master once it mastered its path through staging.

commit 811f504f (2015-07-07): foc: Odroid-X2 basic support

Fix #1597
commit c14fe7e6 (2015-07-21): foc: Odroid-X2 basic support

Fix #1597
closed 2015-07-21 by nfeske