genodelabs/genode #1601: nova: handle out of memory cases of kernel

alex-ab, 2015-06-24:
On Genode/NOVA a process may trigger arbitrary kernel memory consumption during RPC by delegating memory or capabilities. If the kernel memory gets depleted, the kernel panics with a "Out of memory" message and  the overall Genode scenario stops, since the kernel is in-responsive to all processes.

The consumption can be in principal triggered by any "deliberately misbehaving" respectively greedy process, but also will happen during normal day-to-day usage of a developer, e.g. compiling and linking code in Genode's Noux environment. Our nightly test compiling Genode's tool chain in Noux on Genode/Nova already triggers this from time to time. Especially in the context of #1552 this behaviour becomes a show stopper.

Increasing the kernel memory, as currently done as workaround, is sufficient for a given, kernel memory usage wise - fixed and static -  Genode scenario, but does not help for #1552 or greedy Genode's application.

To improve the current situation, where the kernel memory usage is fixed, following steps are undertaken and investigated:
* account kernel memory usage per process
* limit kernel memory usage per process
* detection that a process hits the configured kernel memory limit
* ability to notify somebody (e.g. Genode's core) about the situation
* ability to react to such a notification, e.g.
** either stop the thread of the process or
** transfer kernel memory quota between processes (upgrading limit if desired) or
** free up kernel memory if possible, e.g. revoke memory delegations which can be re-created

With the steps above it should be possible to prevent that the whole kernel goes out of service, instead the greedy process will do so. With the indicated possible reactions we should be able to come up with some flexible policy defined by user-land (e.g. core).

A further improvement could be to go more into the base-hw and seL4 direction, where kernel memory consumption is handled more explicit, which is another topic for another issue.

nfeske, 2015-06-24:
Thanks for the excellent description of this line of work!

alex-ab, 2015-07-22:
The genode topic branch [issue_1601](https://github.com/alex-ab/genode/tree/issue_1601) contains the current state and uses the kernel branch [oom_r9](https://github.com/alex-ab/NOVA/commits/r9_oom) / [oom_r9_debug](https://github.com/alex-ab/NOVA/commits/r9_oom_debug] . The heavy tests seem to work well (bomb, platform, thread and noux_tool_chain) which caused until now the kernel into oom panic - if used the right way maliciously ;-). There are still smaller issues to solve, but the current state should already do well.

nfeske, 2015-07-22:
I just skimmed over the patches. Very nice! I would like to merge all the commits (except for the DEBUG commit) to staging to see if our autopilot is happy. Additionally, I would like to go through the patch https://github.com/alex-ab/genode/commit/d965b11366fd3625f742996d9d55038b015ee84f together with you to get a profound understanding about the mechanism (and possibly document it in the "Under the hood" Chapter of our book).

alex-ab, 2015-07-31:
Updated the current version. The major change is that the cross core ipc case is handled, all the oom handling code in the kernel went to a separate cpp file, vbox as seoul is running again and I tried to apply to my best @nfeske comments. Let's see what the buildbot things about it over the weekend.

commit 3e0fdb69 (2015-08-11): nova: use kernel branch with quota handling

Fixes #1601
commit df662cc2 (2015-08-21): nova: use kernel branch with quota handling

Fixes #1601
closed 2015-08-21 by alex-ab