genodelabs/genode #1605: abort() of main thread should notify parent

ehmry, 2015-06-28:
I'm not sure how much of a hack this is, but I've found the simpliest way to send a signal from abort() is to cause a unresolvable page fault and let the RM service do the rest.

chelmuth, 2015-06-29:
I have two questions

1. What is the rationale behind the limitation of exit signalling to the main thread?
2. Would it be an option to use `Parent_session::exit()` instead of the magical page fault?

ehmry, 2015-06-29:
My rationale to signalling on the main thread is that a program might gracefully handle the abort, and if not the error might eventually propragate to the main thread, if you think any thread should signal, I have no disagreement with that.

I used the magical fault because I figured I should signal with the CPU session exception handler, but I couldn't find a good route and took a shortcut.
I've tested ```env()->parent()->exit(1)``` and I think it works much better.

ehmry, 2015-07-01:
Oops, I didn't realise I didn't change the commit message.

chelmuth, 2015-07-01:
Never mind, I also realized it only after staging was pushed ;-)

commit f29e4e81 (2015-07-01): cxx: abort() triggers page fault when invoked by main threads

Fixes #1605
commit 46b7227e (2015-07-01): cxx: notify parent on abort() (in main thread)

Fixes #1605
closed 2015-07-07 by chelmuth