genodelabs/genode #1607: base: Locked_ptr to weak pointer object hangs infinitely

skalk, 2015-06-30:
When a Locked_ptr is taken to a weak object that is currently within destruction, the thread that tries to obtain the locked object hangs infinitely.

skalk, 2015-06-30:
Above test routine can be used to reproduce the problem.

skalk, 2015-07-02:
@nfeske: I've re-worked the test routine f2e42f6, and provided a fix 5a27b3a

nfeske, 2015-07-02:
Very good! Thanks for discovering, investigating and solving this tricky issue.

skalk, 2015-07-06:
As it turned out, my proposed solution solves the deadlock problem, but introduces a new race condition. Therefore, I've remove the fixed label. Please remove the current half-solution from staging again.

chelmuth, 2015-07-06:
For the record, I removed the proposed fix just now, but kept the extended test.

skalk, 2015-07-07:
@nfeske @chelmuth : I have tried to implement the solution we discussed intensively. Please review commit 3549613 attentively again. Thank you in advance.

alex-ab, 2015-07-07:
Please don't forget to adjust the current user on master, e.g.  repos/base/src/core/include/address_space.h and  repos/base/src/core/include/trace/source_registry.h.

chelmuth, 2015-07-07:
I have no objections to merge this patch. @nfeske, what do you think?

skalk, 2015-07-07:
@alex-ab I did not forget it, I'm currently working on a separate solution where Weak_object<T> should already do that job for you

alex-ab, 2015-07-07:
@skalk: all right ;-)

nfeske, 2015-07-08:
I like the patch a lot. Thanks @skalk for the careful implementation of the idea.

skalk, 2015-07-08:
@chelmuth @nfeske : I've add a fixup 564b30d150187f5943b33d8f8a0af1c379bb9024 that detects missing "lock_for_destruction" calls in the weak object base class, and adds all such calls in the most specific classes that inherit from weak object.

closed 2015-07-21 by skalk

alex-ab, 2015-08-25:
@skalk: for hw we get warnings in the bomb case like:

```
[init -> bomb-master] [2] It's time to kill all my children...
[init -> bomb-master] virtual Bomb_child::~Bomb_child()
Weak object 177e090 not destructed properly there are still dangling pointers to it
cxa pure virtual function called, return addr is 70044314
```

It seems that lock_for_destruction is not issued in platform_pd for base-hw. (all other base-* platform have the call). So fixes b799075b328f42aa0c0f573ff9f033fefa68fabc the issue or do I oversee something ?

skalk, 2015-08-25:
@alex-ab thanks for noticing, nfeske also observed this before. I fear I've overlooked HW when adapting core for the new weak pointer usage.

alex-ab, 2015-08-25:
@nfeske, @chelmuth : please addb799075b328f42aa0c0f573ff9f033fefa68fabc  to staging

nfeske, 2015-08-25:
@alex-ab thanks for the fix. Added to staging.

skalk, 2015-08-25:
@alex-ab : so you're faster, my fixup commit 6b8be6e1a2e7a5a392045462bfadb635ab150b0b came right now

nfeske, 2015-08-25:
@skalk I just added your fixup as well. ;-)

alex-ab, 2015-08-25:
@nfeske  :-D - now we have the some content two times in platform_pd.[h|cc] - just take one

nfeske, 2015-08-25:
@alex-ab thanks for the notice. I wonder how cherry picking the second commit even succeeded. Anyway, I removed the second one. The fixup wouldn't have worked anyway because the original commit is on master already.

commit ee4a0a92 (2015-07-02): base: resolve dead-lock in weak pointer templates

This commit eliminates the mutual interlaced taking of destruction lock,
list lock and weak pointer locks that could lead to a dead-lock situation
when a lock pointer was tried to construct while a weak object is in
destruction progress. It eliminates the destruction lock completely.
Instead, all weak pointers are invalidated and dequeued at the very
beginning of the weak object's destruction. Moreover, the object is
marked as being in destruction progrss so that newly created weak pointer
detect this, and do not try to enqueue. The list opera [... cut]
commit c7032002 (2015-07-08): base: resolve dead-lock in weak pointer templates

This commit eliminates the mutual interlaced taking of destruction lock,
list lock and weak pointer locks that could lead to a dead-lock situation
when a lock pointer was tried to construct while a weak object is in
destruction progress.
Now, all weak pointers are invalidated and dequeued at the very
beginning of the weak object's destruction. Moreover, before a weak pointer
gets invalidated during destruction of a weak object, it gets dequeued, and
the list lock is freed again to avoid the former dead-lock.

Fix #1607
commit 0f05fa6f (2015-07-21): base: resolve dead lock in weak pointer templates

This commit eliminates the mutual interlaced taking of destruction lock,
list lock and weak pointer locks that could lead to a dead-lock situation
when a lock pointer was tried to construct while a weak object is in
destruction progress.
Now, all weak pointers are invalidated and dequeued at the very
beginning of the weak object's destruction. Moreover, before a weak pointer
gets invalidated during destruction of a weak object, it gets dequeued, and
the list lock is freed again to avoid the former dead-lock.

Fix #1607