genodelabs/genode #1614: make run/l4linux problem

thotheolh, 2015-07-02:
I received a 'cp: cannot stat `bin/fb_drv': No such file or directory' while trying to build the Fiasco.OC/L4Linux/x86_32 with version 15.05 of Genode.

    Program core/x86_32/core
        COMPILE  version.o
        LINK     core
    make[1]: Leaving directory `/home/thotheolh/Desktop/genode-15.05/build'
    genode build completed
    Download initramfs ...
    cp: cannot stat `bin/fb_drv': No such file or directory
        while executing
    "exec cp bin/$binary [run_dir]/genode"
        (procedure "copy_and_strip_binaries" line 7)
        invoked from within
    "copy_and_strip_binaries $binaries"
        (procedure "run_boot_dir_x86" line 7)
        invoked from within
    "run_boot_dir_x86 $binaries"
        (procedure "run_boot_dir" line 2)
        invoked from within
    "run_boot_dir $binaries"
        (procedure "build_boot_image" line 2)
        invoked from within
    "build_boot_image  [join $boot_modules " "]"
        (file "/home/thotheolh/Desktop/genode/repos/ports-foc/run/l4linux.run" line 146)
        invoked from within
    "source $include_name"
        ("foreach" body line 6)
        invoked from within
    "foreach include_name [get_cmd_arg --include ""] {
	    # first check if the include name is absolute
	    if {[string first "/" $include_name] == 0} {
		    puts ..."
        (file "/home/thotheolh/Desktop/genode/tool/run/run" line 636)
    make: *** [run/l4linux] Error 1

I found the subject (http://sourceforge.net/p/genode/mailman/genode-main/thread/CANjHPYU98mBWwMkDo9FDFc2QQOL3+7jFk4n4GTzbTt05NbYqEg@mail.gmail.com/) and applied the instructions from the issue:

    make -C libports prepare PKG=x86emu
    make: Entering directory `/home/thotheolh/Desktop/genode/repos/libports'
    x86emu  extract xorg-server-1.12.0.tar.bz2 (xserver)
    x86emu  install include/x86emu
    x86emu  install include/x86emu/x86emu
    make: Leaving directory `/home/thotheolh/Desktop/genode/repos/libports'

And also uncommented the "REPOSITORIES += $(GENODE_DIR)/repos/libports" in the build.conf.

I re-run the "make run/l4linux" and now a new error comes out:

      Program server/report_rom/report_rom
      Program core/x86_32/core
        COMPILE  version.o
        LINK     core
    make[1]: Leaving directory `/home/thotheolh/Desktop/genode-15.05/build'
    genode build completed
    cp: cannot stat `bin/nic_drv': No such file or directory
        while executing
    "exec cp bin/$binary [run_dir]/genode"
        (procedure "copy_and_strip_binaries" line 7)
        invoked from within
    "copy_and_strip_binaries $binaries"
        (procedure "run_boot_dir_x86" line 7)
        invoked from within
    "run_boot_dir_x86 $binaries"
        (procedure "run_boot_dir" line 2)
        invoked from within
    "run_boot_dir $binaries"
        (procedure "build_boot_image" line 2)
        invoked from within
    "build_boot_image  [join $boot_modules " "]"
        (file "/home/thotheolh/Desktop/genode/repos/ports-foc/run/l4linux.run" line 146)
        invoked from within
    "source $include_name"
        ("foreach" body line 6)
        invoked from within
    "foreach include_name [get_cmd_arg --include ""] {
	    # first check if the include name is absolute
	    if {[string first "/" $include_name] == 0} {
		    puts ..."
        (file "/home/thotheolh/Desktop/genode/tool/run/run" line 636)
    make: *** [run/l4linux] Error 1

What else am I still missing ?

chelmuth, 2015-07-02:
You're missing the nic_drv for x86, which is in the *dde_ipxe* repository. Please do `tool/ports/prepare_port dde_ipxe` and the *build.conf* step for this repository too.

thotheolh, 2015-07-03:
This time another new error appeared:

    spawn qemu-system-i386 -no-kvm -m 128 -nographic -serial mon:stdio -smp 2,cores=2 -net nic,model=e1000 -net user -cdrom var/run/l4linux.iso
    couldn't execute "qemu-system-i386": no such file or directory
        while executing
    "spawn qemu-system-i386 -no-kvm  -m 128 -nographic  -serial mon:stdio  -smp 2,cores=2  -net nic,model=e1000 -net user  -cdrom var/run/l4linux.iso"
        ("eval" body line 1)
        invoked from within
    "eval spawn $qemu $qemu_args"
        (procedure "run_power_on" line 70)
        invoked from within
    "run_power_on"
        (procedure "run_genode_until" line 13)
        invoked from within
    "run_genode_until {.*\/ #.*} 220"
        (file "/home/thotheolh/Desktop/genode/repos/ports-foc/run/l4linux.run" line 160)
        invoked from within
    "source $include_name"
        ("foreach" body line 6)
        invoked from within
    "foreach include_name [get_cmd_arg --include ""] {
	    # first check if the include name is absolute
	    if {[string first "/" $include_name] == 0} {
		    puts ..."
        (file "/home/thotheolh/Desktop/genode/tool/run/run" line 636)
    make: *** [run/l4linux] Error 1

Are the anything else I am missing. I am following the Fiasco build webpage on the Genode website (http://genode.org/documentation/platforms/foc) and following as closely as I can but apparently there things I have to do not found on the instruction page.

skalk, 2015-07-03:
> couldn't execute "qemu-system-i386": no such file or directory

It tries to execute the freshly built scenario on top of the QEMU emulator, which obviously is not installed in your system, or not found in your execution pathes. Just install QEMU on your system.

thotheolh, 2015-07-03:
How do I convert the build into a LiveCD and burnt into a CD/DVD instead of using QEMU ?

alex-ab, 2015-07-03:
`` spawn qemu-system-i386 -no-kvm -m 128 -nographic -serial mon:stdio -smp 2,cores=2 -net nic,model=e1000 -net user -cdrom var/run/l4linux.iso ```

The build result in your case is an iso file. You may use this iso to burn it to CD/DVD or using _dd_ to put it on a usb stick and boot it. In general, the results of the invocation of  a run script you find in `<genode-dir>/build/<platform>/var/run/<run-script-name>` and ISOs at `<genode-dir>/build/<platform>/var/run/<run-script-name>.iso`

thotheolh, 2015-07-03:
How do I use the above 'spawn' command ? Do I add them to certain script files ? I don't see the /build/platform>/var/run... but I see the /build/var/run... which am I missing something ?

alex-ab, 2015-07-03:
The spawn line is part of one of your previous posts and at the end of the line you can see the build relateive path to your wanted iso file: var/run/l4linux.iso. So, look into your build directory for the iso. If you created your build directory to /build it is /build/var/run/l4linux.iso. If you haven't specified a build directory during tool/create_builddir invocation, it is at the location as posted before.

thotheolh, 2015-07-03:
So all I need to do is burn the ISO into a CD/DVD for a LiveCD usage ?

alex-ab, 2015-07-03:
Exactly.

chelmuth, 2015-07-07:
@thotheolh may we close this issue?

thotheolh, 2015-07-08:
Yes you may close this thread.

closed 2015-07-08 by chelmuth

LastRitter, 2015-12-13:
I compile it ok by the step below:

./tool/ports/prepare_port dde_ipxe
./tool/ports/prepare_port nova
./tool/ports/prepare_port x86emu

cd repos/base-nova/
make prepare
# make prepare PKG=x86emu
cd -

./tool/create_builddir nova_x86_64
cd ./build/nova_x86_64

vi ./etc/build.conf
# add 
# MAKE += -j4
# 
# uncommenting
# REPOSITORIES += $(GENODE_DIR)/repos/libports
# REPOSITORIES += $(GENODE_DIR)/repos/ports
# REPOSITORIES += $(GENODE_DIR)/repos/gems

make run/demo

skalk, 2015-12-14:
@LastRitter I don't see the relation to this issue. Moreover, I don't understand what the problem is you wanna describe?