genodelabs/genode #1617: nova: core revokes (invalid) memory areas of dataspaces in invalidation

alex-ab, 2015-07-03:
In base-nova  _revoke_ram_ds(Dataspace *) causes local memory mapping to be vanished in core and frees up the virtual region of the local mapping. Additional the _core_local_addr member of the Dataspace gets set to 0, since the virtual region is already added back to the region allocator. Afterwards the destructor of the Dataspace is called, which looks up every rm_session where the dataspace is attached and detach itself. This leads to invocation of rc->unmap() calls which base address around 0. This lead to the effect that what ever in core is mapped there + the original dataspace size vanishes - which shouldn't.

This may lead (for me reproducible) to the phenomenon that a process, which had some core local mappings in the lower virtual region starts causing page faults and the pager resolves this successfully, since it assumes that the core local mapping is valid, which it isn't. Finally the process is in a page-fault live lock loop and seems to 'hang'.

alex-ab, 2015-07-03:
For me it looks like that the virtual region shouldn't be added back to the region_allocator (and therefore the core_local_addr pointer must not set to 0) in _revoke_ram_ds, but at a later point - when it is clear that all detach from all rm_session are done. Suggestions ?

nfeske, 2015-07-06:
For the record, in a short offline discussion between @alex-ab and me, we came to the conclusion that the missing code in ```Ram_session_component::_free_ds``` marked with XXX is the problem: https://github.com/genodelabs/genode/blob/master/repos/base/src/core/ram_session_component.cc#L48

alex-ab, 2015-07-07:
The commit  b4a0b2a51909cdc4572786b6b8ce1a8ca4a391d5 fixes this issue. The other topic I mentioned today morning (using weakobj and weakptr for rm_session<->dataspace which is loosely related gets not fixed, described by #549.

commit a19793cb (2015-07-07): detach ds from all rm sessions before destruction

Fixes #1617
commit c42e7703 (2015-07-21): detach ds from all rm sessions before destruction

Fixes #1617
closed 2015-07-21 by alex-ab