genodelabs/genode #1622: base: more appropriate allocator for IRQs

m-stein, 2015-07-08:


nfeske, 2015-07-08:
Could you please be a bit more elaborate about your motivation behind the issue? How is the current allocator inappropriate? ;-)

m-stein, 2015-07-08:
Both https://github.com/m-stein/genode/commit/5f3df51056884e9f7d8fc84cf49ffa28b1480c27 and https://github.com/m-stein/genode/commit/68bd600b0f6460fad15a2c91ecb0ac474391938a try to fit the IRQ type into the interface of the used ``Range_allocator``. In related discussions, we came to the conclusion that it would be cleaner to switch to another allocator instead. The Range_allocator implementation and interface are more generic then necessary for IRQ allocation. Thus, I now try to introduce an Irq_allocator that uses the Bit_allocator template as back end.

nfeske, 2015-07-08:
Thanks for the explanation!

nfeske, 2017-12-13:
This issue is stale for two years. @m-stein, do you still consider it as relevant, or may we close it?

m-stein, 2017-12-14:
It's still the same. But I've no problems with closing the issue.

closed 2017-12-14 by m-stein