genodelabs/genode #1623: Thread test fails on hw_arndale

chelmuth, 2015-07-08:
With the current staging branch (4b7b440a5ec969881715e1e3d20b225c36d0353b) `make run/thread` on hw_arndale fails like follows.
```
[init -> test-thread] Uncaught exception of type 'N6Genode10Rm_session17Invalid_dataspaceE' (use 'c++filt -t' to demangle)
[init -> test-thread] abort called - thread: 'helper'
```
The test works flawlessly on master (af5c03edaf3db9e8444cad9c7ede1dbb70cc1d1b). My bisect identified commit 0d6b26713696b1d375b2314a9a7fd4991b8ec64d "base: use proper Cpu_session when tracing threads" as the first one failing.

chelmuth, 2015-07-08:
@cnuke Are there hidden/not accounted allocations in the tracing framework?

cnuke, 2015-07-08:
Not that I know of. One issue with commit is https://github.com/genodelabs/genode/commit/0d6b26713696b1d375b2314a9a7fd4991b8ec64d#diff-575e5cd61108cfd019f67c1a91bfe7d9L193 though. We have to detach it after the platform thread has been destructed because will get a page fault on next RPC call otherwise.

chelmuth, 2015-07-08:
It seems the exception is thrown by Thread_base::Thread_base() very early in the log. How could that be?

chelmuth, 2015-08-11:
The issue still triggers each night. Any ideas?

chelmuth, 2015-09-30:
With current staging the issue is till there...

skalk, 2015-10-01:
maybe I'm blind, but both last nights it was working on current staging, wasn't it?

chelmuth, 2015-10-01:
I ran it manually yesterday and it got stuck...

skalk, 2015-10-01:
I see :(

chelmuth, 2015-10-07:
A glimpse into last nights buildbot results reveals that foc_x86_64 and hw_panda may suffer from a similar issue. The former also fails with `Uncaught exception of type 'N6Genode10Rm_session17Invalid_dataspaceE' (use 'c++filt -t' to demangle)`, while the latter just gets stuck.

chelmuth, 2015-11-05:
With help from @m-stein and @skalk I could hunt down the problem with 03b9075. The thread test triggers an issue in the base-hw kernel interface particularly in IPC with capability arguments: The helper threads created in the context-allocator stress test create many threads (that never start btw) concurrently. During the thread creation the `Thread_base` constructor requests the trace-control dataspace for the components CPU session, which is one and the same dataspace (i.e., dataspace capability) for all threads. With the high concurrency of the scenario, we end up with multiple copies of the dataspace capability ID in the UTCBs of the helper threads. At this stage, the `Native_capability` reference counting is not effective as the capability was not yet demarshalled. The cause of issue #1623 is now that one helper demarshals the trace-control dataspace capability but also proceeds further until it destructs the last `Native_capability` instance for the capability, the reference counter drops to 0, and `Kernel::delete_cap()` for this capability is called. In response. the kernel frees the object-ID reference, i.e., capability slot. The UTCB of a second helper still contains the capability ID replied to its request to get the trace-control dataspace, which is now stale and the disaster begins as the `Native_capability` reference counting becomes corrupted by demarshalling a stale capability ID and later deleting a still needed RAM dataspace capability. In the end, the RM session complains an attempt to attach an invalid RAM dataspace (see above).

What we need now is a robust handling of those *in-flight* capability IDs, which where copied into the UTCB of thread performing an IPC call but were not yet demarshalled (and therefore ref-counted by ?`Native_capability`).

My commit contains some instrumentation for debugging and also a log in log.txt.

chelmuth, 2015-12-02:
I merged f83dffb7397eab92a0c94bdaaa7b9883c05f1814 but insist on you to open the "optimization" issue for the evaluation of the UTCB-chain solution.

m-stein, 2015-12-03:
@skalk The solution looks pretty cool to me :)

skalk, 2015-12-03:
@chelmuth I would love to do that, but an "optimization" label is missing unfortunately ;-)

chelmuth, 2015-12-03:
@skalk I'm confident this will not keep you off from opening the issue anyway ;-)

commit 40812d20 (2015-12-02): hw: reference count capabilities in UTCBs

When capabilities are delegated to components, they are added to the UTCB of the
target thread. Before the thread is able to take out the capability id out of
the UTCB and adapt the user-level capability reference counter, it might happen
that another thread of the same component deletes the same capability because
its user-level reference counter reached zero. If the kernel then destroys the
capability, before the same capability id is taken out of all UTCBs, an
inconsitent view in the component is the result.  To keep an consistent view in
the multi [... cut]
commit cb0aaccc (2015-12-04): hw: reference count capabilities in UTCBs

When capabilities are delegated to components, they are added to the UTCB of the
target thread. Before the thread is able to take out the capability id out of
the UTCB and adapt the user-level capability reference counter, it might happen
that another thread of the same component deletes the same capability because
its user-level reference counter reached zero. If the kernel then destroys the
capability, before the same capability id is taken out of all UTCBs, an
inconsitent view in the component is the result.  To keep an consistent view in
the multi [... cut]
commit dc85fc12 (2015-12-07): hw: reference count capabilities in UTCBs

When capabilities are delegated to components, they are added to the UTCB of the
target thread. Before the thread is able to take out the capability id out of
the UTCB and adapt the user-level capability reference counter, it might happen
that another thread of the same component deletes the same capability because
its user-level reference counter reached zero. If the kernel then destroys the
capability, before the same capability id is taken out of all UTCBs, an
inconsitent view in the component is the result.  To keep an consistent view in
the multi [... cut]
commit 60ba210a (2015-12-10): hw: reference count capabilities in UTCBs

When capabilities are delegated to components, they are added to the UTCB of the
target thread. Before the thread is able to take out the capability id out of
the UTCB and adapt the user-level capability reference counter, it might happen
that another thread of the same component deletes the same capability because
its user-level reference counter reached zero. If the kernel then destroys the
capability, before the same capability id is taken out of all UTCBs, an
inconsitent view in the component is the result.  To keep an consistent view in
the multi [... cut]
closed 2015-12-10 by skalk