genodelabs/genode #1624: hw_x86_64: chain boot via bender results in reboot on Qemu

chelmuth, 2015-07-08:
With #1580 fixed I tried to boot hw_x86_64 on my Lenovo x201 with AMT SOL, which works after changing the port manually. The same binary results in instant reboot on Qemu.
```diff
diff --git a/tool/run/boot_dir/hw b/tool/run/boot_dir/hw
index 2abf560..6d5172a 100644
--- a/tool/run/boot_dir/hw
+++ b/tool/run/boot_dir/hw
@@ -166,7 +166,9 @@ proc run_boot_dir {binaries} {
                puts $fh "timeout 0"
                puts $fh "default 0"
                puts $fh "\ntitle Genode on base-hw"
-               puts $fh " kernel /image.elf"
+#              puts $fh " kernel /image.elf"
+               puts $fh " kernel /boot/bender"
+               puts $fh " module /image.elf"
                close $fh
        }
 
```
With the following patch it also reboots when pushing on the stack (as the call would also do later for the return address).
```diff
diff --git a/repos/base-hw/src/core/spec/x86_64/kernel/crt0.s b/repos/base-hw/src/core/spec/x86_64/ker
index 5dffb5a..7c3ff6b 100644
--- a/repos/base-hw/src/core/spec/x86_64/kernel/crt0.s
+++ b/repos/base-hw/src/core/spec/x86_64/kernel/crt0.s
@@ -91,6 +91,9 @@
        leaq _stack_high@GOTPCREL(%rip),%rax
        movq (%rax), %rsp
 
+       push %rax
+       47: jmp 47b
+
        /* uniprocessor kernel-initialization which activates multiprocessor */
        call init_kernel_up
 
```
Without bender it works.

Kensan, 2015-07-08:
I suspect that the image grows larger than the memory mappings of the initial page tables (16 MiB). While rebasing our hw_x86_64_muen-vbox work on the latest staging branch last night, I came across this issue and implemented a fix, see commit 90b02d9.
Could you give the commit a try and see if it also fixes this particular problem?

chelmuth, 2015-07-09:
@Kensan Unfortunately, it's not that easy. The test cases I used were _run/printf_ and _run/thread_ with about 2 MiB _image.elf_.

chelmuth, 2015-07-09:
Now I know for sure that the kernel triple-faults on the `push %rax`. Qemu logs the following CPU state
```
Triple fault
CPU Reset (CPU 0)
RAX=0000000000284f98 RBX=0000000000039400 RCX=00000000c0000080 RDX=0000000000000000
RSI=0000000000322110 RDI=0000000000315f60 RBP=000000000002c6e0 RSP=00b0248c8b000002
R8 =0000000000000000 R9 =0000000000000000 R10=0000000000000000 R11=0000000000000000
R12=0000000000000000 R13=0000000000000000 R14=0000000000000000 R15=0000000000000000
RIP=0000000000201019 RFL=00000046 [---Z-P-] CPL=0 II=0 A20=1 SMM=0 HLT=0
ES =0010 0000000000000000 00000000 00209300 DPL=0 DS16 [-WA]
CS =0008 0000000000000000 00000000 00209800 DPL=0 CS64 [---]
SS =0010 0000000000000000 00000000 00209300 DPL=0 DS16 [-WA]
DS =0010 0000000000000000 00000000 00209300 DPL=0 DS16 [-WA]
FS =0010 0000000000000000 00000000 00209300 DPL=0 DS16 [-WA]
GS =0010 0000000000000000 00000000 00209300 DPL=0 DS16 [-WA]
LDT=0000 0000000000000000 00000000 00008200 DPL=0 LDT
TR =0008 0000000000000580 00000067 00008900 DPL=0 TSS64-avl
GDT=     0000000000258e68 00000037
IDT=     0000000000000000 0000ffff
CR0=80010011 CR2=0000000000000080 CR3=0000000000279000 CR4=00000020
DR0=0000000000000000 DR1=0000000000000000 DR2=0000000000000000 DR3=0000000000000000 
DR6=00000000ffff0ff0 DR7=0000000000000400
CCS=0000000000000000 CCD=0000000000000000 CCO=SARL    
EFER=0000000000000d00
FCW=037f FSW=0000 [ST=0] FTW=00 MXCSR=00001f80
FPR0=0000000000000000 0000 FPR1=0000000000000000 0000
FPR2=0000000000000000 0000 FPR3=0000000000000000 0000
FPR4=0000000000000000 0000 FPR5=0000000000000000 0000
FPR6=0000000000000000 0000 FPR7=0000000000000000 0000
XMM00=00000000000000000000000000000000 XMM01=00000000000000000000000000000000
XMM02=00000000000000000000000000000000 XMM03=00000000000000000000000000000000
XMM04=00000000000000000000000000000000 XMM05=00000000000000000000000000000000
XMM06=00000000000000000000000000000000 XMM07=00000000000000000000000000000000
XMM08=00000000000000000000000000000000 XMM09=00000000000000000000000000000000
XMM10=00000000000000000000000000000000 XMM11=00000000000000000000000000000000
XMM12=00000000000000000000000000000000 XMM13=00000000000000000000000000000000
XMM14=00000000000000000000000000000000 XMM15=00000000000000000000000000000000
```
Any x86 experts around who can point on the bit that is wrong?

Kensan, 2015-07-09:
The stack pointer is way off: **RSP=00b0248c8b000002**. The correct value could possibly be **0x248c8b**. For some reason, the stack pointer is not properly set up. Can you inspect the binary and check what (quad-word) value is at **0x284f98**?

chelmuth, 2015-07-09:
@Kensan I had to dig deeply into bender but found the reason for the corrupted RSP - a corrupt kernel image due to module-ELF region overlap on ELF loading (TUD-OS/morbo#4).

commit 4c3c90e6 (2015-07-09): bender: unconditionally relocate all boot modules

If just one multiboot kernel module was loaded after bender, the
relocation was skipped before. This resulted in a corrupt binary image
on ELF loading if the regions of the boot module and the final program
overlap. Now, all modules are copied below 2 GiB (and out of the way)
before ELF loading.

Fixes #1624

Bender upstream issue is TUD-OS/morbo#4
commit bc827787 (2015-07-21): bender: unconditionally relocate all boot modules

If just one multiboot kernel module was loaded after bender, the
relocation was skipped before. This resulted in a corrupt binary image
on ELF loading if the regions of the boot module and the final program
overlap. Now, all modules are copied below 2 GiB (and out of the way)
before ELF loading.

Fixes #1624

Bender upstream issue is TUD-OS/morbo#4
closed 2015-07-21 by chelmuth