genodelabs/genode #1625: hw_x86_64: read serial I/O port from BDA

chelmuth, 2015-07-08:
On NOVA, the I/O port of the UART is retrieved from the BIOS data area, which is very handy when debugging several systems with standard ports, AMT SOL, or plugin PCI I/O cards. The port is written to the BDA by bender, which scans the PCI bus for an appropriate device. (discovered during debug of #1624)

alex-ab, 2015-07-09:
Please check if this is ok, it is based on #1403. @Kensan, @reet- please have also a look, whether this is ok.

reet-, 2015-07-09:
The code in `serial.h` looks fine to me.

Since we do not provide a BDA to subjects on Muen, we would just continue to use the previous `serial.h` with static port value.

The fixup commit code on the other hand looks a bit special, but it is marked as *fixme* so I assume it is only a temporary solution.

m-stein, 2015-07-10:
Added some commits via my https://github.com/m-stein/genode/tree/1625_hw_x86_64_read_serial_IO_port_from_BDA branch.

chelmuth, 2015-07-10:
Merged 39ce878 and 0e41b61.

m-stein, 2015-07-10:
7077b93e81702ffb423da412eb6ae6e6a69df28f?

chelmuth, 2015-07-10:
Please update staging, it was replaced by 7cff97e80e86d92487e6d7a507b26c4b9305ad6b.

m-stein, 2015-07-10:
Thank you for the hint :)

alex-ab, 2015-07-10:
Could we factor out the Bda class to some place, so that base-nova also can use it ? (Currently on staging there is the old do it by hand version: f7ea7b1506cf2da428fc17362cd3a2110e503347)

m-stein, 2015-07-10:
I think thats a good idea! Shall I fix it?

alex-ab, 2015-07-10:
That would be great.

m-stein, 2015-07-10:
There it is: e78dc3383f6a2adf0e0a9a2451b0d60e031ee75f.

alex-ab, 2015-07-11:
Looks good, thanks.

commit 49ac3e30 (2015-07-09): hw: use bios data area to detect com/uart ports

Fixes #1625
commit 73f7a956 (2015-07-21): hw: use bios data area to detect com/uart ports

Fixes #1625
closed 2015-07-21 by alex-ab