genodelabs/genode #1631: Library qt5_core depends on launchpad library

chelmuth, 2015-07-14:
I think the above-mentioned dependency is artificial and should be removed. Any opinions? @cproc? Currently, any developer using the qt5 library has to enable the _demo_ repository to build her project, which is counterintuitive at least.

cproc, 2015-07-14:
The launchpad library ist currently used for the QProcess implementation.

chelmuth, 2015-07-14:
Do you have a pointer for me?

cproc, 2015-07-14:
https://github.com/genodelabs/genode/blob/master/repos/libports/src/lib/qt5/patches/qt5_qtbase_genode.patch#L90
https://github.com/genodelabs/genode/blob/master/repos/libports/src/lib/qt5/qtbase/src/corelib/io/qprocess_genode.cpp

I can't remember exactly how it works, but after a quick look over the code again, it seems that one has to provide an XML configuration with a list of the programs that can be started together with their RAM quota. The QProcess user provides only the program name and the RAM quota is taken from the configuration and then both name and quota are given to the launchpad library to start the child.

chelmuth, 2015-07-14:
I see. Nevertheless, the feature is not used anywhere (in particular not by *qt_launchpad* and, therefore, untested. Do you think we may remove it? Or should we go for a complete review incl. test case?

chelmuth, 2015-09-30:
I vote for removal of the code (and thereby the artificial dependency).

cproc, 2016-11-08:
Fixup commit 4ab843d2326854150a59582ed8d31ef0b7578be0 updates the port hash.

@chelmuth: please add to staging

cproc, 2016-11-10:
Fixup commit 7a2a9f88e4240c6fa6396f3d293b248420e09baa disables the use of QProcess in Arora.

@chelmuth: please add to staging

commit c78a6e1f (2016-11-08): qt5: remove launchpad-based 'QProcess' implementation

Fixes #1631
commit adc916ce (2016-11-09): qt5: remove launchpad-based 'QProcess' implementation

Fixes #1631
commit fbc46eb3 (2016-11-25): qt5: remove launchpad-based 'QProcess' implementation

Fixes #1631
commit c0e0f287 (2016-11-25): qt5: remove launchpad-based 'QProcess' implementation

Fixes #1631
closed 2016-11-30 by chelmuth