genodelabs/genode #1633: Read-only dataspaces

ehmry, 2015-07-17:
This is not immediate need of mine, but I'd like to make read only dataspaces so that I can deduplicate the ROM memory in fs_rom and use it as a general purpose cache.

My solution was to ammend the RAM session interface to create a duplicate dataspace component with writable set to false. I've tested this on Fiasco, base-hw, and NOVA with a little help from Alex.

I think the safest thing to do with the first dataspace is to invalidate it before returning the second, rather than reference-count the undelying allocations.

Destroying the original dataspace in core does not appear to detach the first dataspaces attachments or invalidate the capability, but it does make the attachments read only. I'm worried that this is leaving some of the core components in a bad state.

I'm working on a test right now but I want to know if these changes look reasonable so far.

nfeske, 2015-07-20:
Thanks for opening this issue. It coincides with a plan of mine to extend the RAM session interface, but in a slightly more general way:

Given a RAM dataspace, I would like to have a way to obtain one or multiple "alias" dataspaces that refer to parts of the original dataspace. Such an alias could be restricted to be read only. This feature would enable us to implement (optional) allocation policies for physical memory outside of core. E,g., on a NUMA system, each NUMA domain could have a separate instance of a RAM service that performs RAM allocations from CPU-local memory. Each instance would obtain a big RAM dataspace from core's RAM service at its start time. Each time when a client performs an allocation, the CPU-local RAM service would create an alias for a portion of its big backing-store dataspace and hand out the dataspace capability of the alias. The operation would look like this:

```Dataspace_capability alias(Dataspace_capability, bool writeable, addr_t base, size_t size)```

This should work recursively. That means, it should be possible to create an alias dataspace from an alias dataspace.

Of course, it must only be possible to obtain an alias with less or equal rights than the original one. E.g., if the supplied dataspace capability refers to a read-only alias dataspace, the resulting dataspace capability will refer to a read-only dataspace regardless whether ```writeable``` was set to true or not. The same applies for the spacial scope of the dataspace. The alias may be smaller but not bigger than the original dataspace.

When a RAM dataspace or an alias gets destroyed, all aliases created from this dataspace should be destroyed as well.

Consequently, within core, we need to maintain a hierarchy of aliases for each RAM dataspace. Right now, the ```Dataspace_component``` class is not very well suited for that. We will possibly need to decompose this class into multiple classes to better capture the different kinds of dataspaces (RAM dataspaces, aliases for RAM dataspaces, managed dataspaces, non-RAM dataspaces).

@ehmry Given the plan outlined above, I am hesitant to include your patch right now because your feature will eventually be superseded. On the other hand, I cannot give you a tangible time of realizing my plan. What do you think how we should proceed?

ehmry, 2015-07-20:
@nfeske Your alias function is much less ad-hoc and covers what I want, so I can  wait for that. I think I could implement alias for simple dataspaces, but I don't understand the exotic dataspaces or the any of the kernels specific memory mapping and paging.

I am content with waiting for now, my use case is that I have two components that store some common state in files, and both components load the files into memory at roughly the some time, so it makes sense to me to have a cache that runs next to the server.

alex-ab, 2018-04-12:
I added two different options to achieve the desired feature, by the simplest (not invasive) way for core. The implementation only work well out for the case, where the creator of the dataspace early in the lifetime of the dataspace diminish write access - so to speak before the capability of the dataspace is passed to clients.

nfeske, 2018-04-13:
I would rule out option B because the side effect on any region map is quite unexpected and may introduce subtle memory-safety problems.

Option B looks like a very good intermediate solution. It covers @ehmry's `static_fs_rom` (#2760) use case while keeping the modifications of core at a minimum. The only important limitation is that existing mappings are not changed from writeable to read-only which is would be a problem if fs_rom shared the dataspace before sealing it. But I cannot imagine a scenario where this is useful.

Still, dropping the rights of the existing mappings might be more consistent (especially across different kernels) and would allow the use of the mechanism for component-local privilege dropping. E.g., the dynamic linker could seal the text segment of shared libraries once loaded. So in the longer term, we should complement the option B with the mapping-flush operation in core. I am afraid that the latter is not easy on all kernels though. In particular on Linux, we need a way to diminish the write access of an existing file handle. I don't know if this is actually possible with the Linux kernel interface.

In @alex-ab's patch, the mechanism is called *diminish_write*, which is quite self-explanatory. In prior conversations, I referred to this mechanism as *sealing* the dataspace, which is shorter and nicer to use as verb, but it is less descriptive. What would be the better term?

alex-ab, 2018-04-13:
In ce59f5d43c9e171abfa5955cda9bbe668c1646e2 I  adjust dynamic_rom_session.h. I had to change the behaviour to ever destruct the old RAM dataspace, since due to the diminish it is not re-usable. Is ithis acceptable @nfeske/all ?

nfeske, 2018-04-13:
@alex-ab I would definitely not go that far. I am afraid that this change would severe impede the performance in many situations that currently profit from repopulating the same dataspace with different data (report_rom, init config updates, session-request ROM).

nfeske, 2018-04-13:
@alex-ab ... as far as I can see, in all these cases, dataspaces are only shared between exactly two parties. So there is practically no security benefit of applying the write-diminishing here.

alex-ab, 2018-04-13:
Ok, I see.

alex-ab, 2018-04-19:
Removed fixed for now, since we had another idea to implement this feature, which I will address asap.

alex-ab, 2018-05-07:
91c9486a121e981c84295d09ac234ed20f15fad3 implements the other idea to support this feature, please review.

nfeske, 2018-05-07:
Cool! Thanks for the picking up this issue again.

I wonder, have you considered adding a `writeable` argument to `Region_map::attach()` instead of the `Rm_connection`? I think that this would be more flexible because the feature could also be used for attaching a RAM dataspace to the local address space in a read-only way.

As a minor cosmetic remark, I think we should strive for consistent spelling of `writeable` vs. `writable`. Technically, both forms are valid but we should foster consistency. In almost all places throughout Genode, we nowadays use the form `writeable` but unfortunately, you took the ancient code in core as starting point for your work, which happens to use the outdated form `writable`. Would you mind adjusting the patch in this respect?

alex-ab, 2018-05-08:
I see, please review 9844043a55fc3c00147f84e3bc48119f32590e00. Maybe we want to extend ```Attached_ram_dataspace``` with a read-only argument in the constructor ?

nfeske, 2018-05-15:
@alex-ab Excellent - that's exactly what I had in mind! Further down the road, we should replace the bool arguments from the public API (using enum types instead) but that's for a different step. I'd actually like to merge the patch as is.

Regarding the `Attached_ram_dataspace`, I agree that such an extension would be sensible. But for now, I'd like to defer this change until we have come up with a good Genode-wide consistent way of how to express such arguments. In contrast to `Region_map`, `Attached_ram_dataspace` is very much visible. So I wouldn't be happy with a boolean argument here. The most pressing use case of your patch is certainly the static_fs_rom, which does not require the `Attached_ram_dataspace` (at least as far as I see).

@ehmry Would you like giving @alex-ab read-only dataspace patch a try for the static_fs_rom? If everything goes well, we could even include it in the release then. ;-)

alex-ab, 2018-05-16:
@ehmry: thanks for the info about the issue with hw - with 9ae079e56cc20940e07f3b4229161b61da106d19 it works for me

chelmuth, 2018-05-16:
Do you know why the W bit has to be checked first? And also, why the I bit is handled after P?

ehmry, 2018-05-16:
I have a fixup for noux an the platform driver at 94082f4d2e07cafd7ca468c7c4925eee6ff1fc51.
6d69344c334eba2612b5182737050ace3ffae23a adds `static_fs_rom` and a test `run/ready_only_rom`.

alex-ab, 2018-05-17:
It turns out that 9ae079e56cc20940e07f3b4229161b61da106d19 is more a optimization and masks the actual root cause. b2d9b3d8922dde7043044b56a3df6f8a5aae4609 fixes the actual issue and 9ae079e56cc20940e07f3b4229161b61da106d19 may or may not be used.

alex-ab, 2018-05-17:
@ehmry: thx for your fixup, I added it with to my issue_branch

ehmry, 2018-05-17:
@alex-ab and a fixup for vbox at bf541681d8d361cd62a1cd09b67c8d944fca51b4.

alex-ab, 2018-05-17:
@ehmry: thx, added, but I extend your fixup to propagate the writeable further

ehmry, 2018-05-18:
Do we want to have this done by the time of release? If so, I will try to complete and test `static_fs_rom`, otherwise I have other things I would like to get done by then. The tradeoff is breaking compatibility with packages that use RM now, or break them later.

nfeske, 2018-05-21:
@ehmry I would merely use it for the depot_rom in the sculpt scenario, not touching any other uses of fs_rom. The reach of potential problems is quite limited in my opinion. But the benefit in terms of startup times and RAM preservations could be pretty noticeable.

nfeske, 2018-05-21:
@alex-ab Thanks for the series of fixups. I merged the commits of your topic branch to staging to let have our buildbot+autopilot have a look. :-)

ehmry, 2018-05-21:
I'm not concerned about problems with `fs_rom`, just that this breaks ABI for a few non-trivial depot packages.

commit 609d3d48 (2018-05-21): base: support to attach RAM dataspaces readonly

Fixes #1633
closed 2018-05-31 by chelmuth