genodelabs/genode #1635: Embed ram_fs in VFS

ehmry, 2015-07-22:


ehmry, 2015-07-22:
Mutable yet self-contained VFS configurations are usefull because temporary files are deleted at exit and storage resources are accounted soley to a single component.

nfeske, 2015-07-22:
That's just perfect timing! @chelmuth just planned to start working on the same idea after his vacation. He will be pleasantly surprised when he comes back. ;-)

I am particularly happy because this change actually clears the way towards unifying most file-system servers (such as ram_fs) into a single "vfs" server. The VFS server would contain a VFS configuration and make the VFS available to its clients via the ```File_system::Session``` interface. The RAM fs would then simply be an instance of the VFS server configured with a ```<ram_fs>``` VFS node. Ultimately, each file system implementation (including rump_fs, ffat_fs, 9p_fs) could have the form of a VFS plugin. @chelmuth and me are even discussing to turn the TCP/IP stacks (lwip and lxip) into VFS plugins offering a plan9-style file-system interface.

In the past, we had three different points of implementation for each file system (libc, noux, and a dedicated fs server). If we follow the path outlined above, there will ultimately be only a single implementation for each file system in the form of a respective VFS plugin. It is then up to the configuration whether the file system is present as a dedicated component (a VFS instance) or local to the application.

ehmry, 2015-07-22:
:+1: I think its a great plan, sounds like it could result in code that is easier to maintain yet it also makes some more interesting things possible.

nfeske, 2015-07-22:
Would you mind to adjust the include guards of the headers you moved to ```include/ram_fs/```?

ehmry, 2015-07-22:
@nfeske fixed it

nfeske, 2015-07-22:
While we are at it, I just applied some additional cosmetics: e59b200c61ddd596e1f4f541e98a667751cb8e62

It occurred to me that now, when the headers have become public, it is certainly not perfect to have the RAM-fs-specific code reside within the ```File_system``` namespace. It seems to be just a matter of time until we will run into a conflict with other file systems that have different implementations of ```File``` or ```Directory```. It might be safer to either move those classes into a different namespace like ```Ram_file_system```, or to prefix the classes with ```Ram_```. On the other hand, maybe this discussion will become moot when following the plan above (then, ram-fs-code could just be moved local to the VFS library because it won't be needed at multiple places anymore). @ehmry What do you think?

ehmry, 2015-07-22:
@nfeske That is a problem that occured to me, but I left it unresolved because I wasn't getting any clashes.

The only issue I found with that is here: https://github.com/genodelabs/genode/blob/master/repos/os/include/file_system/node_handle_registry.h#L14

I can easily bind the classes the registry needs from Ram_fs:: into File_system::, but I think eventually there would be some rewriting to find a middle ground between the VFS handles and the File_system handles, because there is the issue now in my VFS ram_fs that unlinking a node can leave a dangling pointer in a Vfs_handle.

nfeske, 2015-07-22:
This could be solved in two ways. We could either add a reference counter to ```File_system::Node```, which would be consistent with the way POSIX handles open files when unlinking them. The files become invisible but remain to exist until the last file descriptor pointing to the file is closed. Alternatively, we could use the weak-pointer mechanism provided by https://github.com/genodelabs/genode/blob/master/repos/base/include/base/weak_ptr.h A description of this mechanism can be found in Section "8.7.1.  Thread-safe weak pointers" of the documentation http://genode.org/documentation/genode-foundations-15-05.pdf

ehmry, 2015-07-22:
I guess list of potential projects for CCC camp keeps getting longer for me :-)

ehmry, 2015-07-22:
Weak pointer had occured to me, I'll look into it and think about it over the next few days.

chelmuth, 2015-08-03:
@ehmry your efforts are very much appreciated :+1: I'll have a close look next week after returning to Dresden.

chelmuth, 2015-08-14:
@ehmry the patch works flawlessly. I still have one question but it may be that I miss something: Why is it called *ram_fs* and not just *ram* as all other VFS plugins, e.g., *tar* or even *fs*?

ehmry, 2015-08-14:
@chelmuth No reason, I think I had considered that but just left it as is.

chelmuth, 2015-08-14:
Okay, so I'll remove the *_fs* suffix. The last small one: You introduced *SYMLINK_ERR_NO_SPACE* but did not handle it in *vfs_plugin.c*. I'd like to translate it to *ENOSPC*. Does that match your intention?

ehmry, 2015-08-14:
@chelmuth ok let me see, the other thing to decide is if the <ram> node should support the same <content> as the standalone ram_fs.

ehmry, 2015-08-14:
I think I have SYMLINK_ERR_NO_SPACE handled in my latest patch.

ehmry, 2015-08-14:
Updated the commit msg.

chelmuth, 2015-08-14:
I created a *squash* commit from your update - didn't want to screw up staging with a force push currently.

ehmry, 2015-08-14:
ah, right

closed 2015-08-21 by chelmuth

ehmry, 2015-08-31:
I'd like to note that I've done some stress tests, and having a process local ram_fs does give a significant performance boost, and no doubt a gain over the Linux ramfs as well.

nfeske, 2015-08-31:
Very cool!

commit 51be060e (2015-07-22): VFS: local ram_fs instances

Move FS Node implementations from server/ram_fs to include/ram_fs.
Support embedded ram_fs instances in VFS configurations using <ram_fs/>.

Fixes #1635
commit ed70cf5b (2015-08-11): VFS: local ram_fs instances

Move FS Node implementations from server/ram_fs to include/ram_fs.
Support embedded ram_fs instances in VFS configurations using <ram_fs/>.

Fixes #1635
commit 97cefc27 (2015-08-14): squash "VFS: local ram_fs instances" (use new commit message below)

Move FS Node implementations from server/ram_fs to include/ram_fs.
Support embedded ram_fs instances in VFS configurations using <ram/>.
Add 'no space' handling to VFS symlink ops.

Fixes #1635
commit f996697f (2015-08-21): VFS: local ram_fs instances

Move FS Node implementations from server/ram_fs to include/ram_fs.
Support embedded ram_fs instances in VFS configurations using <ram/>.
Add 'no space' handling to VFS symlink ops.

Fixes #1635