genodelabs/genode #1643: Terminal: rich escape sequence support

ehmry, 2015-08-02:
I've noticed that terminal from gems doesn't support the escape sequences produced by terminal_mux and libcurses. Should the terminal decoder be augmented to support more sequences, or should it be frozen for the sake of low complexity?

nfeske, 2015-08-10:
This surprises me a bit because I regularly use ncurses with the terminal when running Vim in Noux. Are you sure that the TERM environment variable of the application is set to ```linux```?

Regarding your question, in the past, I used to extend the escape sequences as needed but tried to stay close to the linux terminal capabilities. Please do not hesitate to extend the decoder if the increase in complexity stays reasonable. As a very useful addition, I would like to see the support for function keys and mouse cursor/buttons added sometime.

ehmry, 2015-08-28:
I'm not going to be working on this anytime soon, so I'm closing the issue.

closed 2015-08-28 by ehmry