genodelabs/genode #1648: VFS backed File_system server

ehmry, 2015-08-16:


chelmuth, 2015-08-17:
My current work circles around socket-based networking. I have a line of changes refactoring the socket-related libc backend and using the VFS plugin in the new socket_openations.c as well as first snippets of a VFS frontend for lxip/lwip. I expect no conflicts with your work that can't be solved once your or my work enters staging.

ehmry, 2015-09-02:
I've renamed this issue from VFS server to VFS improvements because I want to test the corner cases and optimise the VFS before I do the same for VFS as a service, but I intend for the VFS server to close this issue.

I've been optimizing the ram plugin to match the VFS rather than as a server backend, and I naturally decided that every internal directory object in the ramfs should implement the VFS::Directory_service. This has led me to wonder if the VFS would be improved by a more 9P-like interface, in that all operations would take name rather than path arguments, and directories would be accessed using a directory object or handle and a walk operation that moves one directory at a time http://man.cat-v.org/plan_9/5/walk.

My reasoning is that it may be less expensive then the current full path lookup, though it may make walking through composite directories more complicated. But perhaps it would facilitate caching at the VFS layer, which combined with a VFS server, would imply caching for block service backed plugins such as rump plugins.

I'm not sure exactly what the walk function signature would look like or the directory object, and if directories objects would be best allocated on the caller's stack or from within the plugin. I am in favor of keeping an interface that is efficient to call from C.

The question is if this is worth investigating as the VFS server matures?

NobodyIII, 2015-09-03:
As far as improving the VFS goes, it would really be nice to mount filesystems at runtime. That would make file management much easier for users. It seems that implementing the filesystems as libc plugins could help make this feature easier to add.

ehmry, 2015-09-03:
@NobodyIII that shouldn't be too far off, I expect that in the future a GUI toolkit could have an open dialog that was capable of injecting different archives and file servers into the file system at runtime. It should just be a matter of coming up with a good user interface and hooking it into the VFS library.

nfeske, 2015-09-03:
@ehmry The "amend File_system and VFS interface" patch looks very good! The second patch is hard to dissect at a quick glance. I'll need some more time for reviewing it.

I'll try to give you the rationale behind the full-path lookup approach: For now, let us presume that we would not use the entire path for the lookup but some kind of directory handles. Imagine the following VFS configuration:

```
<vfs>
  <tar name="archive.tar"/>
  <fs/>
</vfs>
```

The ```archive.tar``` contains a directory (dir) that contains a file (file) at ```/dir/file```. The ```<fs/>``` node is routed to an external file system like a RAM file system. Initially, the RAM file system is empty. If opening the directory ```/dir```, the VFS would find out that ```/dir``` resides within the tar archive. Hence the open-directory operation would naturally be provided by the tar fs. It would respond to the open-directory request by returning a directory handle. This handle can then be used to obtain the directory listing or to open the files and directories within the directory. This is actually similar to how the ```File_system::Session``` is designed.

Now, another client of the external RAM file system decides to create a new directory "dir" within the formerly empty RAM file system. It then creates a file "file2" within this directory. We arrive at a situation where the same directory ```/dir``` exists in both file systems. When obtaining the directory content for ```/dir```, we would expect to see both "file" and "file2". However, since the directory handle was provided locally by the tar fs, there is no way, it could know that a "file2" appeared in another branch of the VFS. So "file2" would remain invisible unless we would open the directory ```/dir``` again (this time, the VFS would be able to associate the directory handle for "/dir" with both file systems).

By always using the full path for the lookup, this issue cannot occur because all file systems that have a matching path would respond to the lookup. So the aggregate of the content of all involved file systems becomes visible to the user.

However, I clearly see the disadvantage of the full-path lookup from a performance point of view. So in principle, I'd be happy to replace the full-path lookup with a more elegant solution. I just have not found any better way, yet. Granted, I don't know how plan9 solves this problem. I'll probably need to study the link you posted. But for know, I just wanted to share my reasoning behind the current implementation.

nfeske, 2015-09-03:
@NobodyIII I do not think that we should pursue the dynamic change of the VFS at present. Presently, there does not exist a concrete use case. In my opinion, there is no reason a file manage would need to use the VFS for managing new file-system instances. Instead, it could simply operate directly on ```File_system::Sessions```.

I still expect that the principle requirement of modifying the VFS during the runtime of a component will come up sometime in the future. But then, I would like to solve it by letting the VFS atomically respond to a version update of its ```<vfs>``` configuration node instead of providing mount/umount operations. To do that, the VFS would need to employ a mechanism for "diffing" the old and new versions of the ```<vfs>``` configuration. We principally know how to do this. The approach is already employed in the window decorator, which dynamically responds to window-layout changes. But since other issues are much more relevant at present, I would prefer to leave this problem out of the current VFS-related scope of work.

Apart from prioritizing work, I would not underestimate the problem of dynamic VFS changes. Even though the dynamic addition of new file systems to the VFS is not a big problem, the dynamic removal is certainly very complicated. We will need to come up with a sensible life-time management scheme for the open files when their underlying file system vanishes. I'd prefer us not entering this rat hole for now. ;-)

NobodyIII, 2015-09-03:
Okay. Good to know. As for using multiple filesystems, do I need to create
a child process for each filesystem, or is there some way to have multiple
instances of a service used by a single process?

nfeske, 2015-09-03:
@NobodyIII A component can open any number of file-system sessions without the need for any children.

NobodyIII, 2015-09-03:
How would the config for that look?

ehmry, 2015-09-03:
@NobodyIII You can stack multiple file system nodes in the VFS config, each with different session labels.

ehmry, 2015-09-03:
@nfeske The first patch I referenced here is pretty easy to go through, but the second is unreadble as a diff, I basically re-wrote the ram plugin and reverted the ram_fs server back to what it was before the split. The first patch fixes some stuff but I have not tested all the file_system servers against it.

The theme of both patches is preventing problems caused by resource exhaustion, lately I've been creating hundreds of thousands of files and seeing what happens. The problem with the VFS server is that I want to be sure that clients do not expend their session quotas, but its looking like that would put a lower bound on the number of open handles they can open.

With the ram_fs the client session quota is not an issue, because the clients do not pay for metadata, the file handles are essentially a reference to a pointer to the actual element in the file system. With the VFS server however, an open file handle must refence a full path and some permissions metadata, which can eat into quota from the parent if there are many handles open at once.

I see what you mean about the full path lookups, I think I should try and find a quick solution for now and think about it for a while. I think my hangups with the VFS involved a handle leak in the FS plugin, so I should be able to get back to nix stuff.

ehmry, 2015-09-04:
I have the VFS server in a state that I'm happy with. Performance is only slightly worse than ram_fs.
I'm open to changing the name, and I can deduplicate some more of the code if desired.

My `vfs_srv` branch has a few other changes, I threw all my other FS stuff ahead of the server, but I can trim those out.

ehmry, 2015-09-04:
To make a note on the change to void `sync(Node_handle)` at the File_system interface, the idea was that if the VFS has multiple block-backed FSen and caches, it would be nice if one did not have to wait for everything to flush. But after looking at the notify code in `ram_fs`, I made it so that writing to files and creating file in directories will mark nodes as updated, but notifications are only sent when the node handle is closed, and or when sync is called on the node. This way a series of modifications can be made to a node, and then sync operation can be called to send a notification.

ehmry, 2015-09-05:
I have a linux host plugin but it segfaults when its loaded by the library, not sure whats going wrong.

nfeske, 2015-09-07:
Thanks for the posting the ```vfs_srv``` branch. How about naming the component plainly ```vfs```. We don't usually suffix our services with ```_srv```.

I agree with your change of the sync interface.

The Linux plugin crashes because it is built as a ```lx_hybrid``` target (see base-linux/lib/import/import-lx_hybrid.mk). Such targets use the host tool chain and the libraries available on the host. The resulting binary is not compatible with the binaries produced by the Genode tool chain. In particular, they are linked against glibc. Consequently, our dynamic linker is not able to deal with the produced shared object. To solve the problem, the vfs_linux plugin must become independent from the host libraries. Instead of using the ```lx_hybrid``` library, we need to use our Linux syscall bindings. They have the form of the ```syscall``` library hosted within the base-linux repository.

ehmry, 2015-09-07:
I'm checking over the rump plugin now, but while I'm here do we want error handling on the factory create method?
```c++
struct Vfs::File_system_factory
{
	virtual File_system *create(Xml_node node) = 0;
};
```

nfeske, 2015-09-07:
@ehmry That is extremely cool!

The natural approach would be to propagate errors as exceptions. But can you be more specific about the type of error you are facing?

ehmry, 2015-09-07:
rump_fs just throws a bare Genode::Exception(), I haven't run into a failure here yet, so I can leave it.

ehmry, 2015-09-08:
Ok, I've reordered and rewritten some commits so that these changes shouldn't have any effect on the existing FS servers, and the VFS server and rump plugin are ready.

Tests are at libc_vfs_ext2, libc_vfs_fs_ext2, libc_vfs_fs, and libc_vfs_ram.

ehmry, 2015-09-08:
I have two branches off of staging, vfs and vfs_rump.

ehmry, 2015-09-08:
I spoke too soon, there is a problem that the libc tests did hit.

ehmry, 2015-09-11:
I have some finishing touches to put on the VFS server, but I'm running my thinkpad using the rump and ram plugins, is there an ideal way to structure theses commits?

nfeske, 2015-09-11:
@ehmry The commits look truly great and I am fine with structuring them as you did.

I'd like to see them (with the minor ascii_to-refinement I suggested) merged into staging as soon as possible. But first, we need to get @skalk's big restructuring change through. When all tests are happy again, it would be nice of you to rebase your branch on the then-current master.

@cnuke Could you please have a look at the "VFS Rump kernel plugin" patch? I don't feel competent enough about rump to see the possible implications of the changed locking.

chelmuth, 2015-09-11:
@ehmry I like your patch series pretty much, because it applies even the most recent changes on master (e.g., how the build of plugins should be handled) without our intervention. I expect your commits to be merged into master at the beginning of next week. I'm already looking forward to configure the VFS of my next scenario ;-)

ehmry, 2015-09-11:
Ok, my vfs and vfs_rump branches are ready and based on staging. I've add tests to the autopilot list that I think are appropriate.

The only thing I'm worried about is I have are a warning about a bad deallocation when unlinking directories with <ram>. Dosbox is grumpy about permissions on rump file systems, but thats another issue.

ehmry, 2015-09-14:
Be aware, I was getting rump kernel panics but it turned out to be caused by nothing more than a dirty disk structure (++microkernels).

ehmry, 2015-09-18:
This latest patch allows clients to request session roots at vfs server that are symlinks to directories. It also allows the fs_file_system to be created internally from a VFS plugin rather than always with an XML node.

These are both for supporting a nix VFS plugin. This plugin evaluates an expression that creates a directory, and then returns a fs_file_system that is rooted at the evaluation result at the nix store, so in this way we have a very general 'frontend' to nix. The plugin very fragile in regards to routing of various sessions, but I will come up with scenario when I can eliminate some of the silent failures.

nfeske, 2015-09-18:
Dereferencing a symlink specified as root sounds risky. Please consider s VFS with the following content:

```
/user
/user/guest
/credentials/secret_key.txt
```

The VFS config would like this:
```
<policy label="decryptor" root="/credentials"/>
<policy label="guest" root="/user/guest" writeable="yes" />
```

Now, the user "guest" could open a file-system session and would naturally obtain a session with root set to ```/user/guest/```. Since the session is writable, he could create a symlink "foo" pointing to "/credentials". Now, he opens a second file-system session but specifies ```root=foo``` as session argument. The server would append this argument to the root attribute of the session's policy. This would result in ```root=/user/guest/foo```. The server would see that this is a symlink, dereference it, and would hand out the directory ```/credentials``` as the root of the new session.

ehmry, 2015-09-18:
Ah, yes, that is no good, I will have to dereference the symlink elsewhere.

ehmry, 2015-10-05:
rebasing

nfeske, 2015-10-06:
Today, @chelmuth and me finally found the time to review your patches. They are quite a feat, and huge in size. After two hours of sitting together we still could not go through them all. To make our lives easier for merging changes into staging, we would greatly prefer smaller pieces. As the patches stand, they raise a number of questions, which we are, even in our pair-reviewing session, unable to answer by just looking at the code. As the file-system handling is security critical, both @chelmuth and me need to maintain a thorough understanding of the code. @ehmry May you help us with reaching this point?

Our general impression is that you like using C-Style character buffers and C-style string functions, whereby we'd like to avoid them wherever possible because they are bug prone. Everytime I declare a C-style char array, I ask myself if there is a way around it. For example, in many cases, a typedef of a Genode::String is reasonable to use and automatically enforced the null-termination of a string. Or the Genode::Path utility provides higher-level path manipulation functions while enforcing the invariant that a path starts with a ```/```. If those utilities, don't cover your needs, maybe you could consider to sensibly enhance them instead of repeatedly hand-writing C-style code.

Here are our specific questions and remarks:

* The ascii_to patch is fine. We will merge it to staging today.
* The "amend File_system and VFS interface" patch contains far too many things in one patch.
* Adding ```writeable``` as a session argument is fine. At the server side, we should default to ```false``` though. At the ```Connection``` constructor, using writeable as default is fine.
* The changed interface of the sync function is reasonable as we discussed earlier. But we would prefer this change in a  separate patch.
* The ```session_root_path``` function: Is this a candidate for using the ```Genode::Path``` utility? In principle, the function just concatenates the policy argument and the session argument. Could the code not live without the distinction of ```(strcmp("/", path) == 0)```?
* Moving DEFAULT_TX_BUF_SIZE to file_system_session/connection.h is good.
* Even after a long discussion, we could not wrap our heads around your change of the semantics of ```Vfs::Dir_file_system::_sub_path``` and would prefer to keep the original version unless the change is absolutely needed. We failed to understand the reasoning behind the change. Also, it seems to makes the surrounding code more complicated. I.e., the new version contains numerous checks in the form of ```strcmp("/", path, 2)```, which were not needed before. Whereas @chelmuth had no problem following my original code, we both were unable to follow the new version. The commit message just states "Tweak some path lookups in dir_file_system." But since the code will be sensitive for the enforcement of access control with the VFS server, it needs to be easy to understand. How should we go about it?
* How do you think about using the ```Genode::String``` for ```Fs_file_system::Root```? With my recent addition of ```Xml_node::attribute_value``` for ```Genode::String``` arguments, it may become much shorter. See https://github.com/genodelabs/genode/commit/a80d944944e022db769d1ac82b830801e3af8c64
* We do not understand the motivation behind the addition of ```dir_path.remove_trailing('/');``` at various places. What is the problem addressed by this change? I somehow sense that it is connected to the changed path handling in ```Vfs::Dir_file_system```, but I am not sure.
* The fix of the handle leak in mkdir and leaf_path is very good.
* The motivation behind the ```optimize VFS ramfs plugin``` patch is clear to us. Avoiding the indirection is nice.
* We already identified the risk for dangling pointers in the ```Ram_vfs_handle``` struct. It would be sensible to annotate the pointer with an ```XXX``` tagged comment that highlights this deficiency until we fix it.
* The ```walk_path``` function is very hard to read. Maybe it would be worth considering the use of ```Genode::Path```? The ```strncpy``` is prone to write beyond the array boundary. The array is dimensioned with ```i``` whereas ```++i``` is specified as len argument. In principle, the function is similar to ```Ram_fs::Directory::lookup_and_lock```. Have you considered to base your code on this existing (and in my opinion very easy-to-follow) implementation? If yes, why have you disregarded it?
* We should not use ```memset``` to initialize structures like you are doing with ```Stat```. Instead, we could imply initialize the struct members with their default values using the C++11 syntax:
  ```
  struct Stat
  {
    file_size      size   = 0;
    unsigned       mode   = 0;
    unsigned       uid    = 0;
    unsigned       gid    = 0;
    unsigned long  inode  = 0;
    unsigned       device = 0;
  };
  ```

I hope you don't find my remarks overly picky. We are very impressed by your work and would love to integrate it onto the staging branch. But we need to feel confident in the changes made, especially the ones that refer to parts of Genode that will eventually become critical for the overall system security. Would you like to help us with dividing the patches into smaller pieces that are easier to swallow and to review?

ehmry, 2015-10-07:
I don't have any problem with breaking it up, squashing into that big chunk made it easier to move work between branches for me, but I will look over this list and make a series of annotated commits.

ehmry, 2015-10-08:
Everything from 6830309 on should be good, the commits leading the VFS server are required to build, and the commits after fix errors that arrise. The ``libc_vfs_fs`` (client->vfs_server->ram) does not pass, this is due to an error in the current ram fs plugin, so I will now review my new implemtation.

Sorry about all those C strings, I don't know exactly how my changes to dir_file_system work either, but it has something to do with dissapearing nodes, I will isolate that after the ram refactor.

nfeske, 2015-10-08:
Thank you very much for taking the time to structure your commits. You don't need to hurry though. @chelmuth and me will be away for the upcoming two weeks. So please don't worry if you don't get immediate feedback during this time.

ehmry, 2015-10-26:
Rebase

ehmry, 2015-10-26:
I rewrote the ram file system, its a little faster and I think it should be easier to read.

I did however make two changes worth mentioning:
* Recursive unlink is not allowed. Calling destructors recursively is cheap and easier, but in other file systems where the backend doesn't handle recursion this can be expensive. My concern is a thread with a small stack might unlink fine on one file system, but blow the stack in another becase the cost was hidden. Not sure if thats valid, but I suspect this is why POSIX kernels won't unlink recursively.
* ``Symlink(...)`` will overwrite the content of an existing symlink, no prior unlink is needed. This conforms to the ``File_system`` interface and essentially allows atomic replacement of files and directories.

nfeske, 2015-10-28:
Thanks @ehmry for the very good split-up of the commits! I merged the patch series to staging now and added a few fixups. Could you please review them? Also, I left some comments inline.

While reviewing the VFS server, @chelmuth and me could not get a good grasp on the motivation behind the node cache. Wouldn't it be possible to create a node object per handle? This would eliminate the cache as a potential cross-talking point between sessions. The mode could just be stored in the node, which would reduce the difference of the node_handle_registry with the original.

As a side effect of the review, I opened issues #1743 and #1744.

nfeske, 2015-10-28:
@ehmry If you decide to refine the patches that are already on staging, please create fixup commits instead of replacing the original commit.

ehmry, 2015-10-28:
From what I remember the original motivation of the node cache was to deduplicate memory between sessions, because the safe donation size of a client would need to be sizeof(session)+sizeof(buffer)+MAX_NODE_HANDLES*sizeof(Node). The second reason for the cache was to keep sessions consistent between unlinks, as a side effect of this a file that is open in another session cannot be unlinked.

I think a fix for the donation issue could be to add an ``Out_of_metadata`` exeception to the session and then clients catch and donate in the background. I have something like this wrapping the nix ingest connection https://github.com/ehmry/genode-nix/blob/master/include/store_ingest/connection.h

chelmuth, 2015-10-29:
I see that this also boils down to what I was thinking about in my remark a second ago in https://github.com/ehmry/genode/commit/35b5e1e3f6993e1967e27c04025e59336b92f27e. I don't agree that there exists a general *safe donation* and I'm sure that a cache which requires a kind of global/shared quota for all clients is the wrong direction. Shared caches with limited size always require an eviction strategy which we definitely don't want for node handles ;-)

Regarding the handling of `Out_of_metadata` situations I'd like to refer you to the retry pattern, which uses https://github.com/genodelabs/genode/blob/master/repos/base/include/util/retry.h and is applied at https://github.com/genodelabs/genode/blob/master/repos/base/src/base/env/platform_env_common.h#L75 for example.

ehmry, 2015-10-29:
Well, for now there is no eviction stategy and server will just crash. I had noticed the resource usage is different between ram_fs and the vfs server because in ram_fs a handle is more or less a pointer to an object, but in vfs a handle must store atleast full path into the VFS. With rump_fs you have something similar, a integer descriptor and then some sort of session deduplication in the rump kernel.

ehmry, 2015-10-29:
The Out_of_handles exception could be replaced entirely with Out_of_metada if the node registry was a series of dynamically allocated tables, but for now I think MAX_NODE_HANDLES is generous enough.

nfeske, 2015-10-29:
My main concern regarding the node cache is that the VFS will certainly play the role of resource multiplexer, shared between clients of different trust levels. This means it must be resilient against the following scenarios:

* A malicious client may want to make the VFS unavailable to other clients by deliberately crashing it or consuming all of its resources. There are many attack vectors possible (e.g., writing large amounts of data to a RAM fs, triggering exceptions that lead to an abort, exhausting the component's virtual memory) but each additional data structure shared between sessions makes the situation potentially worse. So if we can avoid the shared cache data structure, even it it means to sacrifice memory efficiency or performance, it is the best to remove it.

* The VFS may be used as a information broker between clients of different trust levels. In particular, users may set it up with a "data diode"-like policy where one client is allowed to write information to be read by another client, but not vice versa. Even if both clients are misbehaving, the information-flow policy must be in effect. In such scenarios, any data structure shared between such clients must be thoroughly analyzed for the absence of covert storage channels. The unlink behavior you described represents actually such a channel. Unless a reading session closes a given file, a writing session is unable to unlink it. So bits (the ability whether a file is unlinkable or not) can be transferred from the reading session to the writing session, which would contradict the policy as defined by the system integrator. The best way to counter such problems is to not introduce such shared data structures in the first place.

These concerns may sound a bit extreme but while designing multi-level components like nitpicker, the clipboard, or report_rom, I always try my best to address them. The VFS will end up in a similarly critical role. So I would very much welcome the removal of the shared node cache.

chelmuth, 2015-10-29:
In addition to you first concern I'm not sure how a multi-client VFS may securely spent resources of it clients. In the ram-fs case the session providing RAM quota for file chunks may be closed at any time. This is particularly complicated if different clients have write access to file and may arbitrarily append data. Further, if the VFS uses a remote fs or block service it has to donate resources for session creation and maybe later even upgrade the session resources. In this case, also different clients may at any time close their session and expect their RAM back.

nfeske, 2015-10-29:
@chelmuth Indeed, there are many such problems not yet resolved. My call was just to not knowingly introduce another one. Regarding the resource-related questions, there exists approaches in the back of my head (e.g., RAM consumption could be paid by the server but subjected to session policies, or paid by the clients with the content vanishing when their session is closed, or both as configuration options). But to find the right direction, we first should gather practical experience in using the VFS.

ehmry, 2015-10-29:
I agree with both of you, I don't need any convincing the cache is bad, I tried to get rid of it as I wrote vfs but didn't want to cause a resource regression in relation to ram_fs. I think some of these issues could be addressed at the VFS library, like with a COW plugin.

nfeske, 2015-10-29:
Alright. I think the best way to proceed is merging the current version of the VFS to master and opening a dedicated issue for addressing the concerns discussed above. We added the commits + our fixups to staging yesterday and both the buildbot and the autopilot are happy (including your added test). I will open a dedicated follow-up issue in a few minutes.

commit cb9d8a19 (2015-10-28): Vfs: File_system server backed by VFS library

Test available at libc_vfs_fs.

Fixes #1648
commit b59460dc (2015-11-03): Vfs: File_system server backed by VFS library

Test available at libc_vfs_fs.

Fixes #1648
commit b6c558dc (2015-11-04): Vfs: File_system server backed by VFS library

Test available at libc_vfs_fs.

Fixes #1648
closed 2015-11-04 by chelmuth