genodelabs/genode #1652: hw: Platform_thread::state() returns wrong/incomplete cpu state

alex-ab, 2015-08-21:
In case of a unresolvable page-fault (to small stack) one gets a misleading pagefault warning:

```
[init -> test-thread] 79 : _cpu_session=0x24b1d8 env()->cpu_session()=0x24b1d8
[init -> test-thread -> 80: unresolved pagefault at ip=0 sp=0
```

Actually the fault address is missing and the ip is not 0.  Platform_thread::state() as called by
https://github.com/genodelabs/genode/blob/master/repos/base-hw/src/core/pager.cc#L91
returns bogus values here.

alex-ab, 2015-08-21:
@skalk, @m-stein : 2324a99fde742ed763fb7b82e197b5bd9ebfe31f is rather a quick fix to actual see the valid ip and sp pointers - what do you think ? Still the actual fault reason/address is still missing ...

m-stein, 2015-08-21:
@alex-ab Pushed the outcome of our discussion.

alex-ab, 2015-08-21:
thanks @m-stein 

@nfeske, @chelmuth : please consider ace1f3cf72e2aa3c5d67d5c302cf0bc9e54a8fcd and b8bf183fed7671f368959d7eaaa5cdfb43b55b85 for staging - it should fix the thread test issue on several hw_* platforms

m-stein, 2015-08-21:
@alex-ab Before merging the commits to staging, could you please be more verbose in your first commit message about why pt->state() returns bogus?

m-stein, 2015-08-21:
... and why this is related to hw_x86_64 :-)

alex-ab, 2015-08-21:
the mentioned 2 commits don't fix this issue

alex-ab, 2015-08-21:
The thread test just discovers this bug described here, the 2 commits however fix the thread test. 

Regarding this issue - 'cpu_state_regs()' in repos/base-hw/src/core/spec/x86_64/kernel/thread_base.cc is used by 'Thread_state Platform_thread::state()'. cpu_state_regs is however empty and therefore no state is transfered from the kernel to core, therefore ip/sp are bogus.

m-stein, 2015-08-21:
So, actual, the problem was that the cpu_state_regs things aren't implemented for hw_x86_64. But they are deprecated anyway. The Platform_thread::state functions and thereby also the cpu_state_regs things are not needed anymore when applying your first patch - which is totally fine - but we should remove them.

m-stein, 2015-08-21:
Nevertheless, I would vote for adding a commit message ;-)

skalk, 2015-08-24:
Beside of the above fixes, which I agree to, I'd like to get rid of the 'access_thread_regs' syscall in the HW kernel. It is a rather complicated one, actually an additional call wrapped in a syscall. You have to define which register you would like to access with corresponding platform-specific enums written to the UTCB. First this is error-prone, see above example (it was the incomplete 'access_thread_regs' syscall that lead to wrong state() return values on hw_x86_64). Moreover, it is a lot of additional glue code that needs to be defined per architecture.

Instead, I would access the thread registers from core's side directly via the kernel object. Of course, core has to enforce that the corresponding thread is not running when accessing the registers, which is implicitly the case currently (only used in page-fault situations).

alex-ab, 2015-08-24:
Something like 231d7e28ab3c7723d788d21fb9259ca6373bbd56 ? The only issue here is, that the two Thread_base::state() methods are called by cpu_session_component to retrieve the registers. How to do that ? Presumably it is not used yet on hw (I'm right?) - but gdb could/would on x86 ... Another questions, must the syscallcall ids dense packed ? I just removed access_regs and id 20 becomes unused.

skalk, 2015-08-24:
@alex-ab: one can cast the platform thread's kernel object to Cpu_state, as well as the Thread_state and copy it over directly. It has the same effect as right now.

With respect to the syscall IDs: I would vote for compression here, because HW is not widely used by a lot of developers right now. Apart from that, I think there is no need to stay ABI compatible right now.

skalk, 2015-08-24:
@alex-ab: I forgot: thank you very much for the proposal, I really like the -240 line in sum :+1:

alex-ab, 2015-08-24:
ok, please re-check

skalk, 2015-08-24:
cool!

alex-ab, 2015-08-24:
Ah, what I forgot. For the Platform_thread::state() calls the thread must be suspended - how to do that on hw ?

alex-ab, 2015-08-24:
So, the check that it is not running.

m-stein, 2015-08-24:
@alex-ab There's a syscall ``pause_thread`` in the privileged interface and a syscall ``pause_current_thread`` in the user interface of the Kernel.

m-stein, 2015-08-24:
I'm working on a related patch that removes some more deprecated stuff.

skalk, 2015-08-24:
@alex-ab I do not think that we have to pause the thread implicitely here. AFAIK when GDB needs to stops the thread it does so explicitely.

skalk, 2015-08-24:
In all other use-cases (information are used by core/paging) the thread is already suspended.

m-stein, 2015-08-24:
@alex-ab I've added https://github.com/m-stein/genode/commit/4ccfd68c7ee30a428408b3b4eac4da265d7905eb and https://github.com/m-stein/genode/commit/3689ebe0ac93cd8bbac6f716f60c22b3a32b85e5. Btw. thank you for your patches. I like them as they are.

alex-ab, 2015-08-24:
I don't intend to stop it explicitly. On base-nova you get a Cpu_session::State_access_failed failed exception if you try to invoke Cpu_session:state() on a not-suspended thread - just wanted to make the behaviour similar.

@m-stein: cool, so fixed ?

m-stein, 2015-08-25:
The most recent 8 commits on my branch 1652_hw_Platform_thread_state_returns_wrong_incomplete_cpu_state fix this issue.

chelmuth, 2015-08-31:
Rebasing m-stein/1652_hw_Platform_thread_state_returns_wrong_incomplete_cpu_state produces conflicts. Could you please perform the rebase and squash appropriately?

m-stein, 2015-08-31:
@chelmuth Have rebased m-stein/1652_hw_Platform_thread_state_returns_wrong_incomplete_cpu_state.

chelmuth, 2015-08-31:
Thanks, merged into staging.

closed 2015-09-16 by alex-ab