genodelabs/genode #1654: rpi: GPIO support

rmillo, 2015-08-24:
RaspberryPi GPIO support

rmillo, 2015-08-24:
We have taken into account that each pin can be configured for diferents functionalities (up to 5). The configuration of the functionality of each pin can be set using the attribute **function** inside the **gpio** configuration tag.

The GPIO driver is using the interrupt 49 and have differentiated numbering for **base_hw** and **base_foc**.

Two examples were implemented to test the controller. The first example (`run/gpio_led`) simulates a blinking led (ACT GPIO 16) and the other example (`run/gpio_signal`) uses a pin for input and other for output connected to a push button to alternate the led state.

trimpim, 2015-08-25:
Some months ago I started a similar project (issue #1549) The Idea is to allow to configure the functions of the IO pins through the config for the IoMux.
It can be found here: https://github.com/trimpim/genode/tree/issue1549
Unfortunately I didn't have time over the summer to work on this. I think starting next week I will have time to work on the project again.

m-stein, 2015-08-25:
@rmillo Thank you for these clear commits!

I've added a fixup https://github.com/m-stein/genode/commit/73a7811179fb0bd887586a5ded054fdade59ff80 that suggests to move the IRQ definitions to the driver directory. I think we should not pollute the kernel repos for that, as long as the IRQ is not needed by any other component anyway. This way, the commit also gets more compact with regard to the repos it touches.

The fixup also introduces another way of managing the build configuration of the driver to achieve specification more comfortable. It moves the whole configuration to the gpio library. In new driver implementations that use specifier sub-dirs this is the default as it saves build overhead and the specifier directories in ``lib/mk`` are easier to organize than requires rules that are spread over the target.mk files.

Furthermore I would vote for merging your first and second commit as well as my fixup into one commit. I hope you're ok with the that?

I'll push another fixup for minor style fixes during the day.

rmillo, 2015-08-25:
@trimpim I have followed the thread discusion *"I2C / SPI drivers for the Raspberry Pi"* on the Genode Mailing List, and it was the start point to implement this basic driver. I have been working on it on this summer.

I have checked your branch but in this case you are working with I2C and SPI. Basically this implementation allow work directly with the IO pins to connect another devices. The SPI and I2C can be implemented on top of it.

I'm not using the Iomux, but the IO pins can be configures through the config of the *gpio_drv*. You can see it on the proposed test examples.

rmillo, 2015-08-25:
@m-stein It seems fine, i'm agree with your changes.

In the file `repos/os/lib/mk/hw_rpi/gpio.mk` you must change the first line containing

```lua
INC_DIR += $(REP_DIR)/src/drivers/gpio/rpi/foc
```

this must be set to

```lua
INC_DIR += $(REP_DIR)/src/drivers/gpio/rpi/hw
```

m-stein, 2015-08-26:
@rmillo  
Thank you for the hint regarding my first fix-up :) I've updated it.

I've also added a fix-up https://github.com/m-stein/genode/commit/805d5ef629cc140d48e58d56aa18b18ee1415fe2 that does the following:
* Some minor style fixes like correcting indentation and adding missing spaces
* The complexity especially of gpio.h was significantly reduced by a more intensive use of the MMIO-framework (most notably through register arrays). I think it is worth it to have a look also at the advanced functionalities of this framework when it comes to further driver development in the future. More information on that can be obtained from newer drivers like ``os/src/drivers/ahci/ahci.h``, the framework unit test ``base/src/test/util_mmio``, or documentation such as [release-notes/13.11](http://genode.org/documentation/release-notes/13.11#New_support_for_bit_sets_in_MMIO_API), [release-notes/12.08](http://genode.org/documentation/release-notes/12.08#Base_framework), and [release-notes/12.02](http://genode.org/documentation/release-notes/12.02#MMIO_access_framework). If you have questions regarding this, don't hesitate to ask ;)
* The validation of the GPIO pin-number was done by the Gpio_reg back-end but I think it is better to do it consistently at the beginning of each interface function that receives such an argument. If one hands over a wrong GPIO number the driver shouldn't do anything but providing a hint and leave.

Do you agree with these modifications? If yes, would you mind testing them with your hardware? The outcome should be the same as with the original implementation but I can't verify it currently.

rmillo, 2015-08-26:
@m-stein It seems fine, i'm agree with your changes. I have tested on the hardware and every thing is ok.

m-stein, 2015-08-26:
Thank you for the fast review and feedback :)

m-stein, 2015-08-26:
May we label it as fixed so it can be merged to staging?

rmillo, 2015-08-26:
@m-stein ok

chelmuth, 2015-08-27:
Merged into staging.

commit 09a1b25f (2015-08-27): gpio: RaspberryPI GPIO driver

Fixes #1654
commit 98da4452 (2015-08-31): gpio: RaspberryPI GPIO driver

Fixes #1654
closed 2015-08-31 by chelmuth