genodelabs/genode #1655: ldso: segfault in ld.lib.so after Archlinux system upgrade

ValiValpas, 2015-08-24:
After a system upgrade, I get a segfault in ld.lib.so when running run/ldso:

> segfault at 500b76f8 ip 0000557b37b27578 sp 00000000500b7700 error 6 in ld.lib.so[557b37acf000+6e000]

Note that the sp is still at the top of the stack.

I'm using Archlinux. Please let me know which software versions you are interested in.

ValiValpas, 2015-08-25:
I tried to narrow this down a bit. It seems to be related to the recent kernel update from 4.0 to 4.1.

chelmuth, 2015-08-27:
I changed the title to reflect that the affected platform is Linux. Furthermore, it looks like our magic to reserve and map the context area is broken. As I assume that static programs still work could you please have a look at the memory map `/proc/<pid>/maps` of any Genode component.

chelmuth, 2015-08-27:
Hm, looking at your log snippet again I'm wondering why your stack is at 0x500b7700 and not in the context area between 0x40000000 and 0x50000000... maybe @ssumpf has an idea.

ValiValpas, 2015-08-27:
Here is the memory map:
```
01000000-0102e000 r-xp 00001000 fe:00 9256492                            build/linux_x86/hello/client/hello_client
0102e000-01039000 rwxp 0002f000 fe:00 9256492                            build/linux_x86/hello/client/hello_client
01039000-01070000 rwxp 00000000 00:00 0 
40000000-400e0000 ---p 00000000 00:00 0 
400e0000-40100000 rwxs 00000000 00:21 168595                             /tmp/genode-1000/ds-75 (deleted)
40100000-401f8000 ---p 00000000 00:00 0 
401f8000-40200000 rwxs 00000000 00:21 171148                             /tmp/genode-1000/ds-93 (deleted)
40200000-50000000 ---p 00000000 00:00 0 
7f60be23e000-7f60be240000 rwxs 00000000 00:21 170477                     /tmp/genode-1000/ds-56 (deleted)
7f60be240000-7f60be248000 rwxs 00000000 00:21 168584                     /tmp/genode-1000/ds-72 (deleted)
7f60be248000-7f60be250000 rwxs 00000000 00:21 171025                     /tmp/genode-1000/ds-69 (deleted)
7f60be250000-7f60be279000 rwxs 00000000 00:21 170529                     /tmp/genode-1000/ds-65 (deleted)
7ffdd271c000-7ffdd273d000 rwxp 00000000 00:00 0 
7ffdd27be000-7ffdd27c0000 r--p 00000000 00:00 0                          [vvar]
7ffdd27c0000-7ffdd27c2000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
```

ValiValpas, 2015-08-27:
@chelmuth:
The 0x500b7700 is the initial stack (__initial_sp) which resides in the bss section.

chelmuth, 2015-08-28:
Pretty strange that your Linux kernel does not allow the ld.lib.so program to write its BSS. You may patch repos/base/src/lib/ldso/main.cc:550 ff. to just execute `while (true) ;` and have a look at the access rights in the proc-map of the process. In the best case compare the working kernel version with the error case.

ValiValpas, 2015-08-28:
Thanks, chelmuth. I'll have a look. Your proposed solution, however, won't work as the segfault occurs before main() is entered.

chelmuth, 2015-08-28:
My bad, then it's better to instrument *repos/base/src/platform/x86_64/crt0.s*.

ValiValpas, 2015-08-28:
Sure. Any idea how I can use the instrumented crt0.s only for ldso? Otherwise init will be affected likewise.

chelmuth, 2015-08-28:
In such cases I adapt the run script to build the affected binary only on invocation of `build`.

ValiValpas, 2015-08-28:
Good idea ;-)

Here is what I get on the 4.1 kernel:
```
55da689b5000-55da789b5000 rwxp 00000000 00:00 0 
55da789b5000-55da78a22000 r-xp 00001000 fe:00 9246598                    build/linux_x86/var/libcache/ld/ld.lib.so
55da78a22000-55da78a34000 rwxp 0006e000 fe:00 9246598                    build/linux_x86/var/libcache/ld/ld.lib.so
55da78a34000-55da78a6d000 rwxp 00000000 00:00 0 
7ffc3bbc2000-7ffc3bbe3000 rwxp 00000000 00:00 0                          [stack]
7ffc3bbee000-7ffc3bbf0000 r--p 00000000 00:00 0                          [vvar]
7ffc3bbf0000-7ffc3bbf2000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall
```

And here is what I get on the 4.0 kernel:
```
40000000-50000000 rwxp 00000000 00:00 0
50000000-5006a000 r-xp 00001000 08:11 541557                            build/linux_x86/var/libcache/ld/ld.lib.so
5006a000-5007c000 rwxp 0006b000 08:11 541557                            build/linux_x86/var/libcache/ld/ld.lib.so
5007c000-500b5000 rwxp 00000000 00:00 0
7fff26ce4000-7fff26d05000 rwxp 00000000 00:00 0                         [stack]
7fff26d19000-7fff26d1b000 r--p 00000000 00:00 0                         [vvar]
7fff26d1b000-7fff26d1d000 r-xp 00000000 00:00 0                         [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                 [vsyscall]
```

Strangely, the sections are mapped to wrong addresses in the upper case. Here is what `objdump -ph bin/ld.lib.so` shows:
```

bin/ld.lib.so:     file format elf64-x86-64

Program Header:
    LOAD off    0x0000000000001000 vaddr 0x0000000040000000 paddr 0x0000000040000000 align 2**12
         filesz 0x0000000000000000 memsz 0x0000000010000000 flags ---
    LOAD off    0x0000000000001000 vaddr 0x0000000050000000 paddr 0x0000000050000000 align 2**12
         filesz 0x000000000006cea0 memsz 0x000000000006cea0 flags r-x
    LOAD off    0x000000000006e000 vaddr 0x000000005006d000 paddr 0x000000005006d000 align 2**12
         filesz 0x0000000000011f84 memsz 0x000000000004a668 flags rw-
 DYNAMIC off    0x0000000000073988 vaddr 0x0000000050072988 paddr 0x0000000050072988 align 2**8
         filesz 0x000000000000c5fc memsz 0x0000000000044c64 flags rw-
EH_FRAME off    0x000000000006bfac vaddr 0x000000005006afac paddr 0x000000005006afac align 2**2
         filesz 0x0000000000001ef4 memsz 0x0000000000001ef4 flags r--

Dynamic Section:
  INIT                 0x0000000050022e80
  HASH                 0x0000000050000000
  STRTAB               0x000000005000a0d8
  SYMTAB               0x00000000500024d0
  STRSZ                0x000000000000cda5
  SYMENT               0x0000000000000018
  RELA                 0x0000000050016e80
  RELASZ               0x000000000000c000
  RELAENT              0x0000000000000018
  RELACOUNT            0x00000000000003c1

Sections:
Idx Name          Size      VMA               LMA               File off  Algn
  0 .context_area 10000000  0000000040000000  0000000040000000  00001000  2**0
                  ALLOC
  1 .hash         000024d0  0000000050000000  0000000050000000  00001000  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .dynsym       00007c08  00000000500024d0  00000000500024d0  000034d0  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  3 .dynstr       0000cda5  000000005000a0d8  000000005000a0d8  0000b0d8  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  4 .rela.dyn     0000c000  0000000050016e80  0000000050016e80  00017e80  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  5 .init         00000032  0000000050022e80  0000000050022e80  00023e80  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  6 .text         0004342f  0000000050022ec0  0000000050022ec0  00023ec0  2**4
[... 12 lines cut ...]
                  CONTENTS, ALLOC, LOAD, DATA
 13 .eh_frame     0000a59c  0000000050073010  0000000050073010  00074010  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 14 .gcc_except_table 000019d8  000000005007d5ac  000000005007d5ac  0007e5ac  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 15 .bss          00038668  000000005007f000  000000005007f000  0007ff84  2**8
                  ALLOC
 16 .comment      00000011  0000000000000000  0000000000000000  0007ff84  2**0
                  CONTENTS, READONLY
 17 .debug_aranges 00004ee0  0000000000000000  0000000000000000  0007ffa0  2**4
                  CONTENTS, READONLY, DEBUGGING
 18 .debug_info   001ad95f  0000000000000000  0000000000000000  00084e80  2**0
                  CONTENTS, READONLY, DEBUGGING
 19 .debug_abbrev 000196a0  0000000000000000  0000000000000000  002327df  2**0
                  CONTENTS, READONLY, DEBUGGING
 20 .debug_line   0002f245  0000000000000000  0000000000000000  0024be7f  2**0
                  CONTENTS, READONLY, DEBUGGING
 21 .debug_str    000da397  0000000000000000  0000000000000000  0027b0c4  2**0
                  CONTENTS, READONLY, DEBUGGING
 22 .debug_loc    00142433  0000000000000000  0000000000000000  0035545b  2**0
                  CONTENTS, READONLY, DEBUGGING
 23 .debug_ranges 0005cb20  0000000000000000  0000000000000000  0049788e  2**0
                  CONTENTS, READONLY, DEBUGGING

```

ValiValpas, 2015-08-28:
I just found a llvm bug that sounds similar: https://llvm.org/bugs/show_bug.cgi?id=24155

chelmuth, 2015-08-28:
It may be the strange Linux-kernel policy on DYN-binary load hits you here. Please try in *<build-dir>/var/run/ldso/genode*

`/usr/local/genode-gcc/bin/genode-x86-elfedit --output-type exec ld.lib.so; ./core`

Does it work now?

ValiValpas, 2015-08-28:
Yes, indeed.

chelmuth, 2015-08-28:
So, now we have to come up with a solution that 1) allows linking of components against _ld.lib.so_ while 2) the actual binary is not DYN but EXEC. The background is that the Linux kernel places DYN binaries in virtual memory as it sees fit but we start _ld.lib.so_ as normal executable (on Linux as well as on any microkernel platform) which should not be arbitrarily relocated.

chelmuth, 2015-09-30:
Closed in favor of #1728.

closed 2015-09-30 by chelmuth