genodelabs/genode #1657: fiasco & pistachio: system hang during subsystem destruction

skalk, 2015-08-25:
Under certain conditions when destroying a component (e.g.: starting/killing testnit in demo scenario) core & thereby the whole system hangs. When using the Fiasco kernel debugger one can observe that the context area of the component in destruction is already unmapped, thereby its threads are faulting with respect to their stack pointers. The pager thread in core tries to delegate the fault to the managed dataspace therefore tries to send a signal, but the signal thread in core hangs within a reply IPC to the signal handler thread within the component that is in destruction.

The strange thing is that the signal thread in core hangs (polling actively in the kernel) when trying to reply with a timeout of zero, which looks clearly like a kernel bug. On the other hand the system also hangs on top of pistachio, although I could not determine whether it is due to the same reasons, because pistachio does not offer the same mighty debugging tools.

nfeske, 2015-08-25:
Thanks for documenting this phenomenon. Even though I doubt that we have a pressing need to fix it  as both kernels are quite ancient and unused in practice, it would still be good to get a profound understanding of why the situation occurs in the specific scenario (testnit/demo) but not in others (the automated tests).

nfeske, 2016-05-09:
@skalk It might be worthwhile to try it again with my recent rework of the PD/RM/CPU sessions. Does the problem persist with the current master branch?

skalk, 2016-09-01:
I'm not able to reproduce it nowadays, so I'll close this issue.

closed 2016-09-01 by skalk