genodelabs/genode #1658: base: redesign object pool

skalk, 2015-08-26:
The current object pool implementation hands out pointers to objects, which are can be locked, but do not necessarily need to be locked (e.g. first() accessor). This opens up potential for bugs:

* we might get dangling object pointers when using them without holding the lock
* forgetting the unlock leads to deadlock situations

To eliminate such bugs by design, it would be nice to access objects within the pool with functors/lambdas only. Thereby the pool implementation can take care of the synchronized access, and there is no need to export object pointers at all.

skalk, 2015-09-01:
@nfeske @chelmuth : I've rebased the new object pool to current staging.

chelmuth, 2015-09-01:
Added the commit to staging to give it a spin this night...

nfeske, 2015-09-02:
I just browsed the commit and like it a lot! @skalk Thank you very much for the long-overdue renovation! :-)

commit 5442d9cd (2015-09-01): base: redesign object pool using lambda interface

Instead of returning pointers to locked objects via a lookup function,
the new object pool implementation restricts object access to
functors resp. lambda expressions that are applied to the objects
within the pool itself.

Fix #884
Fix #1658
commit 458b4d6f (2015-09-09): base: redesign object pool using lambda interface

Instead of returning pointers to locked objects via a lookup function,
the new object pool implementation restricts object access to
functors resp. lambda expressions that are applied to the objects
within the pool itself.

Fix #884
Fix #1658
closed 2015-09-09 by skalk