genodelabs/genode #1660: vbox: sporadically vanishing 'interrupt pending' flag

cproc, 2015-08-26:
By default, the EMT thread of the last vCPU handles expired timers. When running VirtualBox with 2 vCPUs, it sporadically happens that the EMT thread of the second CPU clears an 'interrupt pending' flag for the first vCPU after changing the state of a timer device model, which is not expected by our Genode-specific code (failed assertion '!_irq_win' in the recall handler).

The problem did not occur yet when letting the EMT thread of the first vCPU handle the expired timers, so we could do that as an interim fix until the problem has been further investigated.

cproc, 2015-08-26:
@nfeske, @chelmuth: please add the workaround commit to staging

nfeske, 2021-02-26:
This issue is outdated.

closed 2021-02-26 by nfeske