genodelabs/genode #1662: log_trap

ehmry, 2015-08-26:
This is a very simple component that watches the log output of a child and exits when a simple string match occurs.

It would be trivial to find a regex library that I could link to for more robust pattern matching, and not too hard to but in some control flow, but I think it is best to wait until a need for that arises.

ehmry, 2015-08-26:
I have had a strange an issue with the  void exit(int) method on the child_policy interface not being overridden. I want to exit when the child exits but the virtual method underneath is getting called instead.
https://github.com/ehmry/genode/commit/0792342a0960c89961203d0a3bea9c58932873ef#diff-7fb35f8945260f1b978f8db3bd7c04eaR197

ehmry, 2015-08-27:
This is no good without removing escape sequences, I will reference this in another issue when its fixed.

chelmuth, 2015-09-30:
I'm uncertain about this feature. Does anybody like to advocate for a merge? ;-)

ehmry, 2015-09-30:
I too am uncertain now, now that init can exit with a child #1686, I don't know if I will need it.

chelmuth, 2015-10-01:
Okay, I take your answer as the opportunity to close this issue for now. Feel free to reopen in case you feel the need to integrate the server.

closed 2015-10-01 by chelmuth

ehmry, 2015-10-01:
Ok