genodelabs/genode #1663: SD card drivers: Wrong capacity calulation for standard capacity cards

twischer, 2015-08-29:
The capacity calculation for SD cards with a size smaller than 1024MiB fails.
The reason for this problem is the calculation in Host_controller::_sd_card_device_size.
This method should always return the size in 512kiB blocks,
but for standard capacity cards it returns the size in Bytes.

The following fix is working for me.
```patch
diff --git a/repos/os/src/drivers/sd_card/sd_card.h b/repos/os/src/drivers/sd_card/sd_card.h
index d18e62a..7eae84c 100644
--- a/repos/os/src/drivers/sd_card/sd_card.h
+++ b/repos/os/src/drivers/sd_card/sd_card.h
@@ -406,7 +406,7 @@ namespace Sd_card {
 			 * Extract capacity information from CSD register
 			 *
 			 * \throw  Detection_failed
-			 * \return capacity in 512-byte blocks
+			 * \return capacity in 512-kByte blocks
 			 */
 			size_t _sd_card_device_size(Csd const csd)
 			{
@@ -417,7 +417,6 @@ namespace Sd_card {
 				 */
 
 				if (Csd3::Version::get(csd.csd3) == Csd3::Version::STANDARD_CAPACITY) {
-
 					/*
 					 * Calculation of the capacity according to the
 					 * "Physical Layer Simplified Specification Version 4.10",
@@ -426,12 +425,14 @@ namespace Sd_card {
 					size_t const read_bl_len = Csd2::V1_read_bl_len::get(csd.csd2);
 					size_t const c_size      = (Csd2::V1_c_size_hi::get(csd.csd2) << 2)
 					                         |  Csd1::V1_c_size_lo::get(csd.csd1);
+
 					size_t const c_size_mult = Csd1::V1_c_size_mult::get(csd.csd1);
 					size_t const mult        = 1 << (c_size_mult + 2);
 					size_t const block_len   = 1 << read_bl_len;
 					size_t const capacity    = (c_size + 1)*mult*block_len;
+					size_t const capacity512KiB = capacity / (512 * 1024);
 
-					return capacity;
+					return capacity512KiB;
 				}
 
 				if (Csd3::Version::get(csd.csd3) == Csd3::Version::HIGH_CAPACITY)
```

nfeske, 2015-08-31:
Thanks for the notice and the patch! I have actually created a fix for this problem some time ago but somehow missed to merge it. May you give the following commit a try? https://github.com/nfeske/genode/commit/575716f98c028db6fdbfa9dbcd7becd59f34b033 But I wonder, with the capacity calculation corrected, does the actual read/write access work for non-HC cards?

twischer, 2015-09-22:
You are right. The read and write operations has to be patched, too.
I tested the following patch with
```bash
make run/libc_ffat_fs
```
in QEMU for Xilinx Zynq platform.
I do not have a Raspberry Pi to test it.
The SDHCI of the Xilinx Zynq is nearly the same as the controller of a Pi.

```patch
diff --git a/repos/os/src/drivers/sd_card/rpi/driver.h b/repos/os/src/drivers/sd_card/rpi/driver.h
index d0b30bb..8134471 100644
--- a/repos/os/src/drivers/sd_card/rpi/driver.h
+++ b/repos/os/src/drivers/sd_card/rpi/driver.h
@@ -76,7 +76,7 @@ class Block::Sdhci_driver : public Block::Driver
 		 ** Block::Driver interface **
 		 *****************************/
 
-		Genode::size_t block_size() { return 512; }
+		Genode::size_t block_size() { return Sdhci_controller::Block_size; }
 
 		virtual Block::sector_t block_count()
 		{
diff --git a/repos/os/src/drivers/sd_card/rpi/sdhci.h b/repos/os/src/drivers/sd_card/rpi/sdhci.h
index 915d779..c71146b 100644
--- a/repos/os/src/drivers/sd_card/rpi/sdhci.h
+++ b/repos/os/src/drivers/sd_card/rpi/sdhci.h
@@ -131,6 +131,8 @@ struct Sdhci : Genode::Mmio
 
 struct Sdhci_controller : private Sdhci, public Sd_card::Host_controller
 {
+	enum { Block_size = 0x200 };
+
 	private:
 
 		Delayer           &_delayer;
@@ -277,7 +279,7 @@ struct Sdhci_controller : private Sdhci, public Sd_card::Host_controller
 			 */
 			Blksizecnt::access_t v = read<Blksizecnt>();
 			Blksizecnt::Blkcnt::set(v, block_count);
-			Blksizecnt::Blksize::set(v, 0x200);
+			Blksizecnt::Blksize::set(v, Block_size);
 			write<Blksizecnt>(v);
 		}
 
@@ -397,6 +399,16 @@ struct Sdhci_controller : private Sdhci, public Sd_card::Host_controller
 			return Sd_card::Send_relative_addr::Response::Rca::get(read<Resp0>());
 		}
 
+		size_t _block_to_command_address(const size_t block_number)
+		{
+			/* use byte position for addressing with standard cards */
+			if (_card_info.version() == Sd_card::Csd3::Version::STANDARD_CAPACITY) {
+				return block_number * Block_size;
+			}
+
+			return block_number;
+		}
+
 		/**
 		 * Read data blocks from SD card
 		 *
@@ -408,7 +420,7 @@ struct Sdhci_controller : private Sdhci, public Sd_card::Host_controller
 
 			_set_block_count(block_count);
 
-			if (!issue_command(Read_multiple_block(block_number))) {
+			if (!issue_command(Read_multiple_block(_block_to_command_address(block_number)))) {
 				PERR("Read_multiple_block failed, Status: 0x%08x", read<Status>());
 				return false;
 			}
@@ -454,7 +466,7 @@ struct Sdhci_controller : private Sdhci, public Sd_card::Host_controller
 
 			_set_block_count(block_count);
 
-			if (!issue_command(Write_multiple_block(block_number))) {
+			if (!issue_command(Write_multiple_block(_block_to_command_address(block_number)))) {
 				PERR("Write_multiple_block failed, Status: 0x%08x", read<Status>());
 				return false;
 			}
diff --git a/repos/os/src/drivers/sd_card/sd_card.h b/repos/os/src/drivers/sd_card/sd_card.h
index d18e62a..0433620 100644
--- a/repos/os/src/drivers/sd_card/sd_card.h
+++ b/repos/os/src/drivers/sd_card/sd_card.h
@@ -67,7 +67,7 @@ namespace Sd_card {
 
 		struct Version : Bitfield<126 - BIT_BASE, 2>
 		{
-			enum { STANDARD_CAPACITY = 0, HIGH_CAPACITY = 1, EXT_CSD = 3 };
+			enum Type { STANDARD_CAPACITY = 0, HIGH_CAPACITY = 1, EXT_CSD = 3 };
 		};
 
 		struct Mmc_spec_vers : Bitfield<122 - BIT_BASE, 4> { };
@@ -325,11 +325,12 @@ namespace Sd_card {
 
 			unsigned _rca;
 			size_t   _capacity_mb;
+			const Csd3::Version::Type _version;
 
 		public:
 
-			Card_info(unsigned rca, size_t capacity_mb)
-			: _rca(rca), _capacity_mb(capacity_mb)
+			Card_info(unsigned rca, size_t capacity_mb, const Csd3::Version::Type version)
+			: _rca(rca), _capacity_mb(capacity_mb), _version(version)
 			{ }
 
 			/**
@@ -341,6 +342,12 @@ namespace Sd_card {
 			 * Return relative card address
 			 */
 			unsigned rca() const { return _rca; }
+
+			/**
+			 * Returns the version of the card
+			 */
+			Csd3::Version::Type version() const { return _version; }
+
 	};
 
 
@@ -480,7 +487,8 @@ namespace Sd_card {
 					throw Detection_failed();
 				}
 
-				return Card_info(rca, _sd_card_device_size(csd) / 2);
+				return Card_info(rca, _sd_card_device_size(csd) / 2,
+					static_cast<Csd3::Version::Type>(Csd3::Version::get(csd.csd3)));
 			}
 
 			Card_info _detect_mmc()
@@ -525,7 +533,8 @@ namespace Sd_card {
 					throw Detection_failed();
 				}
 
-				return Card_info(rca, device_size);
+				return Card_info(rca, device_size,
+					static_cast<Csd3::Version::Type>(Csd3::Version::get(csd.csd3)));
 			}
 	};
 }
```

ssumpf, 2015-10-01:
@twischer: I have tried your patch on the rpi, unfortunately the driver already fails at the 'Send_if_cond' command, so way before the patch. Thanks for your effort, but after some discussion we have decided that it is not worth to put further effort from our side into standard capacity cards. Also, Qemu is not a good test platform for peripherals, it is good for enabling them, but on hardware the result will usually not work. One minor thing, if you want to fix this issue, you are welcome to, but please sent us a real patch or a GIT commit, because your patch above will not apply since they whole tab/space intention breaks ... and one has to do it by hand. But still, thanks again.

nfeske, 2018-08-24:
This issue is stale for almost 3 years now.

closed 2018-08-24 by nfeske