genodelabs/genode #1664: usb: kingston sticks on usb2.0 ports not working

alex-ab, 2015-09-02:
For setting up a VM completely running from a USB stick, so Genode/Nova and the actual VM vdi image, i tried several 16 GB Kingston USB (USB3.0) sticks:

Datatraveler G4 16GB:
```
Bus 001 Device 011: ID 0951:1666 Kingston Technology
```

They work perfectly fine on a USB3.0 port, but doesn't work on a USB2.0 port. In contrast, I found and used a Intenso 16 GB stick (USB3.0), which works on USB2.0 as USB3.0 ports without any issue.

Intenso 16GB:
```
Bus 001 Device 012: ID 090c:1000 Silicon Motion, Inc. - Taiwan (formerly Feiya Technology Corp.) Flash Drive
```

alex-ab, 2015-09-02:
d73c15275033e3248a4043a23a622f7222c9f43d contains the relevant output of the dde_linux debug messages from the kingston as the intenso on a  usb2.0 controller

chelmuth, 2016-06-22:
Does 16.05 still suffer from this issue?

alex-ab, 2016-06-28:
Unfortunately, yes.

nfeske, 2018-05-31:
I guess this issue is obsolete now, @alex-ab? *crossing-fingers*

alex-ab, 2018-06-01:
Yes.

closed 2018-06-01 by alex-ab