genodelabs/genode #1665: usb: sporadic initialization hickup (ehci-mem.c)

nfeske, 2015-09-02:
Every once in a while when booting the Turmvilla scenario #1552, the USB driver fails to initialize. In this case, the LOG contains the following messages:

```
[init -> usb_drv] dev_info: new high-speed USB device number 2 using ehci-pci
[init -> usb_drv] dev_info: USB hub found
[init -> usb_drv] dev_info: 8 ports detected
[init -> usb_drv] dev_info: new high-speed USB device number 3 using ehci-pci
[init -> usb_drv] dev_info: USB hub found
[init -> usb_drv] dev_info: 4 ports detected
[init -> usb_drv] dev_info: new full-speed USB device number 4 using ehci-pci
[init -> usb_drv] dev_info: new full-speed USB device number 5 using ehci-pci
[init -> usb_drv] dev_info: new low-speed USB device number 6 using ehci-pci
[init -> usb_drv] dev_info: L U
[init -> usb_drv] dev_info: input: USB HID v1.11 Mouse [L U] on usb-dummy-1.2.1/input0
[init -> usb_drv] dev_info: new high-speed USB device number 7 using ehci-pci
[init] child "rtc_drv" announces service "Rtc"
[init -> config_rom] connection for file 'wlan_configuration' requested
[init -> platform_drv] 2:0.0 rewriting IRQ: 11 -> 16
[init -> usb_drv] dev_error: fatal error
[init -> usb_drv] dev_error: HC died; cleaning up
[init -> usb_drv] dev_error: string descriptor 0 read error: -2
[init -> usb_drv] dev_info: USB disconnect, device number 6
[init -> platform_drv] 2:0.0 rewriting IRQ: 16 -> 16
[init -> usb_drv] dev_info: USB disconnect, device number 7
[init -> usb_drv] BUG: failure at /home/no/src/genode/contrib/dde_linux-f127ceb9224245419b4fc046839a77555eb12c33/src/lib/usb/drivers/usb/host/ehci-mem.c:72/qh_destroy()!
```

In this case, the system fails to come up. Nitpicker and CLI monitor appear in screen but no prompt shows up in the CLI monitor window.

chelmuth, 2015-12-03:
Maybe we should double-check if this issue is still valid?

chelmuth, 2016-06-22:
Does this still happen with 16.05 or could we close this issue?

nfeske, 2016-06-22:
The issue refers to the old USB driver. I'll close it.

closed 2016-06-22 by nfeske